grblHAL/iMXRT1062 #1: SD Card support

infamous-panda, 2021-04-08:
What would the SD Card typically be used?

Is it for storing and running g-code programs? If so how would it be accessed from Gcode Sender?

terjeio, 2021-04-08:
> What would the SD Card typically be used? Is it for storing and running g-code programs? 

Yes.

>  If so how would it be accessed from Gcode Sender?

When detected a simple tab is presented with a list of the files on the card. Double click on the filename to run a file.

[image]

I need to add more functionality to this and tidy it up, already present in grblHAL is a rewind option where the same program can be run again by simply pressing _Cycle Start_. Upload to the card should be added as well, I am considering using the z-modem protocol for that. Upload has to be implemented at both ends.

infamous-panda, 2021-04-08:
Cool, Does this have a speed or reliability benefit over streaming usb or
ethernet?

terjeio, 2021-04-08:
> Does this have a speed or reliability benefit over streaming usb or ethernet?

I don't really know. I would guess reading data off the card is faster than usb or ethernet streaming. From the testing I have done (which is not much) reliability is good - cannot remember last time I saw any issues with the implementation.

Ok, I did a quick test streaming _Landscape.nc_ in check mode: via USB it took about 9 seconds, from the card less than one second. However, there is an issue with check mode I'll have to fix - program end is not handled correctly when streaming from the card...

infamous-panda, 2021-04-08:
That seems like a massive gain...

terjeio, 2021-04-30:
Check mode bug is fixed in the latest build, and ioSender is updated with a context menu in the file list and an Upload button if YModem file transfer is enabled in the controller (by setting `SDCARD_ENABLE` to 2). Upload functionality is experimental/in beta state for now as it sometimes fails. If it does the next try is usually successful.

closed 2021-04-30 by terjeio