grblHAL/iMXRT1062 #2: GrblHAL Hangs if EEPROM is not present - iMXRT1062

mlewus, 2021-06-15:
If the code is built for EEPROM and you boot without the EEPROM attached, the program does not respond to the USB serial port and appears to be hung.

This was an issue because the EEPROM is located on the breakout board but it is more convenient for me to build and load code with the Teensy out of its socket. I spent an hour debugging before I realized my error. Dumb, I know, but an error code would have been helpful. The same thing will presumably happen if the EEPROM becomes defective so it would be more generally useful than just my corner case.

I would suggest a code or message that resolves to something like "EEPROM not found" so it is clear that the build and load were successful even though the EEPROM is missing. If you will assign a new numeric code I'll make a pull request and make the required changes.

terjeio, 2021-06-15:
Use alarm 16 _Alarm_SelftestFailed_, see the [MSP432 i2.c code](https://github.com/grblHAL/MSP432P401R/blob/f2530ffc992552510a93121c11a7dafd52a653d7/i2c.c) for how to raise it and report the reason. That code justs checks for a stuck bus, you have add a check for the I2C EEPROM present as well.

mlewus, 2021-06-15:
That helps a lot, thanks. I also intend to implement the ability to set spindle speed as a RPM value. That is mostly a port of the work that you did in driver.C for MSP432P401R and was discussed in https://github.com/terjeio/grblHAL/discussions/298

Housekeeping: do you prefer that I create a fork to my own repository or do you wish to create a new branch in the project repository? Either way I will issue a pull request when I’m ready.

terjeio, 2021-06-15:
Create a fork to your own repo and a PR from that.