grblHAL/iMXRT1062 #5: Null pointer issue in TCPStream.c

andrewmarles, 2021-06-24:
I think there might be a problem with TCPStream.c that breaks ethernet on the IMXRT1062 at the moment.  I am pretty sure that the call hal.stream_select(NULL) goes nowhere and so while you can enable ethernet and the device will come up and ping, as soon as you open a telnet session and send a command it immediately locks up the board.

I reverted these calls to the previous selectStream function and it appears to work ok - I can open a telnet session and interact with the device as expected.

If there are updates needed to the iMX driver to align it with the updated stream switching functionality, I can certainly take a crack at a PR if there is an example driver that is known to work.  It looks to me like it is just that the stream_select function pointer isn't defined in the driver.

Cheers,

terjeio, 2021-06-24:
Are you sure your source code is up to date and that the Arduino IDE is using it? Line 2222 in driver.c sets the function pointer as it should - and it works for me, albeit with an yet to be released driver.c:

https://github.com/grblHAL/iMXRT1062/blob/9843c73a48d8e666c14b45af8caf48efac6f9f0a/grblHAL_Teensy4/src/driver.c#L2222-L2223

andrewmarles, 2021-06-24:
Thanks, my bad.  I had not merged properly with some local driver updates.

closed 2021-06-24 by andrewmarles