grblHAL/iMXRT1062 #7: Building with >3 Axes.

phil-barrett, 2021-07-14:
In grbl/config.h, line 38, set N_AXIS to 5
`#define N_AXIS 5 // Number of axes`
and compile.  Get numerous errors of the form

> D:\Documents\Arduino\libraries\grblHAL_Teensy4\src\T41U5XBB_map.h:26:2: error: #error "Axis configuration is not supported!"
>  #error "Axis configuration is not supported!"
>   
When N_AXIS is set to 4, the problem does not occur.

This occurs because N_ABC_MOTORS is getting set to 3 when N_AXIS is 5. 
I put #warning messages in driver_opts.h and T41U5XBB_map.h and it looks like N_ABC_AXIS is 2 and N_GANGED is 0 so when N_ABC_MOTORS is set in driver_opts.h it should be 2. Yet in T41U5XBB_map.h, N_ABC_MOTORS is 3.
I placed this in T41U5XBB_map.h before the error test around line 24.
```
#if N_ABC_MOTORS == 3
#warning "N_ABC_MOTORS == 3"
#elif N_ABC_MOTORS == 4
#warning "N_ABC_MOTORS == 4"
#endif
```
The compiler prints out:

> #warning "N_ABC_MOTORS == 3"

Using Arduino 1.8.15 and Teensyduino 1.54

my_machine in both cases (4 and 5 AXIS):
```
/*
  my_machine.h - configuration for IMXRT1062 processor (on Teensy 4.x board)

  Part of grblHAL

  Copyright (c) 2020 Terje Io

  Grbl is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  Grbl is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  along with Grbl.  If not, see <http://www.gnu.org/licenses/>.
*/

// BOARD_T40X101 and BOARD_T41U5XBB by Phil Barrett: https://github.com/phil-barrett/grblHAL-teensy-4.x

// NOTE: Only one board may be enabled!
// If none is enabled pin mappings from generic_map.h will be used
//#define BOARD_T40X101
#define BOARD_T41U5XBB
//#define BOARD_T41U5XBB_SS // For a modified T41U5XBB board, allows spindle sync to be enabled.
//#define BOARD_T41BB5X_PRO
//#define BOARD_CNC_BOOSTERPACK
//#define BOARD_MY_MACHINE // Add my_machine_map.h before enabling this!

// Configuration
// Uncomment to enable, for some a value > 1 may be assigned, if so the default value is shown.

/*
              Plugin: | ETHERNET¹ | SDCARD¹ | KEYPAD | EEPROM | N_AXIS |
----------------------|-----------|---------|--------|--------|--------|
BOARD_T40X101         | no        | no      | yes    | yes³   | max 4  |
BOARD_T41U5XBB        | yes       | yes     | yes    | yes³   | max 5  |
BOARD_T41BB5X_PRO     | yes       | yes     | yes    | yes    | max 5  |
BOARD_CNC_BOOSTERPACK | yes²      | yes     | yes    | yes    | max 3  |

¹ Teensy 4.1 only
² External magjack.
³ EEPROM is optional and must be added to the board.

N_AXIS has a default value of 3, edit grbl\config.h to increase.

*/

#define USB_SERIAL_CDC      2 // 1 for Arduino class library and 2 for PJRC C library. Comment out to use UART communication.
//#define USB_SERIAL_WAIT     1 // Wait for USB connection before starting grblHAL.
//#define BLUETOOTH_ENABLE    1 // Set to 1 for HC-05 module. Requires Bluetooth plugin.
//#define SPINDLE_HUANYANG    1 // Set to 1 or 2 for Huanyang VFD spindle. Requires spindle plugin.
//#define QEI_ENABLE          1 // Enable quadrature encoder interfaces. Max value is 1. Requires encoder plugin.
#define ETHERNET_ENABLE     1 // Ethernet streaming. Requires networking plugin.
//#define SDCARD_ENABLE       1 // Run gcode programs from SD card, requires sdcard plugin.
//#define KEYPAD_ENABLE       1 // I2C keypad for jogging etc., requires keypad plugin.
//#define PLASMA_ENABLE       1 // Plasma/THC plugin. To be completed.
//#define MCP3221_ENABLE      1 // Enable analog input via MCP3221 ADC.
//#define PPI_ENABLE          1 // Laser PPI plugin. To be completed.
//#define ODOMETER_ENABLE     1 // Odometer plugin.
//#define OPENPNP_ENABLE      1 // OpenPNP plugin. To be completed.
//#define FANS_ENABLE         1 // Enable fan control via M106/M107. Requires fans plugin.
//#define EEPROM_ENABLE       1 // I2C EEPROM support. Set to 1 for 24LC16(2K), 2 for larger sizes. Requires eeprom plugin.
//#define EEPROM_IS_FRAM      1 // Uncomment when EEPROM is enabled and chip is FRAM, this to remove write delay.
//#define SPINDLE_SYNC_ENABLE 1 // Enable spindle sync support (G33, G76). !! NOTE: Alpha quality - enable only for test or verification.
                                // Currently only available for BOARD_T41BB5X_PRO and BOARD_T41U5XBB_SS.
//#define ESTOP_ENABLE        0 // When enabled only real-time report requests will be executed when the reset pin is asserted.
                                // Note: if left commented out the default setting is determined from COMPATIBILITY_LEVEL.

// If the selected board map supports more than three motors ganging and/or auto-squaring
// of axes can be enabled here.
//#define X_GANGED            1
//#define X_AUTO_SQUARE       1
//#define Y_GANGED            1
//#define Y_AUTO_SQUARE       1
//#define Z_GANGED            1
//#define Z_AUTO_SQUARE       1
// For ganged axes the limit switch input (if available) can be configured to act as a max travel limit switch.
// NOTE: If board map already has max limit inputs defined this configuration will be ignored.
//#define X_GANGED_LIM_MAX    1
//#define Y_GANGED_LIM_MAX    1
//#define Z_GANGED_LIM_MAX    1
//

#if ETHERNET_ENABLE > 0
#define TELNET_ENABLE           1 // Telnet daemon - requires Ethernet streaming enabled.
#define FTP_ENABLE              1 // Ftp daemon - requires SD card enabled.
#define WEBSOCKET_ENABLE        1 // Websocket daemon - requires Ethernet streaming enabled.
#define NETWORK_HOSTNAME        "GRBL"
#define NETWORK_IPMODE          1 // 0 = static, 1 = DHCP, 2 = AutoIP
#define NETWORK_IP              "192.168.5.1"
#define NETWORK_GATEWAY         "192.168.5.1"
#define NETWORK_MASK            "255.255.255.0"
#define NETWORK_TELNET_PORT     23
#define NETWORK_WEBSOCKET_PORT  80
#define NETWORK_HTTP_PORT       80
#endif

```

phil-barrett, 2021-07-14:
This may be related.  When enabling ganging (uncommenting #define Y_GANGED 1 in my_machine.h), it uses the next available axis motor rather than the last one. So when N_AXIS is 3, it uses A as the ganged motor.  When N_AXIS is 4, it uses B. 

I don't know if it was intentional, there may be a good reason to do it this way.  But someone upgrading a 3 Axis machine with a ganged axis to the latest grblHAL will have to either say they have 4 axes or change their wiring. That is a pretty common configuration - lots of WorkBees out there.

terjeio, 2021-07-14:
Odd, I do not get the error. `N_ABC_MOTORS` is defined in _grbl/driver_opts.h_, is `N_GANGED` somehow getting defined to 1 for you there when it should not? Can you check that?

> This may be related. When enabling ganging (uncommenting #define Y_GANGED 1 in my_machine.h), it uses the next available axis motor rather than the last one.

I do not believe it is in any way related. Pin assignments is done in _grbl/motor_pins.h_ and it uses defined values from _grbl/driver_opts.h_ to do that. Could it be the Arduino build system messing up - it copies the source code to a temp folder before compiling and sometims that fails (by not copying all changed files).

> I don't know if it was intentional, there may be a good reason to do it this way.

This is intentional, and is warned about in the [changelog](https://github.com/grblHAL/core/blob/master/changelog.md).

One of the reasons behind this was to always leave the unused motor pins from top down so that they could be reassigned as aux outputs without too complex preprocessor code. This approach may also have made the pins vs. motor assignment preprocessor code simpler?

phil-barrett, 2021-07-14:
I don't see it anywhere in the driver_opts.h.  I will dredge through the build output to see if there is something.  Strange that it changes.  I will drop in a bunch more "warning #ifs" to try and track down where it happens. Dredging though #ifs makes my head hurt.

I understand why you made the change.  I will post a note for my customers to be aware of it. Early adopter tax?

terjeio, 2021-07-14:
Calculating number of motors is done [here](https://github.com/grblHAL/core/blob/222ba55724741b1ab3b699f3ae4953cc90bc30f5/driver_opts.h#L74-L76).

I forgot about the new `$pins` command when I replied, what is the output when `N_AXIS` is four? Are all five motors having pins assigned?

> I understand why you made the change. I will post a note for my customers to be aware of it. Early adopter tax?

I guess so, the good thing is that the settings are now consistent across drivers and easier to set up in the map file.
The RP2040 driver will get ganged/squared axis support configured the same way even when using shift registers for the step/dir signals...

phil-barrett, 2021-07-14:
In the case of N_AXIS == 4 and Y_GANGED == 1,  it works correctly. ioSender see XYZA and the B motor is correctly ganged to Y.  Here is the $pins output. It shows the Y ganging (though is ambiguous - could we have Y1 and Y2 instead of just Y in both cases?) this is for T41BB5X_Pro_map.h, by the way.

> $pins
> [PIN:40,Emergency stop]
> [PIN:16,Feed hold]
> [PIN:17,Cycle start]
> [PIN:15,Probe]
> [PIN:20,X limit min]
> [PIN:21,Y limit min]
> [PIN:22,Z limit min]
> [PIN:23,A limit min]
> [PIN:36,Aux input 0]
> [PIN:30,Aux input 1]
> [PIN:31,Aux input 2]
> [PIN:14,Aux input 3]
> [PIN:2,X step]
> [PIN:4,Y step]
> [PIN:6,Z step]
> [PIN:8,A step]
> [PIN:26,Y step]
> [PIN:3,X dir]
> [PIN:5,Y dir]
> [PIN:7,Z dir]
> [PIN:9,A dir]
> [PIN:27,Y dir]
> [PIN:10,X enable]
> [PIN:35,Y enable]
> [PIN:39,Z enable]
> [PIN:38,A enable]
> [PIN:37,Y enable]
> [PIN:12,Spindle on]
> [PIN:11,Spindle direction]
> [PIN:19,Flood]
> [PIN:18,Mist]
> [PIN:34,Aux out 0]
> [PIN:32,Aux out 1]
> [PIN:33,Aux out 2]

When I have N_AXIS 4 (in grbl/config.h) and #define Y_GANGED 1 in my_machine.h, I see this output from the compiler
> d:\documents\arduino\libraries\grblhal_teensy4\src\grbl/driver_opts.h:76:2: warning: #warning "N_ABC_AXIS == 1" [-Wcpp]
>  #warning "N_ABC_AXIS == 1"
>   ^
> d:\documents\arduino\libraries\grblhal_teensy4\src\grbl/driver_opts.h:92:2: warning: #warning "N_GANGED == 0" [-Wcpp]
>  #warning "N_GANGED == 0"
>   ^

Here is the code I added to driver_opts.h starting at line 73.

```
#if N_ABC_AXIS == 1
#warning "N_ABC_AXIS == 1"
#endif

#if N_ABC_AXIS == 2
#warning "N_ABC_AXIS == 2"
#endif

#if N_ABC_AXIS == 3
#warning "N_ABC_AXIS == 3"
#endif

#if N_ABC_AXIS == 4
#warning "N_ABC_AXIS == 4"
#endif

#if N_GANGED == 0
#warning "N_GANGED == 0"
#endif

#if N_GANGED == 1
#warning "N_GANGED == 1"
#endif

#if N_GANGED == 2
#warning "N_GANGED == 2"
#endif

```

phil-barrett, 2021-07-14:
With N_AXIS set to 5 in grbl/config.h, I get these errors:

> In file included from D:\Documents\Arduino\libraries\grblHAL_Teensy4\src\driver.h:72:0,
>                  from D:\Documents\Arduino\libraries\grblHAL_Teensy4\src\driver.c:27:
> D:\Documents\Arduino\libraries\grblHAL_Teensy4\src\T41BB5X_Pro_map.h:25:2: error: #error "Axis configuration is not supported!"
>  #error "Axis configuration is not supported!"
>   ^
> In file included from D:\Documents\Arduino\libraries\grblHAL_Teensy4\src\driver.c:87:0:
> D:\Documents\Arduino\libraries\grblHAL_Teensy4\src\grbl/motor_pins.h:424:4: error: #error "B_AXIS pins are not available"
>    #error "B_AXIS pins are not available"
>     ^
> D:\Documents\Arduino\libraries\grblHAL_Teensy4\src\grbl/motor_pins.h:427:29: error: 'M4_STEP_PIN' undeclared here (not in a function)
>  #define B_STEP_PIN          M4_STEP_PIN
>                              ^
> D:\Documents\Arduino\libraries\grblHAL_Teensy4\src\driver.c:357:68: note: in expansion of macro 'B_STEP_PIN'
>      { .id = Output_StepB,           .port = &stepB,         .pin = B_STEP_PIN,              .group = PinGroup_StepperStep },
>                                                                     ^
> In file included from D:\Documents\Arduino\libraries\grblHAL_Teensy4\src\driver.c:87:0:
> D:\Documents\Arduino\libraries\grblHAL_Teensy4\src\grbl/motor_pins.h:237:29: error: 'M5_STEP_PIN' undeclared here (not in a function)
>  #define Y2_STEP_PIN         M5_STEP_PIN
>                              ^
> D:\Documents\Arduino\libraries\grblHAL_Teensy4\src\driver.c:366:68: note: in expansion of macro 'Y2_STEP_PIN'
>      { .id = Output_StepY,           .port = &stepY2,        .pin = Y2_STEP_PIN,             .group = PinGroup_StepperStep },
>                                                                     ^
> In file included from D:\Documents\Arduino\libraries\grblHAL_Teensy4\src\driver.c:87:0:
> D:\Documents\Arduino\libraries\grblHAL_Teensy4\src\grbl/motor_pins.h:430:29: error: 'M4_DIRECTION_PIN' undeclared here (not in a function)
>  #define B_DIRECTION_PIN     M4_DIRECTION_PIN
>                              ^
> D:\Documents\Arduino\libraries\grblHAL_Teensy4\src\driver.c:378:68: note: in expansion of macro 'B_DIRECTION_PIN'
>      { .id = Output_DirB,            .port = &dirB,          .pin = B_DIRECTION_PIN,         .group = PinGroup_StepperDir },
>                                                                     ^
> In file included from D:\Documents\Arduino\libraries\grblHAL_Teensy4\src\driver.c:87:0:
> D:\Documents\Arduino\libraries\grblHAL_Teensy4\src\grbl/motor_pins.h:240:29: error: 'M5_DIRECTION_PIN' undeclared here (not in a function)
>  #define Y2_DIRECTION_PIN    M5_DIRECTION_PIN
>                              ^
> D:\Documents\Arduino\libraries\grblHAL_Teensy4\src\driver.c:387:68: note: in expansion of macro 'Y2_DIRECTION_PIN'
>      { .id = Output_DirY,            .port = &dirY2,         .pin = Y2_DIRECTION_PIN,        .group = PinGroup_StepperDir },
>

terjeio, 2021-07-14:
> though is ambiguous - could we have Y1 and Y2 instead of just Y in both cases?

I think so, more work has to be done on the $pins reporting. $pins output should also be extended to include pins such as UART, PWM, I2C etc. which are not plain GPIO. One step at a time... 

What happens if you compile 5 axes for the pro board? Still erroring out? If not then your T41U5XBB_map.h is bad?
I just copied it from github, again it compiles without errors for me.

phil-barrett, 2021-07-14:
Both the Pro and regular board have the same errors. Both map files are from the repo - I pulled them down July 10.  This is the full build output from Arduino when N_AXIS is set to 5 (in grbl/config.h).  Using the Pro map file (selected from my_machine.h) Unfortunately, it wraps lines so the output is hard to read.

Note the section ResolveLibrary(grblHAL_Teensy4.h) starting at actual line 9.  There are several errors that say "Error while detecting libraries included by D:\Documents\Arduino\libraries\grblHAL_Teensy4\src\driver.c"  (the file name changes, same path, though). The log file for N_AXIS 4 build is the same except no error messages (i.e., same ResolveLibrary, Alternatives for, and so on messages). 

In the N_AXIS 4 situation, it does not generate the errors.

> C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\philba\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\philba\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Documents\Arduino\libraries -fqbn=teensy:avr:teensy41:usb=serial,speed=600,opt=o2std,keys=en-us -ide-version=10815 -build-path C:\Users\philba\AppData\Local\Temp\arduino_build_838793 -warnings=none -build-cache C:\Users\philba\AppData\Local\Temp\arduino_cache_662115 -verbose D:\Documents\Arduino\libraries\grblHAL_Teensy4\examples\grblHAL_Teensy4_Upload\grblHAL_Teensy4_Upload.ino
> C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\philba\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\philba\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Documents\Arduino\libraries -fqbn=teensy:avr:teensy41:usb=serial,speed=600,opt=o2std,keys=en-us -ide-version=10815 -build-path C:\Users\philba\AppData\Local\Temp\arduino_build_838793 -warnings=none -build-cache C:\Users\philba\AppData\Local\Temp\arduino_cache_662115 -verbose D:\Documents\Arduino\libraries\grblHAL_Teensy4\examples\grblHAL_Teensy4_Upload\grblHAL_Teensy4_Upload.ino
> Using board 'teensy41' from platform in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr
> Using core 'teensy4' from platform in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr
> Detecting libraries used...
> "C:\\Program Files (x86)\\Arduino\\hardware\\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=154 -DARDUINO=10815 -DARDUINO_TEENSY41 -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\cores\\teensy4" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_838793\\sketch\\grblHAL_Teensy4_Upload.ino.cpp" -o nul
> Alternatives for grblHAL_Teensy4.h: [grblHAL_Teensy4@1.1.7]
> ResolveLibrary(grblHAL_Teensy4.h)
>   -> candidates: [grblHAL_Teensy4@1.1.7]
> "C:\\Program Files (x86)\\Arduino\\hardware\\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=154 -DARDUINO=10815 -DARDUINO_TEENSY41 -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\cores\\teensy4" "-ID:\\Documents\\Arduino\\libraries\\grblHAL_Teensy4\\src" "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_838793\\sketch\\grblHAL_Teensy4_Upload.ino.cpp" -o nul
> "C:\\Program Files (x86)\\Arduino\\hardware\\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=154 -DARDUINO=10815 -DARDUINO_TEENSY41 -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\cores\\teensy4" "-ID:\\Documents\\Arduino\\libraries\\grblHAL_Teensy4\\src" "D:\\Documents\\Arduino\\libraries\\grblHAL_Teensy4\\src\\bluetooth\\hc_05.c" -o nul
> "C:\\Program Files (x86)\\Arduino\\hardware\\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=154 -DARDUINO=10815 -DARDUINO_TEENSY41 -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\cores\\teensy4" "-ID:\\Documents\\Arduino\\libraries\\grblHAL_Teensy4\\src" "D:\\Documents\\Arduino\\libraries\\grblHAL_Teensy4\\src\\driver.c" -o nul
[... cut ...]
> D:\Documents\Arduino\libraries\grblHAL_Teensy4\src\driver.c:148:28: warning: 'LimitB' defined but not used [-Wunused-variable]
>  static gpio_t stepB, dirB, LimitB;
>                             ^
> Using previously compiled file: C:\Users\philba\AppData\Local\Temp\arduino_build_838793\libraries\grblHAL_Teensy4\main.c.o
> Using library grblHAL_Teensy4 at version 1.1.7 in folder: D:\Documents\Arduino\libraries\grblHAL_Teensy4 
> Using library teensy41_ethernet-master at version 2.0.2 in folder: D:\Documents\Arduino\libraries\teensy41_ethernet-master 
> Error compiling for board Teensy 4.1.
>

terjeio, 2021-07-14:
Does it compile if you remove the guard for the M4 motor defines? Change
```
#if N_ABC_MOTORS == 2
#define M4_AVAILABLE
#define M4_STEP_PIN         (26u)
#define M4_DIRECTION_PIN    (27u)
#define M4_LIMIT_PIN        (28u)
#define M4_ENABLE_PIN       (37u)
#endif
```
to
```
#define M4_AVAILABLE
#define M4_STEP_PIN         (26u)
#define M4_DIRECTION_PIN    (27u)
#define M4_LIMIT_PIN        (28u)
#define M4_ENABLE_PIN       (37u)
```

If you add warnings for N_ABC_MOTORS what is the output? It should be 2 but is 1?.

phil-barrett, 2021-07-14:
In the case of N_AXIS 5, N_ABC_MOTORS is 3.  (should be 2?)

> In file included from D:\Documents\Arduino\libraries\grblHAL_Teensy4\src\driver.h:72:0,
>                  from D:\Documents\Arduino\libraries\grblHAL_Teensy4\src\enet.c:22:
> D:\Documents\Arduino\libraries\grblHAL_Teensy4\src\T41BB5X_Pro_map.h:25:2: error: #error "Axis configuration is not supported!"
>  #error "Axis configuration is not supported!"
>   ^
> In file included from D:\Documents\Arduino\libraries\grblHAL_Teensy4\src\driver.h:37:0,
>                  from D:\Documents\Arduino\libraries\grblHAL_Teensy4\src\driver.c:27:
> D:\Documents\Arduino\libraries\grblHAL_Teensy4\src\grbl/driver_opts.h:80:2: warning: #warning "N_ABC_AXIS == 2" [-Wcpp]
>  #warning "N_ABC_AXIS == 2"
>   ^
> D:\Documents\Arduino\libraries\grblHAL_Teensy4\src\grbl/driver_opts.h:92:2: warning: #warning "N_GANGED == 0" [-Wcpp]
>  #warning "N_GANGED == 0"
>   ^
> D:\Documents\Arduino\libraries\grblHAL_Teensy4\src\grbl/driver_opts.h:114:2: warning: #warning "driver_opts: N_ABC_MOTORS == 3" [-Wcpp]
>  #warning "driver_opts: N_ABC_MOTORS == 3"
>   ^
> In file included from D:\Documents\Arduino\libraries\grblHAL_Teensy4\src\driver.h:72:0,
>                  from D:\Documents\Arduino\libraries\grblHAL_Teensy4\src\driver.c:27:
> 

Here is the code in drivr_opts.h that generates that output:
```
#ifndef X_GANGED
#define X_GANGED            0
#endif
#ifndef X_AUTO_SQUARE
#define X_AUTO_SQUARE       0
#endif
#ifndef Y_GANGED
#define Y_GANGED            0
#endif
#ifndef Y_AUTO_SQUARE
#define Y_AUTO_SQUARE       0
#endif
#ifndef Z_GANGED
#define Z_GANGED            0
#endif
#ifndef Z_AUTO_SQUARE
#define Z_AUTO_SQUARE       0
#endif
#ifndef X_GANGED_LIM_MAX
#define X_GANGED_LIM_MAX    0
#endif
#ifndef Y_GANGED_LIM_MAX
#define Y_GANGED_LIM_MAX    0
#endif
#ifndef Z_GANGED_LIM_MAX
#define Z_GANGED_LIM_MAX    0
#endif

#if X_AUTO_SQUARE && !X_GANGED
#undef X_GANGED
#define X_GANGED 1
#endif
#if Y_AUTO_SQUARE && !Y_GANGED
#undef Y_GANGED
#define Y_GANGED 1
#endif
#if Z_AUTO_SQUARE && !Z_GANGED
#undef Z_GANGED
#define Z_GANGED 1
#endif

#if N_ABC_AXIS == 1
#warning "N_ABC_AXIS == 1"
#endif

#if N_ABC_AXIS == 2
#warning "N_ABC_AXIS == 2"
#endif

#if N_ABC_AXIS == 3
#warning "N_ABC_AXIS == 3"
#endif

#if N_ABC_AXIS == 4
#warning "N_ABC_AXIS == 4"
#endif

#if N_GANGED == 0
#warning "N_GANGED == 0"
#endif

#if N_GANGED == 1
#warning "N_GANGED == 1"
#endif

#if N_GANGED == 2
#warning "N_GANGED == 2"
#endif

#define N_GANGED (X_GANGED + Y_GANGED + Z_GANGED)
#define N_AUTO_SQUARED (X_AUTO_SQUARE + Y_AUTO_SQUARE + Z_AUTO_SQUARE)
#define N_ABC_MOTORS (N_ABC_AXIS + N_GANGED)

#if N_ABC_MOTORS == 1
#warning "driver_opts: N_ABC_MOTORS == 1"
#endif
#if N_ABC_MOTORS == 2
#warning "driver_opts: N_ABC_MOTORS == 2"
#endif
#if N_ABC_MOTORS == 3
#warning "driver_opts: N_ABC_MOTORS == 3"
#endif

```

phil-barrett, 2021-07-14:
And finally, when N_AXIS is 4, N_ABC_MOTORS is 2.  Off by one error somewhere?

terjeio, 2021-07-14:
> And finally, when N_AXIS is 4, N_ABC_MOTORS is 2. Off by one error somewhere?

Thanks, this is obviously wrong.

Change
`#define N_ABC_MOTORS (N_ABC_AXIS + N_GANGED)`
to
`#define N_ABC_MOTORS N_ABC_AXIS`

Compiles? Somehow `N_GANGED` is not 0?

terjeio, 2021-07-14:
`N_ABC_MOTORS` is only defined in driver_opts.h and not modified elsewhere. Is the Arduino build system playing games with us?
Have you tried with deleting the arduino_build_* and arduino_cache_* folders created by the IDE in your temp folder?
If not, close the IDE and do so. Search for `%TEMP%` to open your temp folder.

phil-barrett, 2021-07-14:
Closed Arduino IDE, emptied temp folder (it did contain arduino build and cache dirs).  No change, same problem.

Changed 
`#define N_ABC_MOTORS (N_ABC_AXIS + N_GANGED)`
to
`#define N_ABC_MOTORS N_ABC_AXIS`

Now that causes an expected error and not the cascade of other errors.

> In file included from D:\Documents\Arduino\libraries\grblHAL_Teensy4\src\driver.c:87:0:
> D:\Documents\Arduino\libraries\grblHAL_Teensy4\src\grbl/motor_pins.h:199:2: error: #error "B-axis motor is used for ganged Y motor"
>  #error "B-axis motor is used for ganged Y motor"
>   ^
> In file included from D:\Documents\Arduino\libraries\grblHAL_Teensy4\src\driver.h:37:0,
>                  from D:\Documents\Arduino\libraries\grblHAL_Teensy4\src\enet.c:22:
>

terjeio, 2021-07-14:
Ok, so `Y_GANGED` is non-zero, I assume you have commented out `Y_GANGED` and `Y_AUTO_SQUARE` in my_machine.h so it must be set non-zero somewhere else - but where is that? `Y_GANGED` is set in my_machine.h if commented in, set to1 or 0 in driver_opts.h if unset depending on whether Y_AUTO_SQUARE is uncommented in in my_machine.h, and referenced in motor_pins.h. Let us explicitly set it to 0 by adding
```
#undef Y_GANGED
#define Y_GANGED 0
```
just before 
```
#define N_GANGED (X_GANGED + Y_GANGED + Z_GANGED)
#define N_AUTO_SQUARED (X_AUTO_SQUARE + Y_AUTO_SQUARE + Z_AUTO_SQUARE)
#define N_ABC_MOTORS (N_ABC_AXIS + N_GANGED)
```
in driver_opts.h and try again. Still failing?

phil-barrett, 2021-07-14:
But, that change causes N_AXIS 4 to get this same error for the A AXIS

> In file included from D:\Documents\Arduino\libraries\grblHAL_Teensy4\src\driver.c:87:0:
> D:\Documents\Arduino\libraries\grblHAL_Teensy4\src\grbl/motor_pins.h:164:2: error: #error "A-axis motor is used for ganged Y motor"
>  #error "A-axis motor is used for ganged Y motor"
>   ^
> In file included from D:\Documents\Arduino\libraries\grblHAL_Teensy4\src\driver.h:37:0,
>                  from D:\Documents\Arduino\libraries\grblHAL_Teensy4\src\i2c.c:12:
>

phil-barrett, 2021-07-14:
I need to do a quick reset here.  In the past, setting N_AXIS to 5 allowed ganging of an axis.  I was assuming that is still the case and duplicated the previous settings in my_machine and map files.  But, I am a bit slow sometimes - N_AXIS 5 and Y_GANGED 1 should generate an error in the new system.   Though, the cascade of errors led me to think it was a problem.  The onslaught of errors is a consequence of the Arduino IDE, unlike a real Make system that will exit on first error return.

Am I right here?

terjeio, 2021-07-14:
>  In the past, setting N_AXIS to 5 allowed ganging of an axis.

Possibly due to no checks beeing done in the past. If N_AXIS is 5 you have to have a six motor BOB if enabling a ganged/squared axis.
Now it triggers the, error as it should, if the configuration is for more motors than available.

> The onslaught of errors is a consequence of the Arduino IDE, unlike a real Make system that will exit on first error return.

I just tried with STM32CubeIDE, there is a bunch of error generated but nothing like the Arduino output. I guess it depends on how many places driver.h is beeing included.

```
21:46:42 **** Build of configuration Debug F411 Nucleo64 for project GRBL Driver STM32F401 ****
make -j6 all 
arm-none-eabi-gcc "../spindle/huanyang.c" -mcpu=cortex-m4 -std=gnu11 -g -DUSE_HAL_DRIVER -DDEBUG -DSTM32F411xE -DNUCLEO_F411 -c -I"D:/Projects/STM/GRBL Driver STM32F401" -I../Inc -I../Drivers/CMSIS/Include -I../Drivers/STM32F4xx_HAL_Driver/Inc -I../Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -I../Drivers/CMSIS/Device/ST/STM32F4xx/Include -I../Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I../Drivers/STM32F4xx_HAL_Driver/Inc/Legacy -I"D:/Projects/STM/GRBL Driver STM32F401/FatFs" -I../USB_DEVICE/App -I../USB_DEVICE/Target -Os -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"spindle/huanyang.d" -MT"spindle/huanyang.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "spindle/huanyang.o"
arm-none-eabi-gcc "../spindle/modbus.c" -mcpu=cortex-m4 -std=gnu11 -g -DUSE_HAL_DRIVER -DDEBUG -DSTM32F411xE -DNUCLEO_F411 -c -I"D:/Projects/STM/GRBL Driver STM32F401" -I../Inc -I../Drivers/CMSIS/Include -I../Drivers/STM32F4xx_HAL_Driver/Inc -I../Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -I../Drivers/CMSIS/Device/ST/STM32F4xx/Include -I../Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I../Drivers/STM32F4xx_HAL_Driver/Inc/Legacy -I"D:/Projects/STM/GRBL Driver STM32F401/FatFs" -I../USB_DEVICE/App -I../USB_DEVICE/Target -Os -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"spindle/modbus.d" -MT"spindle/modbus.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "spindle/modbus.o"
arm-none-eabi-gcc "../sdcard/sdcard.c" -mcpu=cortex-m4 -std=gnu11 -g -DUSE_HAL_DRIVER -DDEBUG -DSTM32F411xE -DNUCLEO_F411 -c -I"D:/Projects/STM/GRBL Driver STM32F401" -I../Inc -I../Drivers/CMSIS/Include -I../Drivers/STM32F4xx_HAL_Driver/Inc -I../Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -I../Drivers/CMSIS/Device/ST/STM32F4xx/Include -I../Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I../Drivers/STM32F4xx_HAL_Driver/Inc/Legacy -I"D:/Projects/STM/GRBL Driver STM32F401/FatFs" -I../USB_DEVICE/App -I../USB_DEVICE/Target -Os -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"sdcard/sdcard.d" -MT"sdcard/sdcard.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "sdcard/sdcard.o"
arm-none-eabi-gcc "../sdcard/ymodem.c" -mcpu=cortex-m4 -std=gnu11 -g -DUSE_HAL_DRIVER -DDEBUG -DSTM32F411xE -DNUCLEO_F411 -c -I"D:/Projects/STM/GRBL Driver STM32F401" -I../Inc -I../Drivers/CMSIS/Include -I../Drivers/STM32F4xx_HAL_Driver/Inc -I../Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -I../Drivers/CMSIS/Device/ST/STM32F4xx/Include -I../Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I../Drivers/STM32F4xx_HAL_Driver/Inc/Legacy -I"D:/Projects/STM/GRBL Driver STM32F401/FatFs" -I../USB_DEVICE/App -I../USB_DEVICE/Target -Os -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"sdcard/ymodem.d" -MT"sdcard/ymodem.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "sdcard/ymodem.o"
arm-none-eabi-gcc "../odometer/odometer.c" -mcpu=cortex-m4 -std=gnu11 -g -DUSE_HAL_DRIVER -DDEBUG -DSTM32F411xE -DNUCLEO_F411 -c -I"D:/Projects/STM/GRBL Driver STM32F401" -I../Inc -I../Drivers/CMSIS/Include -I../Drivers/STM32F4xx_HAL_Driver/Inc -I../Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -I../Drivers/CMSIS/Device/ST/STM32F4xx/Include -I../Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I../Drivers/STM32F4xx_HAL_Driver/Inc/Legacy -I"D:/Projects/STM/GRBL Driver STM32F401/FatFs" -I../USB_DEVICE/App -I../USB_DEVICE/Target -Os -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"odometer/odometer.d" -MT"odometer/odometer.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "odometer/odometer.o"
arm-none-eabi-gcc "../motors/trinamic.c" -mcpu=cortex-m4 -std=gnu11 -g -DUSE_HAL_DRIVER -DDEBUG -DSTM32F411xE -DNUCLEO_F411 -c -I"D:/Projects/STM/GRBL Driver STM32F401" -I../Inc -I../Drivers/CMSIS/Include -I../Drivers/STM32F4xx_HAL_Driver/Inc -I../Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -I../Drivers/CMSIS/Device/ST/STM32F4xx/Include -I../Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I../Drivers/STM32F4xx_HAL_Driver/Inc/Legacy -I"D:/Projects/STM/GRBL Driver STM32F401/FatFs" -I../USB_DEVICE/App -I../USB_DEVICE/Target -Os -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"motors/trinamic.d" -MT"motors/trinamic.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "motors/trinamic.o"
In file included from ../Inc/driver.h:132,
                 from ../odometer/odometer.c:27:
../Inc/st_morpho_map.h:32:2: error: #error "Axis configuration is not supported!"
   32 | #error "Axis configuration is not supported!"
      |  ^~~~~
In file included from ../Inc/driver.h:132,
                 from ../sdcard/sdcard.h:30,
[... cut ...]
make: *** [sdcard/subdir.mk:24: sdcard/ymodem.o] Error 1
make: *** Waiting for unfinished jobs....
make: *** [sdcard/subdir.mk:22: sdcard/sdcard.o] Error 1
make: *** [motors/subdir.mk:19: motors/trinamic.o] Error 1
make: *** [odometer/subdir.mk:19: odometer/odometer.o] Error 1
make: *** [spindle/subdir.mk:22: spindle/huanyang.o] Error 1
"make -j6 all" terminated with exit code 2. Build might be incomplete.

21:46:43 Build Failed. 11 errors, 0 warnings. (took 1s.74ms)
```

> Am I right here?

I am not sure - it depends on compiler settings?

phil-barrett, 2021-07-15:
Sorry to have wasted your time.  I should have done some study before posting.  Will close.

On the upgrading-ganged-machines issue.  I will craft a map file that swaps A and B pins so that the customer doesn't have to change their wiring.

closed 2021-07-15 by phil-barrett