grblHAL/iMXRT1062 #10: Limit switch/sensor issues and Homing

phil-barrett, 2021-07-23:
Recent build (source updated July 3).  I have NO sensors (NPN Prox) so inverted limit pins ($5=15).  They work, I get Alarm 1 when hard limits are enabled and a is sensor triggered.  However, homing does not work.  When I start homing, the first Axis (Z) moves a short distance and throws an Alarm 8. Individual axis homing ($HX, $HY, $HZ) does the same thing - short move then alarm. I believe it related to how NO sensors are handled. 

I have tested this on a bare teensy with T41U5XBB_map.h and T41BB5X_map.h selected from my machine. Both behave the same.  I first found this while testing homing on my Avid Pro4848 running grblHAL on the T41ProBB board..

It looks to be correctly working with $5=0, i.e. NC sensors. With the limit pins not connected to ground (i.e. open and thus triggered),  I get this:
> <Idle|MPos:0.000,0.000,0.000,0.000|Bf:35,1023|FS:0,0|Pn:PXYZA>
and if I short a limit pin to ground, say X, I will get
> <Idle|MPos:0.000,0.000,0.000,0.000|Bf:35,1023|FS:0,0|Pn:PYZA>
and the Signal "LED" in ioSender will go off.  

However, with $5=15, i.e. NO sensors and no limit pins shorted to ground (i.e. not triggered), I get:
> <Idle|MPos:0.000,0.000,0.000,0.000|Bf:35,1023|FS:0,0|Pn:PXYZA>
Thus saying that all the limit input are triggered.  With hard limits enabled the machine will run programs just fine. If I short a limit pin to ground there is no change in the idle message but an Alarm 8 is thrown. So, even though grblHAL does not report a change in the limit sensors, it does know to throw an alarm 8.  This causes the problem for homing since it thinks the limit switch is already triggered?

The correct behavior should have the idle message show  Pn:P (I have not dealt with the Probe yet) when no sensors are triggered.  When a sensor is triggered the Pn: message should show the triggered Axis.

Here is the initial console info for the Pro board.  For the original board, everything is the same except for BOARD:

> [VER:1.1f.20210629:]
> [OPT:VNMSL,35,1024,4,0]
> [NEWOPT:ENUMS,RT+,ES,TC,ETH]
> [FIRMWARE:grblHAL]
> [NVS STORAGE:*FLASH]
> [DRIVER:iMXRT1062]
> [DRIVER VERSION:210703]
> [DRIVER OPTIONS:USB.2]
> [BOARD:T41BB5X Pro]
> [AUX IO:4,3,0,0]
> [IP:0.0.0.0]
screen shot of Settings: Grbl
[image]

terjeio, 2021-07-25:
This is due to a regression, change this line
https://github.com/grblHAL/iMXRT1062/blob/ac334100bf374a7211dd287d8274e3dbea744602/grblHAL_Teensy4/src/driver.c#L996
to 
`    signals.min.mask = settings.limits.invert.mask;`
to fix it.

I am currenly busy (have visitors) so a fix from me will be committed later.

phil-barrett, 2021-07-25:
No problem.  Thanks for the quick feedback.  Enjoy your guests!

phil-barrett, 2021-07-26:
I tested the change and it does stop the Alarm 8 when homing.  However, the NO sensors are still showing as triggered (in ioSender and the idle messages) and homing now immediately throws an Alarm 1.

terjeio, 2021-07-26:
Weird - it works for me...

Hopefully I will be able to commit the latest version later today.

closed 2023-01-29 by terjeio