grblHAL/iMXRT1062 #12: Teensy40 Unable to connect

Bx3mE, 2021-08-02:
changing in my_machine.h :
#define BOARD_T40X101
//#define BOARD_T41U5XBB

and leaving the following commented out:
//#define USB_SERIAL_CDC      2
OR setting it to 
#define USB_SERIAL_CDC      2

Compiling with PlatformIO Uploading with Teensy Loader 1.54

Used both FTDI-Driver and built in Microsift Serial Driver on Win10 to test, both configured to 115200 8N1.
Used UGS and PUTTY to test connection.

terjeio, 2021-08-02:
When compiling with Arduino both methods works for me. Something amiss in [platformio.ini](https://github.com/grblHAL/iMXRT1062/blob/master/grblHAL_Teensy4/platformio.ini)? `default_envs` is set for teensy41, have you changed that?

"Unable to connect" is a bit vague - do you mean unable to communicate? If unable to connect then you may have driver issues, are the ports showing up in the device manager? If so, is connection is refused?

I do not have PlatformIO installed and are not familiar with the ini-file structure so I am afraid I am not of much help regarding that.

Bx3mE, 2021-08-04:
Connected but no data coming on the stream. I tried with putty too and no
response to any input. (Tested "$$", "G0 X3", $I" etc.) My board is a blanc
Teensy 4.0 with no pins connected to anything. I´d suspect the board to
return an error if in E-Stop or similar state but nothing. After grounding
all pins, i actually made it respond but my goal was to achieve steprate of
between 600k steps/s and 4M steps/s to driva a galvo but it seems i will
not get close to that. I maxed out at 200... I will abandon this for the
time beeing...

terjeio, 2021-08-04:
>  I tried with putty too and no response to any input.

If estop is asserted (default if all pins left open) only a real-time request, `?`, is responded to.

> my goal was to achieve steprate of between 600k steps/s and 4M steps/s to driva a galvo

IMO not possible with a microcontroller. IIRC my FPGA based Smoothstepper (for Mach3) tops out at 1M steps/s.
If the g-code contains a lot of small movements max step rate is limited by the planner buffer size, larger  -> faster up to a point. 300KHz is possible, and one user reported a 600KHz step rate after modifying limits (min step pulse with is currently 2.5us) - but that was from a move from a single g-code command.

You have step/dir driven galvos? I made an experiment earlier with a 16 bit I2S D/A converter for analog output instead of step/dir signals - but I did not tune anything to find the max frequency possible. This was with a STM32F446 dev. board. Here is a XY plot on my scope I made then:

[image]

Bx3mE, 2021-08-06:
I made my own Galvo controller that takes step/dir and outputs XY2-100. The protocol is 2MHz and sentences are 20bit so 100000 positions per second, but the galvo is much faster with a setting time of 0.4ms. I am currently giving it a try to ignore any planning and just calculate the number of positions I can manage to send in the timeframe of the upcoming move and then divide the travel of the command move with that number of positions and output as many positions along that path as I can manage. so far it looks good as the setting time outperforms any need for correction of mechanical issues normally present as jitter jerk etc.

Bx3mE, 2021-08-06:
Would you have any interest in helping out in developing this?

terjeio, 2021-08-07:
> The protocol is 2MHz and sentences are 20bit so 100000 positions per second

Sounds like a job for the Pi Pico to generate the XY2-100 protocol stream - via a PIO program. We've made a driver for grblHAL for it that outputs step/dir signals via [PIO code](https://github.com/grblHAL/RP2040/blob/master/driverPIO.pio) and shift registers with sub microsecond resolution of the step signals...

> Would you have any interest in helping out in developing this?

Interesting, but I have far too much on may hands to get involved.

Bx3mE, 2021-08-07:
ok Tnks, I got my head straight yesterday and got on track to make a implementation. As I dont think it needs the same kind of complexity as a 3d printer firmware maybe I can get away with a minimalistic implementation. I will push it to OpenGalvo when done...

Bx3mE, 2021-08-17:
So I came around to making a first pre alpha version of my implementation: https://github.com/opengalvo/OPAL