grblHAL/iMXRT1062 #14: File not found : altpc.h

Tiktiki, 2021-09-10:
Error on compiling using VSCode and PlatformIO
Cannot find file altpc.h

src\networking\httpd.c:104:24: fatal error: lwip/altcp.h: No such file or directory

Careful exploration of grblHAL folders and .pio/libdeps confirms the file does not exist.

Tiktiki, 2021-09-10:
Linked to commit 39aa72ff0e74879dfeb7ab51a535e0fcf06e39dc 10 hours ago...

terjeio, 2021-09-10:
Bummer - code seemingly last updated in 2003 has dependencies on code from 2017. How could that be...

Fix [committed](https://github.com/grblHAL/Plugin_networking/commit/80c3f4d2238b3be801b6909edf0fdca9d32dcf24). Will add instructions on how to update lwIP when WebUI support for this driver is added.

Tiktiki, 2021-09-10:
Seems like a normal git clone off the https://git.savannah.nongnu.org/git/lwip.git repo (master) includes the files.
I wonder if the repo that is currently pointed to has been modified (deleted files) somehow... even if it does not show in history.

terjeio, 2021-09-15:
> Seems like a normal git clone off the https://git.savannah.nongnu.org/git/lwip.git repo (master) includes the files.

I have tried that - it did not work "out-of-the-box" so I ditched that attempt. Copying altpc.h to the original library, adjusting some code, and modifying lwipopts.h did - and [WebUI](https://github.com/luc-github/ESP3D-webui) can now be fired up:

[image]

Is this of interest? It is early days so more testing and tuning needs to be done...

Tiktiki, 2021-09-15:
That would be a great addition. Is it limited to ESP32 family or does it run on any network-enabled controller (running grblHAL)

terjeio, 2021-09-15:
I have the new code running on a STM32F756 and the iMXRT1062 (Teensy4.1), I have not yet tested with the other non ESP32 network capable controllers. The ESP32 driver uses the ESP32 http daemon - the new code a daemon based on the lwIP supllied code.

karoria, 2021-09-27:
Hi @terjeio 
I am unable to compile without a single change in source. See the log:

#################################################################
> Executing task in folder grblHAL_Teensy4: platformio run <

Processing teensy41 (board: teensy41; platform: teensy; framework: arduino)
------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/teensy/teensy41.html
PLATFORM: Teensy (4.13.1) > Teensy 4.1
HARDWARE: IMXRT1062 600MHz, 512KB RAM, 7.75MB Flash
DEBUG: Current (jlink) External (jlink)
PACKAGES: 
 - framework-arduinoteensy 1.154.0 (1.54) 
 - tool-teensy 1.154.210805 (1.54) 
 - toolchain-gccarmnoneeabi 1.50401.190816 (5.4.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 95 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <uSDFS> 1.1.2+sha.52b0ae3
|   |-- <MSC-non-blocking> 1.0.0+sha.5a2e6f4
|   |   |-- <USBHost_t36> 0.1
|-- <MSC-non-blocking> 1.0.0+sha.5a2e6f4
|   |-- <USBHost_t36> 0.1
|-- <lwip> 2.0.2+sha.58c266f
Building in release mode
Compiling .pio/build/teensy41/src/bluetooth/hc_05.c.o
Compiling .pio/build/teensy41/src/driver.c.o
Compiling .pio/build/teensy41/src/eeprom/eeprom_24AAxxx.c.o
Compiling .pio/build/teensy41/src/eeprom/eeprom_24LC16B.c.o
In file included from src/driver.c:87:0:
src/grbl/motor_pins.h:692:13: warning: 'motor_iterator' defined but not used [-Wunused-function]
 static void motor_iterator (motor_iterator_callback_ptr callback)
             ^
Compiling .pio/build/teensy41/src/encoder/encoder.c.o
Compiling .pio/build/teensy41/src/enet.c.o
Compiling .pio/build/teensy41/src/fans/fans.c.o
In file included from src/networking/networking.h:82:0,
                 from src/enet.c:39:
src/networking/httpd.h:49:24: fatal error: lwip/altcp.h: No such file or directory
compilation terminated.
*** [.pio/build/teensy41/src/enet.c.o] Error 1
=============================== [FAILED] Took 3.77 seconds ===============================

Environment    Status    Duration
-------------  --------  ------------
teensy41       FAILED    00:00:03.772
========================= 1 failed, 0 succeeded in 00:00:03.772 =========================
The terminal process "platformio 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
##################################################

Any clues to find lwip/altcp.h ? Further, it doesn't seem to be a directory named lwip anywhere in source tree. Please help.

Regards,
Ravi

terjeio, 2021-09-27:
@karoria I see now that when networking and SD card is enabled in my_machine.h the http daemon get enabled as well - uncommenting this line helps?

https://github.com/grblHAL/iMXRT1062/blob/7005e09e83f3c0e2f8b99c38a606c33c8baa0e0e/grblHAL_Teensy4/src/my_machine.h#L94

With networking enabled you have to install the lwip library, and to enable the httpd dameon you have to patch it, see this [link](https://github.com/grblHAL/iMXRT1062#networking-plugin) for how to install it for the Arduino IDE - the code has to be copied inside the grblHAL source to make it compile with PIO?

The patch for the lwip library (adds  lwip/altcp.h and modifies some configurations) - info [here](https://github.com/grblHAL/Plugin_WebUI/tree/3bc2b569057495f66e891c88bd073bc71ace8b83/lwIP%20patch).

I'll disable the http daemon in the next commit (unless the webui plugin is enabled)).

Tiktiki, 2021-09-27:
> Hi @terjeio
> I am unable to compile without a single change in source. See the log:
> 
> #################################################################
> 
> > Executing task in folder grblHAL_Teensy4: platformio run <
> 
> ## Processing teensy41 (board: teensy41; platform: teensy; framework: arduino)
> Verbose mode can be enabled via `-v, --verbose` option
> CONFIGURATION: https://docs.platformio.org/page/boards/teensy/teensy41.html
> PLATFORM: Teensy (4.13.1) > Teensy 4.1
> HARDWARE: IMXRT1062 600MHz, 512KB RAM, 7.75MB Flash
> DEBUG: Current (jlink) External (jlink)
> PACKAGES:
> 
> * framework-arduinoteensy 1.154.0 (1.54)
> * tool-teensy 1.154.210805 (1.54)
> * toolchain-gccarmnoneeabi 1.50401.190816 (5.4.1)
>   LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
>   LDF Modes: Finder ~ chain, Compatibility ~ soft
>   Found 95 compatible libraries
>   Scanning dependencies...
>   Dependency Graph
>   |--  1.1.2+sha.52b0ae3
>   |   |--  1.0.0+sha.5a2e6f4
>   |   |   |-- <USBHost_t36> 0.1
>   |--  1.0.0+sha.5a2e6f4
>   |   |-- <USBHost_t36> 0.1
>   |--  2.0.2+sha.58c266f
>   Building in release mode
>   Compiling .pio/build/teensy41/src/bluetooth/hc_05.c.o
>   Compiling .pio/build/teensy41/src/driver.c.o
>   Compiling .pio/build/teensy41/src/eeprom/eeprom_24AAxxx.c.o
>   Compiling .pio/build/teensy41/src/eeprom/eeprom_24LC16B.c.o
>   In file included from src/driver.c:87:0:
>   src/grbl/motor_pins.h:692:13: warning: 'motor_iterator' defined but not used [-Wunused-function]
>   static void motor_iterator (motor_iterator_callback_ptr callback)
>   ^
>   Compiling .pio/build/teensy41/src/encoder/encoder.c.o
>   Compiling .pio/build/teensy41/src/enet.c.o
>   Compiling .pio/build/teensy41/src/fans/fans.c.o
>   In file included from src/networking/networking.h:82:0,
>   from src/enet.c:39:
>   src/networking/httpd.h:49:24: fatal error: lwip/altcp.h: No such file or directory
>   compilation terminated.
>   *** [.pio/build/teensy41/src/enet.c.o] Error 1
>   =============================== [FAILED] Took 3.77 seconds ===============================
> 
> Environment Status Duration
> 
> teensy41 FAILED 00:00:03.772
> ========================= 1 failed, 0 succeeded in 00:00:03.772 =========================
> The terminal process "platformio 'run'" terminated with exit code: 1.
> 
> Terminal will be reused by tasks, press any key to close it.
> ##################################################
> 
> Any clues to find lwip/altcp.h ? Further, it doesn't seem to be a directory named lwip anywhere in source tree. Please help.
> 
> Regards,
> Ravi

Also, there is a problem with the version 4.13 of the Platform module. Force it to use version 4.12 or you will have errors anyways.

karoria, 2021-09-28:
> @karoria I see now that when networking and SD card is enabled in my_machine.h the http daemon get enabled as well - uncommenting this line helps? [...]

No success till now.
1. Commenting / Uncommenting "#define HTTP_ENABLE 1" could not help.
2. I am not sure about path of lwip library, I use github desktop on mac and open project with VS code with PIO, I just clone iMXRT1062 project and everything else is imported automatically to the project (including core, which settles inside grbl folder).
3. I doubt whether "iMXRT1064/grblHAL_Teensy4/src/networking" is to be replaced with "https://github.com/ddrown/teensy41_ethernet.git" as it seems both are for network plugins. If they are not same/interchangeable, where should be path of the second repo?
4. @Tiktiki Kindly guide me how to install version 4.12 and force the build process to use the same.
Waiting for reply.
Thanks,
Ravi

Tiktiki, 2021-09-28:
Here is the issue and fix for the platform version:

https://github.com/grblHAL/core/issues/47

terjeio, 2021-09-28:
I do not have PIO installed so I do not know where to copy the lwIP library or how to set it up for compilation. In the Arduino IDE it is imported via a menu selection and lives outside the grblHAL source code, can PIO do it in a similar way?

Here is my Arduino library folder structure with the lwIP library marked:

[image]

karoria, 2021-09-28:
@Tiktiki, installed 4.12 but no luck. Last lines of terminal are as below:

src/networking/httpd.c:255:2: error: #error HTTP v0.9 support has been removed!
 #error HTTP v0.9 support has been removed!
  ^
src/networking/httpd.c: In function 'get_http_content_length':
src/networking/httpd.c:923:58: error: 'cont_len' undeclared (first use in this function)
         hs->response_hdr.string[hs->response_hdr.next] = cont_len;
                                                          ^
src/networking/httpd.c:923:58: note: each undeclared identifier is reported only once for each function it appears in
src/networking/httpd.c: At top level:
src/networking/httpd.c:340:20: warning: 'conn_close' defined but not used [-Wunused-variable]
 static const char *conn_close = "Connection: Close" CRLF CRLF;
                    ^
src/networking/httpd.c:342:20: warning: 'conn_keep2' defined but not used [-Wunused-variable]
 static const char *conn_keep2 = "Connection: keep-alive" CRLF "Content-Length: ";
                    ^
*** [.pio/build/teensy41/src/networking/httpd.c.o] Error 1
===================== [FAILED] Took 240.39 seconds =====================

Environment    Status    Duration
-------------  --------  ------------
teensy41       FAILED    00:04:00.394
================= 1 failed, 0 succeeded in 00:04:00.394 =================
The terminal process "platformio 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

terjeio, 2021-09-28:
> src/networking/httpd.c:255:2: error: #error HTTP v0.9 support has been removed!

Getting closer now, have you replaced _lwipopts.h_ in the lwIP library folder with the one from the [patch](https://github.com/grblHAL/Plugin_WebUI/tree/3bc2b569057495f66e891c88bd073bc71ace8b83/lwIP%20patch)? Or are PIO messing up?

karoria, 2021-09-28:
> I do not have PIO installed so I do not know where to copy the lwIP library or how to set it up for compilation. In the Arduino IDE it is imported via a menu selection and lives outside the grblHAL [...]

Getting really closer:

> Executing task in folder grblHAL_Teensy4: platformio run <

Processing teensy41 (board: teensy41; platform: teensy@4.12; framework: arduino)
------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/teensy/teensy41.html
PLATFORM: Teensy (4.12.0) > Teensy 4.1
HARDWARE: IMXRT1062 600MHz, 512KB RAM, 7.75MB Flash
DEBUG: Current (jlink) External (jlink)
PACKAGES: 
 - framework-arduinoteensy 1.153.0 (1.53) 
 - toolchain-gccarmnoneeabi 1.50401.190816 (5.4.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 93 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <uSDFS> 1.1.2+sha.52b0ae3
|   |-- <MSC-non-blocking> 1.0.0+sha.5a2e6f4
|   |   |-- <USBHost_t36> 0.1
|-- <MSC-non-blocking> 1.0.0+sha.5a2e6f4
|   |-- <USBHost_t36> 0.1
|-- <lwip> 2.0.2+sha.58c266f
|-- <SD> 1.2.2
|   |-- <SPI> 1.0
Building in release mode
Compiling .pio/build/teensy41/src/enet.c.o
Compiling .pio/build/teensy41/src/networking/TCPStream.c.o
Compiling .pio/build/teensy41/src/networking/WsStream.c.o
Compiling .pio/build/teensy41/src/networking/fs.c.o
Compiling .pio/build/teensy41/src/networking/ftpd.c.o
Compiling .pio/build/teensy41/src/networking/http_upload.c.o
Compiling .pio/build/teensy41/src/networking/httpd.c.o
Compiling .pio/build/teensy41/src/networking/libraries/lwip-ftpd-sd/src/lwip_ftpd-sd.cpp.o
Compiling .pio/build/teensy41/src/networking/libraries/lwip-ftpd-sd/src/vfs.cpp.o
src/networking/libraries/lwip-ftpd-sd/src/lwip_ftpd-sd.cpp: In function 'void send_next_directory(ftpd_datastate*, tcp_pcb*, int)':
src/networking/libraries/lwip-ftpd-sd/src/lwip_ftpd-sd.cpp:461:51: warning: taking address of temporary [-fpermissive]
   fsd->vfs_dirent = &(fsd->vfs_dir->openNextFile());
                                                   ^
src/networking/libraries/lwip-ftpd-sd/src/lwip_ftpd-sd.cpp:465:94: warning: format '%d' expects argument of type 'int', but argument 4 has type 'uint32_t {aka long unsigned int}' [-Wformat=]
     else len = sprintf(buffer, "%s  %d\r\n", fsd->vfs_dirent->name(), fsd->vfs_dirent->size());
                                                                                              ^
src/networking/libraries/lwip-ftpd-sd/src/lwip_ftpd-sd.cpp:465:94: warning: format '%d' expects argument of type 'int', but argument 4 has type 'uint32_t {aka long unsigned int}' [-Wformat=]
Compiling .pio/build/teensy41/src/networking/multipartparser.c.o
Compiling .pio/build/teensy41/src/networking/networking.c.o
Compiling .pio/build/teensy41/src/networking/sfifo.c.o
Compiling .pio/build/teensy41/src/networking/sha1.c.o
src/networking/libraries/lwip-ftpd-sd/src/lwip_ftpd-sd.cpp: At global scope:
src/networking/libraries/lwip-ftpd-sd/src/lwip_ftpd-sd.cpp:169:20: warning: 'month_table' defined but not used [-Wunused-variable]
 static const char *month_table[12] = {
                    ^
Compiling .pio/build/teensy41/src/networking/src/api/api_lib.c.o
Compiling .pio/build/teensy41/src/networking/src/api/api_msg.c.o
Compiling .pio/build/teensy41/src/networking/src/api/err.c.o
Compiling .pio/build/teensy41/src/networking/src/api/netbuf.c.o
Compiling .pio/build/teensy41/src/networking/src/api/netdb.c.o
Compiling .pio/build/teensy41/src/networking/src/api/netifapi.c.o
Compiling .pio/build/teensy41/src/networking/src/api/sockets.c.o
Compiling .pio/build/teensy41/src/networking/src/api/tcpip.c.o
Compiling .pio/build/teensy41/src/networking/src/apps/httpd/fs.c.o
Compiling .pio/build/teensy41/src/networking/src/apps/httpd/httpd.c.o
Compiling .pio/build/teensy41/src/networking/src/apps/lwiperf/lwiperf.c.o
Compiling .pio/build/teensy41/src/networking/src/apps/mdns/mdns.c.o
Compiling .pio/build/teensy41/src/networking/src/apps/mqtt/mqtt.c.o
Compiling .pio/build/teensy41/src/networking/src/apps/netbiosns/netbiosns.c.o
Compiling .pio/build/teensy41/src/networking/src/apps/snmp/snmp_asn1.c.o
[... 367 lines cut ...]
.pio/build/teensy41/src/networking/src/apps/httpd/fs.c.o: In function `fs_open':
/Users/karoria/Documents/GitHub/iMXRT1062/grblHAL_Teensy4/src/networking/src/apps/httpd/fs.c:66: multiple definition of `fs_close'
.pio/build/teensy41/src/networking/fs.c.o:/Users/karoria/Documents/GitHub/iMXRT1062/grblHAL_Teensy4/src/networking/fs.c:56: first defined here
.pio/build/teensy41/src/networking/src/apps/httpd/fs.c.o: In function `fs_open':
/Users/karoria/Documents/GitHub/iMXRT1062/grblHAL_Teensy4/src/networking/src/apps/httpd/fs.c:66: multiple definition of `fs_read'
.pio/build/teensy41/src/networking/fs.c.o:/Users/karoria/Documents/GitHub/iMXRT1062/grblHAL_Teensy4/src/networking/fs.c:56: first defined here
.pio/build/teensy41/src/networking/src/apps/httpd/fs.c.o: In function `fs_open':
/Users/karoria/Documents/GitHub/iMXRT1062/grblHAL_Teensy4/src/networking/src/apps/httpd/fs.c:66: multiple definition of `fs_bytes_left'
.pio/build/teensy41/src/networking/fs.c.o:/Users/karoria/Documents/GitHub/iMXRT1062/grblHAL_Teensy4/src/networking/fs.c:56: first defined here
.pio/build/teensy41/src/networking/src/apps/httpd/httpd.c.o: In function `httpd_init':
[... cut ...]
*** [.pio/build/teensy41/firmware.elf] Error 1
============================== [FAILED] Took 56.49 seconds ==============================

Environment    Status    Duration
-------------  --------  ------------
teensy41       FAILED    00:00:56.492
========================= 1 failed, 0 succeeded in 00:00:56.492 =========================
The terminal process "platformio 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

@terjeio, I am also OK with Arduino IDE if it is little more easier.

karoria, 2021-09-28:
For everybody's information, "Teensy41_ethernet-master" directory goes to "iMXRT1062/grblHAL_Teensy4/.pio/libdeps/teensy41/" path. I am using PIO with VS code in Mac.

terjeio, 2021-09-28:
"multiple definitions" errors is due to similar source beeing compiled? This is coming from where? PIO or conflicting Arduino libs?

_Compiling .pio/build/teensy41/src/networking/libraries/lwip-ftpd-sd/src/lwip_ftpd-sd.cpp.o_

karoria, 2021-09-28:
I am not sure. I have already messed up a lot of code. Will start clean, may be tomorrow, as I am leaving from work now. Or will try later tonight. Will update here.

Tiktiki, 2021-09-28:
What compiler are you using?
I had troubles with Atom since it would refuse to load some libraries.

If you want the compiler to include a library you downloaded and placed somewhere, you have to go and modify the platformio.ini file, there is a section where it lists the includes. Change from a git command to a local directory and it will use that as include source.

And yeah, at that point I would go with a clean start/git pull.

karoria, 2021-09-29:
@Tiktiki 
Where to find for compiler in PIO? As I mentioned earlier, I copied libraries in ".pio/libdeps" manually. I have not changed anything in platformio.ini

closed 2023-08-26 by terjeio