grblHAL/iMXRT1062 #15: error: 'msSCSICapacity_t' does not name a type   msSCSICapacity_t msCapacity;

Tiktiki, 2021-09-10:
Error on compile with fresh clone.

Windows 10 Pro 20H2 build 19042.1165
VSCode 1.60.0
PlatformIO IDE v2.3.3
Teensy 4 v4.12.0
Python v3.9.7:1016ef3

STR:
git clone --recurse-submodules https://github.com/grblHAL/iMXRT1062.git
In PlatformIO terminal>pio run

`Compiling .pio\build\teensy41\lib7a0\uSDFS\ff.c.o
In file included from .pio\libdeps\teensy41\MSC-non-blocking\src\msc.h:31:0,
                 from .pio\libdeps\teensy41\MSC-non-blocking\src\MassStorageDriver.cpp:29:
C:\Users\gawyn\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2045:2: error: 'msSCSICapacity_t' does not name a type
  msSCSICapacity_t msCapacity;
  ^
C:\Users\gawyn\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2046:2: error: 'msInquiryResponse_t' does not name a type
  msInquiryResponse_t msInquiry;
  ^
C:\Users\gawyn\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2047:2: error: 'msRequestSenseResponse_t' does not name a type
  msRequestSenseResponse_t msSense;
  ^
C:\Users\gawyn\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2048:2: error: 'msDriveInfo_t' does not name a type
  msDriveInfo_t msDriveInfo;
  ^
C:\Users\gawyn\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2067:31: error: 'msSCSICapacity_t' has not been declared
  uint8_t msReadDeviceCapacity(msSCSICapacity_t * const Capacity);
                               ^
compilation terminated due to -fmax-errors=5.
In file included from .pio\libdeps\teensy41\MSC-non-blocking\src\msc.h:31:0,
                 from .pio\libdeps\teensy41\MSC-non-blocking\src\MassStorageHost.cpp:31:
C:\Users\gawyn\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2045:2: error: 'msSCSICapacity_t' does not name a type
  msSCSICapacity_t msCapacity;
  ^
C:\Users\gawyn\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2046:2: error: 'msInquiryResponse_t' does not name a type
  msInquiryResponse_t msInquiry;
  ^
C:\Users\gawyn\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2047:2: error: 'msRequestSenseResponse_t' does not name a type
  msRequestSenseResponse_t msSense;
  ^
C:\Users\gawyn\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2048:2: error: 'msDriveInfo_t' does not name a type
  msDriveInfo_t msDriveInfo;
  ^
C:\Users\gawyn\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2067:31: error: 'msSCSICapacity_t' has not been declared
  uint8_t msReadDeviceCapacity(msSCSICapacity_t * const Capacity);
                               ^
compilation terminated due to -fmax-errors=5.
*** [.pio\build\teensy41\libac5\MSC-non-blocking\MassStorageDriver.cpp.o] Error 1
*** [.pio\build\teensy41\libac5\MSC-non-blocking\MassStorageHost.cpp.o] Error 1
========================================================================================= [FAILED] Took 16.48 seconds =========================================================================================

Environment    Status    Duration
-------------  --------  ------------
teensy41       FAILED    00:00:16.481
==================================================================================== 1 failed, 0 succeeded in 00:00:16.481 ====================================================================================`

cmidgley, 2021-09-11:
Hi - I also just downloaded and hit that problem.  I found a discussion over on `core` that said it's an issue with Teensy libraries - you need to back out to version 4.12.  I did that, and it compiles now.  See [this issue](https://github.com/grblHAL/core/issues/47) for details.

The instructions in that issue change the library from 4.13 to 4.12 globally, but the steps aren't totally clear (but can be figured out).  However, I found a much easier way that works for me.   In the `platform.io` file (in your `grblHAL_Teensy4` folder), locate the `[env:teensy41]` section (for Teensy 4.1, if using 4.0 use the other section) and then change this line:

```
platform = ${common_teensy.platform}
```

to

```
platform = ${common_teensy.platform}@4.12
```

Tiktiki, 2021-09-11:
Yup, the platform update broke something.

fix in platformio.ini solves the issue for now.

dresco, 2022-02-02:
platformio.ini is pinning the teensy platform to 4.12, which pulls in teensyduino 1.53. This version is now thought to have issues.

This was done to avoid build errors with the SD card library, which is currently pointing at;
```
https://github.com/wwatson4506/uSDFS#uSDFS-non-blocking
https://github.com/wwatson4506/MSC#MSC-non-blocking
```

If we change the SD card library to the source listed in this repositories README.md -  `https://github.com/WMXZ-EU/uSDFS` (and drop the MSC library) - then I think we can get rid of this 4.12/1.53 dependency.

We could then either pin the teensy platform to 4.13 (teensyduino 1.54), or let it use the latest - which is currently 4.14 (teensyduino 1.55). 

Thoughts? I can make a PR for the ini file changes if in agreement..

terjeio, 2022-02-02:
> This version is now thought to have issues.

It has been confirmed 1.53 has issues, it caused random delays in grblHAL and who knows what else. 1.54 fixed the random delay issue.

> If we change the SD card library to the source listed in this repositories README.md - https://github.com/WMXZ-EU/uSDFS (and drop the MSC library) - then I think we can get rid of this 4.12/1.53 dependency.

The problem with this is that this library is buggy. I submitted a PR with a fix that was rejected, and the maintainer then tried to fix it himself - and that did not work. I guess the only way forward is to include the libary in the download.

BTW if you get around to write a driver for the STM32H7 then the bug is related to syncing a cache version of data with the RAM version.

So wait a bit with this - until the USB issue is fixed.

dresco, 2022-02-04:
> BTW if you get around to write a driver for the STM32H7 then the bug is related to syncing a cache version of data with the RAM version.

Thanks, I've got it building for H7 now (based off your F7 driver, just needed a few register & peripheral changes). Now just have to wait for hardware to arrive for testing - may be some time!

andrewmarles, 2022-10-02:
Is this issue resolved?  Is using `https://github.com/WMXZ-EU/uSDFS` a viable solution?  Does this preclude SD card support?

terjeio, 2022-10-02:
> Is this issue resolved

Yes, 1.54+ fixes the random delay issue.

> Is using https://github.com/WMXZ-EU/uSDFS a viable solution?

Yes, my [patched version](https://github.com/grblHAL/iMXRT1062/tree/master/patches) works - but not the official one.

> Does this preclude SD card support?

No - recent builds even works with WebUI v3.

closed 2023-08-26 by terjeio