grblHAL/iMXRT1062 #19: Initial test of webui, ftp and SD

karoria, 2021-09-30:
Hi @terjeio ,
Just wanted to share my initial impressions after playing some time on webui, ftp and sd.
1. FTP works like a charm
2. Uploading/deleting files from sd via webui and ftp both works perfect.
3. Webui is not responding to any manual command entry in the text box area.
4. Cant run program from webui.
5. No state or coordinate updates in webui.

It seems, webui is still under initial development. I want to know how I can map html, css, js project to internal bindings. I mean, how to host a server, send commands to serial console using webpage, so that I csn design my own app.

terjeio, 2021-09-30:
> Webui is not responding to any manual command entry in the text box area.

Is your websocket port 81? And is the browser connecting to that? Here is what I get in the Firefox debug output for network traffic (opens with F12) on startup:

[image]

The 101 response is what you want to see when opening the websocket.

> When I reload webui in browser, it hangs and doesnt show up any data.

It reloads just fine for me, any clues from looking at the debug output?

> It seems, webui is still under initial development.

It is but should be usable, what is missing is mainly access control - I have not spent time looking into why that does not work. Authentication seems to work so it should not be too hard to fix.

>  I want to know how I can map html, css, js project to internal bindings. I mean, how to host a server, send commands to serial console using webpage, so that I csn design my own app.

Currently you have to study the [webui server code](https://github.com/grblHAL/Plugin_WebUI/blob/main/server.c) and perhaps also take a look at the webui traffic.
Basically you have to write handlers for the different uri's and register them with the http daemon. Based on the request url the daemon will call back the appropriate handler. Serving files is easy, dynamic content is a bit more involved as the daemon only works with files. So either you have to write the content to a file and return that or do as I have done - trapping the stream output and redirecting it to memory and then returning data from that, a virtual file. There are function calls for handling virtual files so not too hard.
The _webui_init()_ function registers the handlers and is called at the end of the _driver_init()_ function. Tip: if you name your init function _my_plugin_init()_ it will be called automagically - no need to change any code.

BTW I committed an update yesterday - this allows saving of WebUI preferences to the www folder.

karoria, 2021-10-01:
Sorry replying you late @terjeio. I could not check what you asked. My router had some issues. Then I tried to use a crossover ethernet cable for direct communication with teensy4.1 but no success. I use mac with usb to ethernet converter cable. Can that be a problem? Also, #define network_ipmode value 0 and 1 both tried. I am well conversant with LAN and ip address assignment. For dhcp mode I assume it accepts dhcp ip assignment rather being a dhcp host. Am I right?

terjeio, 2021-10-01:
>  I could not check what you asked.

Why not? Which browser are you using? I believe most has a dev mode nowadays, F12 opens dev panes/windows with Firefox and Edge - IIRC Chrome and likely others too. The websocket port is in the controller settings - `$307`. Just send `$307` to get the value back or `$307=81` to set it. Cycle power if changing it as it is only read on startup.

Since you had ftp and webui up and running your other network settings were correct, no need to change them.

> For dhcp mode I assume it accepts dhcp ip assignment rather being a dhcp host. Am I right?

Yes.

I wonder if PIO is using the latest Teensy library, 1.53 had problems with intermittent delays which were fixed in 1.54.
From issue #14 it seems this could be so from this line: `framework-arduinoteensy 1.153.0 (1.53)` in [this comment](https://github.com/grblHAL/iMXRT1062/issues/14#issuecomment-929187918). This could be a problem?

karoria, 2021-10-02:
Sorry if I could not explain you properly. I meant my network router is out of order, so cant connect my grblHAL to my mac. Anyway, today I will get new but still I would like to know direct connection with crossover cable and its settings.
I must say, not only you have built a wonderful project, you are a "superman" in support also. I can see the number of drivers developed and maintained by your "one man army". Many thanks for all this!

karoria, 2021-10-02:
Got my router up and running. Played further with ip and webui. Following was observed:
1. Websocket connection shows failed in chrome developer tools. I set ws port to 81. It shows: websocket connection to 'ws://192.168.0.80:81/' failed.
2. Network only works if network ipmode is set to 1 (dhcp) in my_machine.h. I couldn't get it work with static ip. I want to make direct communication with crossover cable. Any clues?
3.. Webui can display and upload files and can run a program without showing status and axis position.

terjeio, 2021-10-02:
1. What is your `$307` setting? It should be `81`. If it is is a firewall in your machine blocking traffic on port 81? If the traffic passes through your router could it be blocking it?
2. Static IP is working for me (I have not tested with a dedicated ethernet port but I assume that would work too). Note that after changing network settings a power cycle is required for the Teensy as there is no reset button. `<ESC>` followed by `<CTRL>`+`<T>` in a terminal window may be used as well as this will attempt a cold restart.
Is this via a secondary ethernet port? Netmasks, gateway and IP-adresses are all good? Ping is working?
3. This is to be expected when websocket communication is not up.

You can [test websocket communication](https://github.com/grblHAL/Plugin_networking#simple-websocket-test-app) directly with the [wschat.html](https://github.com/grblHAL/Plugin_networking/blob/master/wschat.html) file, edit the IP address and port number in line 103 before opening it.

karoria, 2021-10-02:
wschat.html connects and communicates perfectly. But webui developer tools shows connection failure!
Will dig more on this and will let you know.
Thanks.
Ravi

terjeio, 2021-10-02:
That is odd, can you connect with wschat.html in another tab/window when the webui is running?

karoria, 2021-10-02:
Yes. I have started webui in pc1 (windows) and wschat.html in pc2 (mac) at the same time. Got success in wschat. However, webui is observed not working in pc2 earlier. So, my doubts are less on firewall and more on webui code/library  or any setting thereof where ip is not updated inside any code due to broken links?

karoria, 2021-10-02:
I forgot to check with same pc, same browser in different tab. Will do it tomorrow as it is night here and I am back home.

terjeio, 2021-10-02:
What is the response to the websocket connect in the webui (looking at the dev output)?
Also, be aware that a websocket can only serve one connection at a time so start with the webui first.

karoria, 2021-10-02:
I am sure I have started webui first. It was not successful. Then I startes wschat, which was successful. Will give more info on dev tools information of webui page tomorrow.

karoria, 2021-10-03:
1. Static IP now works. My ignorance of $301 setting. I was changing #define NETWORK_IPMODE in my_machine.h and recompiled and flashed each time I tried. I forgot EEPROM doesn't get erased when flashing and it remembers old settings.
2.  I can confirm webui has issues with websocket which was started first on chrome browser. Then, in another tab started wschat which connects successfully. 
3. Following is the dev tools log:
(index):17 Connect to board
(index):17 Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/94.0.4606.61 Safari/537.36
(index):17 No IP for websocket, use default
(index):17 Socket is 192.168.1.77:81
(index):17 Fw identification:FW version:1.1f # FW target:grbl-embedded # FW HW:Direct SD # primary sd:/sd # secondary sd:none # authentication:no # webcommunication: Sync: 81# hostname:GRBL
(index):17 Init UI
(index):17 Get settings
(index):17 Get macros
(index):17 Get preferences
(index):17 WebSocket connection to 'ws://192.168.1.77:81/' failed: 
startSocket @ (index):17
(index):17 ws error Event
(index):17 Disconnected
/macrocfg.json?1633243403758&PAGEID=:1 Failed to load resource: the server responded with a status of 404 (File not found)
(index):17 Error 404 : <html><body><h2>404: The requested file cannot be found.</h2></body></html>

/preferences.json?1633243403760&PAGEID=:1 Failed to load resource: the server responded with a status of 404 (File not found)
(index):17 Error 404 : <html><body><h2>404: The requested file cannot be found.</h2></body></html>

(index):17 enable ping
(index):17 .g, .G, .gco, .GCO, .gcode, .GCODE, .nc, .NC, .ngc, .NCG, .tap, .TAP, .txt, .TXT
(index):17 Socket is 192.168.1.77:81
(index):17 WebSocket connection to 'ws://192.168.1.77:81/' failed: 
startSocket @ (index):17
(index):17 ws error Event
(index):17 Disconnected
(index):17 Socket is 192.168.1.77:81
(index):17 WebSocket connection to 'ws://192.168.1.77:81/' failed: 
startSocket @ (index):17
(index):17 ws error Event
(index):17 Disconnected
(index):17 Socket is 192.168.1.77:81
(index):17 WebSocket connection to 'ws://192.168.1.77:81/' failed: 
startSocket @ (index):17
(index):17 ws error Event
(index):17 Disconnected
(index):17 Socket is 192.168.1.77:81
(index):17 WebSocket connection to 'ws://192.168.1.77:81/' failed: 
startSocket @ (index):17
(index):17 ws error Event
(index):17 Disconnected
(index):17 Socket is 192.168.1.77:81
(index):17 WebSocket connection to 'ws://192.168.1.77:81/' failed: 
startSocket @ (index):17
(index):17 ws error Event
(index):17 Disconnected
(index):17 Socket is 192.168.1.77:81
(index):17 WebSocket connection to 'ws://192.168.1.77:81/' failed: 
startSocket @ (index):17
(index):17 ws error Event
(index):17 Disconnected
(index):17 Socket is 192.168.1.77:81
(index):17 WebSocket connection to 'ws://192.168.1.77:81/' failed: 
startSocket @ (index):17
(index):17 ws error Event
(index):17 Disconnected
(index):17 Socket is 192.168.1.77:81
(index):17 WebSocket connection to 'ws://192.168.1.77:81/' failed: 
startSocket @ (index):17
(index):17 ws error Event
(index):17 Disconnected
(index):17 Socket is 192.168.1.77:81
(index):17 WebSocket connection to 'ws://192.168.1.77:81/' failed: 
startSocket @ (index):17
(index):17 ws error Event {isTrusted: true, type: 'error', target: WebSocket, currentTarget: WebSocket, eventPhase: 2, …}
(index):17 Disconnected
(index):17 Socket is 192.168.1.77:81
(index):17 WebSocket connection to 'ws://192.168.1.77:81/' failed: 
startSocket @ (index):17
(index):17 ws error Event {isTrusted: true, type: 'error', target: WebSocket, currentTarget: WebSocket, eventPhase: 2, …}
(index):17 Disconnected

karoria, 2021-10-03:
Here are the lines from wschat (newest on top) if that helps:

<Idle|MPos:-0.800,28.124,15.000|Bf:35,1023|FS:0,0|Pn:PXYZHS>
<Idle|MPos:-0.800,28.124,15.000|Bf:35,1023|FS:0,0|Pn:PXYZHS|WCO:0.000,0.000,0.000|Ov:100,100,100>
<Idle|MPos:-0.800,28.124,15.000|Bf:35,1023|FS:0,0|Pn:PXYZHS>
<Idle|MPos:-0.800,28.124,15.000|Bf:35,1023|FS:0,0|Pn:PXYZHS>
[VER:1.1f.20210928:] [OPT:VNMSL,35,1024,3,0] [NEWOPT:ENUMS,RT+,ES,TC,SED,ETH,FTP,SD] [FIRMWARE:grblHAL] [NVS STORAGE:*FLASH] [DRIVER:iMXRT1062] [DRIVER VERSION:210919] [DRIVER OPTIONS:USB.2] [BOARD:T41U5XBB] [AUX IO:4,3,0,0] [NETCON:Websocket] [IP:192.168.1.77] [PLUGIN:KEYPAD v1.11] [PLUGIN:WebUI v0.01] [PLUGIN:SDCARD v1.02] ok
[MSG:WEBSOCKET STREAM ACTIVE]

karoria, 2021-10-03:
This is how my webui screen shows. Coordinates are not there. Status like "JOG", "RUN", "IDLE" are also absent
[image]
.

terjeio, 2021-10-03:
This is very odd - a Chrome issue? Have you tried with another browser?
And is it possible to see the network traffic in the dev output - is there any response header coming back? If so compare with wschat.html traffic. "failed" is not much to work from. I would like to know if the connection fails completely or if there is a status code returned. 

[This is the function](https://github.com/grblHAL/Plugin_networking/blob/0a564ec00b91ed9bf718457a5663bc54da257208/WsStream.c#L467) that is called on a websocket connect, and is where some kind of debug output could be added.
A simple one can be to switch on flood coolant:
```
coolant_state_t state = { .flood = 1 };
hal.coolant.set_state(state);
```

The missing data in the WebUI is simple to explain as these are extracted from the websocket stream - no connection, no data to show.

Yoy can ignore the 404 responses - until you save preferences in the WebUI these will show up. Note that saving these is only possible with the latest commit.

karoria, 2021-10-03:
OMG. Right. Chrome issue. But we can't really call it a browser issue. wschat runs good in chrome! I have checked with safari browser where webui works! but again it doesn't work with opera! I am not good at coding but I will try myself to findout issues. I will post here if I can get anything. I suspect issues with index.html.gz. Will dig more to it. Thanks.

terjeio, 2021-10-03:
Ok, there is a browser compatibility list [here](https://github.com/luc-github/ESP3D-WEBUI/wiki/Browser-compatibility-for-ESP3D).

_index.html.gz_ is a compressed version of the minimized javascript code. This means that the [current source code](https://github.com/luc-github/ESP3D-WEBUI) is what you will have to dig into.

Note that I have only changed some hardcoded strings to "grblHAL" in the _index.html.gz_ supplied, my plan is to try to add proper support for all grblHAL settings and I just wanted so check out how to build it.

karoria, 2021-10-03:
Noted. Thanks.

karoria, 2021-10-03:
I think, luc is primarily working on ESP3D for all ESP boards which has very little memory for web app storage so he might have selected gulp and other libraries based on that. If we fresh start with a complete web app, it should be even less time consuming than understanding and editing ESP3D code (as we have plenty of storage). I have beginner to intermediate level skill to develop nodejs web applications. The only thing I really don't know is binding of server side code to js. Is there any api readily available for me to act as template?

terjeio, 2021-10-03:
>  Is there any api readily available for me to act as template?

Websocket comms can be used "as-is" as it is just a serial stream wrapper for the standard grbl request/response protocol.
However, this protocol is perhaps not the easiest to handle in javascript as it requires custom parsing. A protocol based on JSON formatted data would be easier? [g2core](https://github.com/synthetos/g2) has a JSON based protocol that could be worth taking a look at?

By [adding a few function pointers](https://github.com/grblHAL/core/blob/71dc5ac3331e3d0d195486c0bfd8605ef3e3f1a7/core_handlers.h#L35-L50) to the core it will be possible to replace the standard grbl protocol with a new one, even at run-time. This could be part of the core or in a plugin.

Using http requests (rest API?) on top of a protocol could also be an option, the WebUI uses that but relies on a websocket connection to get back some of the responses - perhaps because that was the only workable solution?

There is one downside to http requests and repsonses - overhead, you cannot avoid that. I don't know if that could be an issue, especially for high speed laser engraving?

I am not entirely sure about what you define as the API as there are several levels an API might be defined, if seen from the browser then the [grblHAL WebUI](https://github.com/grblHAL/Plugin_WebUI/blob/main/server.c) API can serve as an example as it parses http requests and deliver repsonses back both via html and a websocket stream.

karoria, 2021-10-03:
OK. I was under impression that you have modified ESP3D code to suit with grblHAL but now I can understand server.c is the file you wrote as a "middleware". Correct me if I am wrong.
I will try to understand server.c (I am very poor in C and C++, so it is not my piece of cake) as well as ESP3D webui (this I can understand little more as I am well versed with html, css and js).
I will play with both for some days to get more insight. If I can get useful outcome, I will be happy to contribute to this wonderful project.

terjeio, 2021-10-03:
> Correct me if I am wrong.

You are entirely correct, the WebUI code is (not yet) modified by me. _server.c_ (and _commands.c_) is the backend/middleware for the WebUI protocol.

karoria, 2021-10-03:
Thanks for the info.

karoria, 2021-10-03:
Let me tell you, all the network based protocols works really fast. Previously I have used ESP3D webui with grbl_esp32 project but due to wifi, file upload speed was very slow. I always prefer FTP client (I use FileZilla) which is very convenient in uploading as well as deleting files and folders from SD and with native ethernet of teensy4.1, it is blazing fast. Upload speed in webui is also very fast due to ethernet. This is for information/feedback purpose.

mundsen, 2021-12-26:
I`m trying to activate the webui, but get this error

[image]

I have the grblHAL Teensy 4.1 Pro controller

#define USB_SERIAL_CDC      2 // 1 for Arduino class library and 2 for PJRC C library. Comment out to use UART communication.
//#define USB_SERIAL_WAIT     1 // Wait for USB connection before starting grblHAL.
//#define BLUETOOTH_ENABLE    1 // Set to 1 for HC-05 module. Requires and claims one auxillary input pin.
//#define HUANYANG_ENABLE     1 // Set to 1 or 2 for Huanyang VFD spindle. Enables spindle plugin.
//#define MODBUS_ENABLE       1 // Set to 1 for auto direction, 2 for direction signal on auxillary output pin.
#define WEBUI_ENABLE        1 // Enable ESP3D-WEBUI plugin along with networking and SD card plugins.
//#define WEBUI_INFLASH       1 // Store WebUI files in flash instead of on SD card.
#define ETHERNET_ENABLE     1 // Ethernet streaming. Enables networking plugin.
//#define SDCARD_ENABLE       1 // Run gcode programs from SD card, enables sdcard plugin.
//#define QEI_ENABLE          1 // Enable quadrature encoder interfaces. Max value is 1. Requires encoder plugin.
//#define KEYPAD_ENABLE       1 // Set to 1 for I2C keypad, 2 for other input such as serial data
//#define PLASMA_ENABLE       1 // Plasma/THC plugin. To be completed.
//#define MCP3221_ENABLE      1 // Enable analog input via MCP3221 ADC.
//#define PPI_ENABLE          1 // Laser PPI plugin. To be completed.
//#define ODOMETER_ENABLE     1 // Odometer plugin.
//#define OPENPNP_ENABLE      1 // OpenPNP plugin. To be completed.
//#define FANS_ENABLE         1 // Enable fan control via M106/M107. Enables fans plugin.
#define EEPROM_ENABLE       1 // I2C EEPROM support. Set to 1 for 24LC16 (2K), 3 for 24C32 (4K - 32 byte page) and 2 for other sizes. Enables eeprom plugin.
//#define EEPROM_IS_FRAM      1 // Uncomment when EEPROM is enabled and chip is FRAM, this to remove write delay.
//#define SPINDLE_SYNC_ENABLE 1 // Enable spindle sync support (G33, G76). !! NOTE: Alpha quality - enable only for test or verification.
                                // Currently only available for BOARD_T41BB5X_PRO and BOARD_T41U5XBB_SS.
//#define ESTOP_ENABLE        0 // When enabled only real-time report requests will be executed when the reset pin is asserted.
                                // Note: if left commented out the default setting is determined from COMPATIBILITY_LEVEL.

// If the selected board map supports more than three motors ganging and/or auto-squaring
// of axes can be enabled here.
//#define X_GANGED            1
//#define X_AUTO_SQUARE       1
//#define Y_GANGED            1
//#define Y_AUTO_SQUARE       1
//#define Z_GANGED            1
//#define Z_AUTO_SQUARE       1
// For ganged axes the limit switch input (if available) can be configured to act as a max travel limit switch.
// NOTE: If board map already has max limit inputs defined this configuration will be ignored.
//#define X_GANGED_LIM_MAX    1
//#define Y_GANGED_LIM_MAX    1
//#define Z_GANGED_LIM_MAX    1
//

#if ETHERNET_ENABLE > 0
#define TELNET_ENABLE           1 // Telnet daemon - requires Ethernet streaming enabled.
#define WEBSOCKET_ENABLE        1 // Websocket daemon - requires Ethernet streaming enabled.
#ifdef SDCARD_ENABLE
//#define FTP_ENABLE              1 // Ftp daemon - requires SD card enabled.
//#define HTTP_ENABLE             1 // http daemon - requires SD card enabled.
#endif
// The following symbols have the default values as shown, uncomment and change as needed.
#define NETWORK_HOSTNAME        "GRBL"
#define NETWORK_IPMODE          1 // 0 = static, 1 = DHCP, 2 = AutoIP
//#define NETWORK_IP              "192.168.5.1"
//#define NETWORK_GATEWAY         "192.168.5.1"
//#define NETWORK_MASK            "255.255.255.0"
//#define NETWORK_FTP_PORT        21
//#define NETWORK_TELNET_PORT     23
//#define NETWORK_HTTP_PORT       80
#if HTTP_ENABLE
//#define NETWORK_WEBSOCKET_PORT  81
#else
//#define NETWORK_WEBSOCKET_PORT  80
#endif
#endif

terjeio, 2021-12-27:
@mundsen Have you [patched lwIP](https://github.com/grblHAL/iMXRT1062#networking-plugin)?

mundsen, 2021-12-27:
yes, but I`ll try one more time, in case I have done something wrong

mundsen, 2021-12-30:
Started using a new download from git.. but still the same :-(

terjeio, 2021-12-30:
> Started using a new download from git.. but still the same :-(

Try deleting the _src/apps/httpd_ folder in the ethernet library, seems like I have done that and forgotten about it.

mundsen, 2021-12-30:
Eureka :-) Deleted the folder - now it compiles without errors

I`ll upload and test later

mundsen, 2021-12-31:
[image]

WebUi up and running :-)  x and y axis moves fine

mundsen, 2021-12-31:
Is it possible/hard to alter the text/icons for spindle, flood and mist?

Sample (for my laser setup) - Spindle = Laser, Flod = Exhaust fan and Mist = AirAssist ?

+ If I open LightBurn or ioSender I can move the axis..as normal, but if I open the web UI, I can no longer move the axis.. from ioSender/LB - only from the Web UI - until I close the Web UI window - is this normal? not possible using them at the same time?

terjeio, 2021-12-31:
> Is it possible/hard to alter the text/icons for spindle, flood and mist?

If you set up a [development environment](https://github.com/luc-github/ESP3D-WEBUI/wiki/Compilation---Development) for the WebUI I guess it would not be too hard. I did that to modify a few strings earlier and managed to recompile.

> If I open LightBurn or ioSender I can move the axis..as normal, but if I open the web UI, I can no longer move the axis.. from ioSender/LB - only from the Web UI - until I close the Web UI window - is this normal?

Yes.

> not possible using them at the same time?

No, but I may implement some kind of protocol later for switching between connected clients. This since I regard allowing clients to intermingle gcode commands as a safety hazard.
It can be noted that the current behaviour is not how I want it to be either, a network client will take over full control regardless of the current machine state. It should not be allowed to unless the machine is in a state where it is safe to do so.

closed 2023-08-26 by terjeio