grblHAL/iMXRT1062 #21: Teensy 4.1 hangs at full buffer and trigger of emergency stop

rcp1, 2021-10-13:
Hi,

thanks for your great work with grblHal.

I am encountering an issue regarding jogging. I am currently trying to implement my own gamepad driver for jogging in cncjs with grblHal on a T41U5XBB board and do send a lot of `$J=G91G21X-3.3Y0.0Z0.0F1000.00` jog commands to grblHal.  Moreover - due to the unfinished nature of my code or my laziness - I do not check for the Ack from grblHal. Thus, these jog commands stack up until plan and receive buffer are full.

If I trigger with full buffers the emergency stop pin the Teensy is becoming completely unresponsible, only a hard power reset helps.

It may be a stupid question, but is it expected that a full buffer leads to becoming unresponsive? Is it the task of the sender to stop sending in sight of a full buffer? grblHal is my first experience with grbl-based SW and I would have thought that the expected behavior is that grblHal under no circumstances becomes unresponsive.

I can recreate the crash / becoming unresponsive with IOSender, too, if I set the jog values to unreasonable amounts and jog like crazy with keyboard on all axes:

```
  <Jog>
    <KeyboardEnable>false</KeyboardEnable>
    <FastFeedrate>3000</FastFeedrate>
    <SlowFeedrate>3000</SlowFeedrate>
    <StepFeedrate>3000</StepFeedrate>
    <FastDistance>300</FastDistance>
    <SlowDistance>300</SlowDistance>
    <StepDistance>300</StepDistance>
  </Jog>
```

Here are my build settings ($I):
```
[VER:1.1f.20211010:]
[OPT:VNMZSL,35,1024,3,0]
[NEWOPT:ENUMS,RT+,HOME,ES,EXPR,TC,SED]
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[DRIVER:iMXRT1062]
[DRIVER VERSION:210930]
[DRIVER OPTIONS:USB.2]
[BOARD:T41U5XBB]
[AUX IO:4,3,0,0]
```

Thank you for your time.

rcp1, 2021-10-14:
Okay, so in general this is already an edge case since the sender / I does / do not care about the buffer.

And yes, I've just tested with USB_SERIAL_CDC as 1 and it happens, too.

```
[VER:1.1f.20211010:]
[OPT:VNMZSL,35,1024,3,0]
[NEWOPT:ENUMS,RT+,HOME,ES,EXPR,TC,SED]
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[DRIVER:iMXRT1062]
[DRIVER VERSION:210930]
[DRIVER OPTIONS:USB.1]
[BOARD:T41U5XBB]
[AUX IO:4,3,0,0]
```

dresco, 2021-10-14:
@rcp1 Out of interest, I was hacking around with a gamepad addon for CNCjs as well. Was working generally okay, but had a troublesome memory leak on the Pi that I never got to the bottom of..

I've just stuck my code up on github in case it's helpful to you - https://github.com/dresco/cncjs-pendant-gamecontroller

closed 2023-08-26 by terjeio