grblHAL/iMXRT1062 #25: REQ: Add serial driver information for Serial8 on Teensy 4.1 as used in grblHAL2000 controller board

5ocworkshop, 2021-10-31:
As this happened within another issue that is appropriately now closed, and it is specific to Teensy based boards, I thought I should tow this over here.  Serial 8 has been working well with these changes:
--

I am happy to report the dragon has been slayed:
```
Port /dev/ttyAMA0, 21:38:24
Press CTRL-A Z for help on special keys

GrblHAL 1.1f ['$' or '$HELP' for help]
```

Here is the entry I created for Serial8 on the Teensy 4.1.  I apologize for not submitting a patch but I think it best someone who really understands it and how to add it to the desired structure make the necessary changes so it can coexist with various board/pin combinations:

```
// JAC adjusted per https://github.com/PaulStoffregen/cores/blob/master/teensy4/HardwareSerial8.cpp
// Applies to grblHAL2000 board.  Serial8 is routed to the 40 pin Pi format expansion header on the board

static const uart_hardware_t uart5_hardware =
{
    .port = &IMXRT_LPUART5,
    .ccm_register = &CCM_CCGR3,
    .ccm_value = CCM_CCGR3_LPUART5(CCM_CCGR_ON),
    .irq = IRQ_LPUART5,
    .irq_handler = uart_interrupt_handler,
    .rx_pin = {
        // .pin = 21,
        .pin = 34, // JAC For connection to Pi Header
        .mux_val = 1,
        .select_reg = &IOMUXC_LPUART5_RX_SELECT_INPUT,
        .select_val = 1
    },
    .tx_pin = {
        // .pin = 20,
        .pin = 35, // JAC for connection to Pi Header
        .mux_val = 1,
        .select_reg = &IOMUXC_LPUART5_TX_SELECT_INPUT,
        .select_val = 1
    }
};
```

terjeio, 2021-10-31:
I have added a new definition for this already but not yet commited it. Will do so soon.

terjeio, 2021-11-01:
Update now commited, port is selected by
`#define UART_PORT 8`
in the map file or _my_machine.h_.

Should it be default for the GRBLHAL200 map?

5ocworkshop, 2021-11-07:
Great, thank you Terje.

Enabling it should be the default for the HAL2000 board, but it is not the primary interface, normally.
  That should still default to USB and be able to be switched to Ethernet.  It should be available as secondary serial.

terjeio, 2021-11-09:
> Enabling it should be the default for the HAL2000 board, but it is not the primary interface, normally.

Enabled in the latest commit.

closed 2021-11-09 by terjeio