grblHAL/iMXRT1062 #26: Swapping spindle PWM pin from 13 to 12 breaks comapilation

karoria, 2021-11-16:
Hi @terjeio 
As I have shown interest to use teensy4.1 onboard LED as status light with blink patterns, I considered to use GPIO 12 for spindle PWM. But I can't compile that way with following messages in PIO.

> Executing task in folder grblHAL_Teensy4: platformio run <

Processing teensy41 (board: teensy41; platform: teensy; framework: arduino)
----------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/teensy/teensy41.html
PLATFORM: Teensy (4.12.0) > Teensy 4.1
HARDWARE: IMXRT1062 600MHz, 512KB RAM, 7.75MB Flash
DEBUG: Current (jlink) External (jlink)
PACKAGES: 
 - framework-arduinoteensy 1.153.0 (1.53) 
 - toolchain-gccarmnoneeabi 1.50401.190816 (5.4.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 93 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <uSDFS> 1.1.2+sha.52b0ae3
|   |-- <MSC-non-blocking> 1.0.0+sha.5a2e6f4
|   |   |-- <USBHost_t36> 0.1
|-- <MSC-non-blocking> 1.0.0+sha.5a2e6f4
|   |-- <USBHost_t36> 0.1
|-- <lwip> 2.0.2+sha.58c266f
|-- <lwip> 2.0.2
Building in release mode
Compiling .pio/build/teensy41/src/driver.c.o
Compiling .pio/build/teensy41/src/grbl/alarms.c.o
Compiling .pio/build/teensy41/src/grbl/coolant_control.c.o
Compiling .pio/build/teensy41/src/grbl/corexy.c.o
src/driver.c: In function 'spindle_set_speed':
src/driver.c:1177:24: error: 'settings_t {aka struct <anonymous>}' has no member named 'sp'
             if(settings.sp)
                        ^
In file included from src/driver.c:87:0:
src/driver.c: At top level:
src/grbl/motor_pins.h:692:13: warning: 'motor_iterator' defined but not used [-Wunused-function]
 static void motor_iterator (motor_iterator_callback_ptr callback)
             ^
*** [.pio/build/teensy41/src/driver.c.o] Error 1
======================================= [FAILED] Took 6.62 seconds =======================================

Environment    Status    Duration
-------------  --------  ------------
teensy41       FAILED    00:00:06.622
================================= 1 failed, 0 succeeded in 00:00:06.622 =================================
The terminal process "platformio 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

terjeio, 2021-11-16:
Ouch - seems like a copy/paste error. Remove line 1177 and it should be ok.
I'll commit a fix later as I have some other pending changes.

Thanks for reporting.

karoria, 2021-11-16:
It works! Thanks for prompt reply as always.

closed 2021-12-28 by karoria