grblHAL/iMXRT1062 #27: unknown type name 'io_stream_properties_t

fireup924, 2021-12-02:
Get an error when trying to compile current version
"uart.c: 184:8: error: unknown type name 'io_stream_properties_t"

Can not seem to find where it should be defined?

terjeio, 2021-12-02:
It is defined [here](https://github.com/grblHAL/core/blob/637fa80eb328b042754716969674caff598425d9/stream.h#L234-L239). Did you [pull the submodules](https://github.com/grblHAL/core/wiki/Compiling-GrblHAL#using-git-command-line) when you downloaded/updated?

fireup924, 2021-12-03:
I was using Sourcetree to do a pull/update of current version. Somehow it did not update stream.h or stream.c. I had to delete my local copy and did a fresh clone to get all new files and it compiles now. Thanks.

closed 2021-12-03 by fireup924