grblHAL/iMXRT1062 #33: Z-probe support?

mundsen, 2022-01-20:
Hi, does grblHAL support different probes for auto set z-height (auto focus) ?

I have a laser with x,y and Z (+rotary)

I`m interested in a function to auto set focus height for the laser

I could mount a simple switch at the laser module that is triggered when it hits the material surface and then set the focus distance in software.

one alternative is to configure this switch as an z-end stop and home the z-axis when I add a new object to engrave..

Or is there probe support that can handle this?  like a button that triggers a macro that probes the surface and sets the focus?

terjeio, 2022-01-20:
> Hi, does grblHAL support different probes for auto set z-height (auto focus) ?

Not out of the box, the next commit will have a [new core event](http://svn.io-engineering.com/grblHAL/html/core__handlers_8h.html) for probe start so perhaps you can write a plugin that subscribe to that for switching probe input. There is already an event for probe completed.

> Or is there probe support that can handle this? like a button that triggers a macro that probes the surface and sets the focus?

That could be an option as well, the [macro plugin](https://github.com/grblHAL/Templates/tree/master/my_plugin) can be used?

mundsen, 2022-02-09:
Can a laser follow a z-axis path?

If using a rotating axis, the height (z-axis) of the object you want to engrave can vary

Is it possible in any way, using a z-probe to make a z-axis path that can be used to adjust the z-height while engraving?

Image for illustration only

[image]

terjeio, 2022-02-09:
> Can a laser follow a z-axis path?

Yes.

> Is it possible in any way, using a z-probe to make a z-axis path that can be used to adjust the z-height while engraving?

Yes. Many senders supports that - by creating a height map and transform the gcode.

lawrence-jeff, 2022-11-07:
@mundsen  - Did you ever work this out, I'm converting an old proprietary laser to GRBLHAL and it has a head mounted touch sensor that I'm trying to figure out how to incorporate

closed 2023-08-26 by terjeio