grblHAL/iMXRT1062 #34: Teensy 4.1 hangs on custom board

karoria, 2022-02-12:
Dear all, I need help as I am stuck with teensy4.1 hang issues. This may not be a grblHAL related issue though. I need some favour. My observations / descriptions are as under:
1. I am using my own webUI named grblTouch. I have connected RPi4 with 7 inch display to grblHAL teensy4.1 custom board via a network router with ethernet cables.
2. The setup runs smoothly when I use a particular brand SMPS (wago), but not with other different brands (phoenix, delta, rishabh) with same 24V 5A rating.
3. Now focussing one one non-working setup with rishabh SMPS, it works while I have not connected limit switches. This means it is obviously EMI issues. But it doesn't false trigger hard limit ever. That means is is 24v and 0v lines which enters with emi noise to pcb.
4. Upon attaching scope to 24v and 0v lines at different places in circuit, it shows steady 24v without spikes. After a 24v to 5v buck converter (common ground) in circuit, it shows spikes. 3.3v from the teensy board shows maximum spikes. It ranges from 4.8v to 1.0v. To my surprise, I checked with Wago SMPS again and it shows same level of spikes. But with that smps, the circuit performs well. 
I am now clueless. Can somebody point me to right direction about what to check?
Sorry for asking hardware (it seems like that) issue here.

karoria, 2022-02-12:
Sorry to bother you friends. Problem solved. Grounded the shields of shielded cables coming from limit switches.

terjeio, 2022-02-12:
Was this with a board having opto-couplers in front of the limit inputs?

karoria, 2022-02-12:
Yes. My limit switches are opto-isolated. I use inductive proximity sensors as limit switches which operates on 24V which is dropped by resistor on PCB to suit for the opto input.

closed 2022-09-27 by terjeio