grblHAL/iMXRT1062 #41: Bluetooth for Android sender

mac7988, 2022-06-07:
Can I hookup a hc-05 module to tx rx for serial communication?

Do I need to modify the code?

Thank you in advance

mac7988, 2022-07-15:
Hi,

I  hooked up my hc-05  module to tx1 rx1 (pin 0 and 1) but I have no serial connection. The module connects just no data. 

I am using Android grbl sender

terjeio, 2022-07-15:
Have you enabled the [Bluetooth plugin](https://github.com/grblHAL/Plugins_Bluetooth) in [my_machine.h](https://github.com/grblHAL/iMXRT1062/blob/5cf91f3bb9baa4def8f3da7356dacbd383a6e4a3/grblHAL_Teensy4/src/my_machine.h#L57)?
Did you connect the state signal?
Is the $377 setting value 1 (e.g. following a successful [auto configure](https://github.com/grblHAL/Plugins_Bluetooth#auto-configure))?

mac7988, 2022-07-15:
I used a precompiled version of your code.
$377 is not there when I serial send $+

terjeio, 2022-10-20:
> I used a precompiled version of your code.

I am working on a [web app](http://svn.io-engineering.com:8080/) for building binaries, you may try that to create a version with Bluetooth enabled.

terjeio, 2023-01-27:
See this [blog post](https://www.grbl.org/single-post/using-the-grblhal-web-builder) for how to use the Web Builder.

closed 2023-01-27 by terjeio