grblHAL/iMXRT1062 #42: USB host support

karoria, 2022-07-12:
Hi @terjeio 
I was wondering if I can activate native USB host of teensy4.1 and use it as secondary source to read programs from a usb pen drive in addition to sd card. I hope it should be easy. I use PIO for compilation. Can you guide me through the process on which library to include for it and how to host its space alongwith sd on server (including ftp).
Thanks in advance and a big salute for whatever you have done for the DIY CNC community.
Regards,
Ravi

terjeio, 2022-07-13:
It seems the SD card library already [supports USB drives](https://github.com/WMXZ-EU/uSDFS#--07-may-2019). I guess [sdcard.c](https://github.com/grblHAL/Plugin_SD_card/blob/master/sdcard.c) needs to be changed to mount the USB drive instead of the SD card to make it work. More changes is likely needed to make it work alogside the SD card.

karoria, 2022-07-13:
Ok. Will try it. Thanks.

karoria, 2022-09-17:
Hi @terjeio sorry I couldn't find what needs to be changed in sdcard.c to mount USB drive instead of SD card. Will you please point me?

terjeio, 2022-09-17:
I believe it is the [device name set here](https://github.com/grblHAL/Plugin_SD_card/blob/357cd53ebc69bb4d79df789a6054609f76d90a97/sdcard.c#L49-L51).

Perhaps this snippet from one of the uSDFS examples is a clue for what to use?
```
#if TEST_DRV == 0
  const char *Dev = "0:/";  // SPI
#elif TEST_DRV == 1
  const char *Dev = "1:/";  // SDHC
#elif TEST_DRV == 2
  const char *Dev = "2:/";  // USB
#endif
```

karoria, 2022-09-18:
Thanks for hint. I got the definition in driver.c changed it to 2:/ and USB opens through FTP. But there are some weird files along with. I have formatted it with FAT32 but the www directory has some different name like WW~1 I think it should be related to FS or FTP permissions. my webUI doesn't open inspite I have put the www directory in root of USB drive.

terjeio, 2022-09-18:
> the www directory has some different name like WW~1

Long filename support is not enabled for FatFs for some reason?

You may also try WebDAV via WinSCP - this does not keep track of current working directory that may mess things up with ftp.

> my webUI doesn't open inspite I have put the www directory in root of USB drive.

What about [ESP3D WebUI](https://github.com/luc-github/ESP3D-WEBUI/tree/3.0/dist/CNC/GRBLHal) - same?
And do you use the webui backend?
If so then you may try WEBUI_ENABLE 2 instead of 1 to force the v2 protocol - if set to 1 I try to auto detect which one to use. Also be aware that the webui backend is in a somewhat fluid state now due to v3 support beeing finalized.
Oh, and your webui code is gzipped (index.html.gz)?

karoria, 2022-09-19:
I am not using ESP3D webUI. But will check with it. I am using grblTouch which is not gzipped, so simple to work with. I also don't use any webUI backend (actually I don't understand what do you mean by backend) but I can surely tell you that I have not used any of the code which is inside ESP3D webUI. Will try with Luc's webUI and post the results here as I get time.

terjeio, 2022-09-19:
> (actually I don't understand what do you mean by backend)

Ok, this may explain it - I have removed support for fetching _index.html_ from the _www_ directory if no [backend](https://github.com/grblHAL/Plugin_WebUI/blob/main/server.c) is present. You will have to move it too the root directory or implement a mini backend the [redirects the request](https://github.com/grblHAL/Plugin_WebUI/blob/2fe5ef2046e7414fc54bb88a48b25e1d2c0584d6/server.c#L727-L762).

A backend is code that processes http requests in different ways depending on the url provided, it is set up by registering [url handlers](https://github.com/grblHAL/Plugin_WebUI/blob/2fe5ef2046e7414fc54bb88a48b25e1d2c0584d6/server.c#L772-L886) with the http server. It may also register a [file redirector](https://github.com/grblHAL/Plugin_WebUI/blob/2fe5ef2046e7414fc54bb88a48b25e1d2c0584d6/server.c#L791) with the server that get called when a file is not found - this can be used to look up the file elsewhere in the filing system.

The iMXRT1062 driver has gotten littlefs support, this is for storing files in flash. You may use this to store the grblTouch code, with file redirection you can check if the code is present on the sdcard, if not fall back to littlefs.

karoria, 2022-09-19:
Ok. That sounds interesting. Actually I am using server.c but my code is little older and there may be lots of changes by you. Will play around with your suggestions. BTW, will you point me on how to save my grblTouch web app to flash? I can see the path you have mentioned in server.c is /embedded for it.

terjeio, 2022-09-19:
>> BTW, will you point me on how to save my grblTouch web app to flash? I can see the path you have mentioned in server.c is /embedded for it.

The actual file system path is /littlefs - the /embedded file system path is read-only and files are stored in a [.c file](https://github.com/grblHAL/Plugin_WebUI/blob/main/fs_embedded.c). The WebUI index.html.gz stored in this file is for updating the WebUI image in littlefs, either when _index.html.gz_ is not found in the _/_  or _/www_ directories or when http://<ip>/forcefallback=yes is opened. All this is handled via file redirection in server.c - and you may redirect whatever filenames you want.

With WinSCP I open /littlefs mount via WebDAV by specifying _/littlefs_ as the remote directory in advanced settings.

[image]

ftp access to mounts other than the root mount is not yet working - I'll fix this later.

A bit more complicated than before I know, but the http server daemon is a lot more flexible now.

If storing grblTouch in flash you really should minimize the javascript and gzip the result- you will save a lot of flash that way. It might not even be possible to store it in flash if not doing so.