grblHAL/iMXRT1062 #43: Can I add the 7th and 8th axis support.？

pzh11001, 2022-07-27:
I checked part of the source code and found that there are definitions of U-axis and V-axis in many places. I try to imitate the programming, add the U-axis and v-axis code where there is ABC axis code, simply and completely imitate the macro definition and variable declaration of ABC axis. Unfortunately, although I have checked it many times to make sure that the U-axis and V-axis codes have been added to all places with ABC axis codes. The compilation can pass smoothly without any errors, and it can also be downloaded to teensy4.1 normally. But when I try to control with the serial assistant, the MCU microcomputer didn't respond at all, and the USB serial port of the computer was constantly lost and reconnected. This shows that the MCU is constantly restarting. There must be something wrong. I think grblHAL should definitely support 8 or 9 axes running at the same time. What should I do?

pzh11001, 2022-07-27:
i use  ArduinoIDE1.8.19。  Just in config.h set N_AXIS . If axis is larger than 6, MCU will crash. As long as 6 axes or less are set. Everything is back to normal. Why doesn't the code I added work

terjeio, 2022-07-27:
> I think grblHAL should definitely support 8 or 9 axes running at the same time. 

8 axes should be relatively easy to add since a byte is used to hold information such as step and direction output. 9 (or more) requires two bytes and will trigger more significant changes?

> What should I do?

Debug? Support for U and V is work (in slow) progress for me and there may be issues such as settings/parameter storage allocation that has to be resolved.  It is a pity that the Teensy is not easily debuggable, myself I would use a Nucleo STM32F756 board to debug as it has plenty of pins.

For the iMXRT1062: if you have a UART <> USB breakout board you may either add debug output to a UART channel or switch the primary output stream to that. Debug output can be enabled in _grbl/config.h_ (`DEBUGOUT`) and writing is via `debug_write()`.

pzh11001, 2022-07-27:
> 8 axes should be relatively easy to add since a byte is used to hold information such as step and direction output. 9 (or more) requires two bytes and will trigger more significant changes?
>
>
> Debug? [...]

I see. Thank you very much. I think I need to prepare a UART to USB device. Connect to uart1 of teensy4.1.

pzh11001, 2022-07-27:
Finally found the problem, The starting macro definition in 《nvc.h》  file is too small. After the expansion, grbHALl will not crash and can operate normally. Next, the 7-axis and 8-axis motors still don't rotate, but the command "?" of grblhal has been able to correctly show the coordinates, and I already know where to find the problem. It should be solved soon!

Thank you again for your help!

pzh11001, 2022-07-27:
Debug output can be enabled in _grbl/config.h_ (`DEBUGOUT`) and writing is via `debug_write()`.

Finally found the problem, The starting macro definition in 《nvc.h》 file is too small. After the expansion, grbHALl will not crash and can operate normally. Next, the 7-axis and 8-axis motors still don't rotate, but the command "?" of grblhal has been able to correctly show the coordinates, and I already know where to find the problem. It should be solved soon!

Thank you again for your help!

pzh11001, 2022-07-27:
> 

Finally found the problem, The starting macro definition in 《nvc.h》 file is too small. After the expansion, grbHALl will not crash and can operate normally. Next, the 7-axis and 8-axis motors still don't rotate, but the command "?" of grblhal has been able to correctly show the coordinates, and I already know where to find the problem. It should be solved soon!

Thank you again for your help!

closed 2022-08-22 by pzh11001

terjeio, 2022-09-19:
> The starting macro definition in 《nvc.h》 file is too small

I've fixed this, and related stuff, in the latest commit.