grblHAL/iMXRT1062 #45: Controller hardlocks after Invalid G-Code Command when using Ethernet connection

Dietz0r, 2022-09-02:
Controller:
Teensy4.1 on grblHAL2000 Board
Firmware:
https://github.com/Expatria-Technologies/iMXRT1062/releases/tag/v1.03 
ioSender: 2.0.40 (latest from Jul. 18)

Issue:
When issuing a wrong G-Code Command (e.g.: G1 x200,1) the controller hardlocks after issuing any other new commands. Keyboard jogging in between works until the controller hardlocks. Connection cannot be reestablished with controller over any means. Issue is only present on Ethernet connection. Issue cannot be replicated on USB Serial connection. 

I took a video of the exception: https://www.youtube.com/watch?v=cb4VMLmiTx

terjeio, 2022-09-03:
It is a sender issue, will create a edge build with fix soon. The lock can be cleared by sending a system command, eg. `$I`, from the MDI.

Dietz0r, 2022-09-03:
As always i'm astonished how fast you are.
Could you provide any info on what went wrong? I'd like to understand it a bit better. 

Also: Checked with PuTTy over Ehternet Telnet and no lock ... so you seem to be spot on!

terjeio, 2022-09-14:
> Could you provide any info on what went wrong? I'd like to understand it a bit better.

A bug in the [websocket/telnet code](https://github.com/terjeio/ioSender/blob/29f206a11ba632bbbda509c6aa70a51616a031f1/CNC%20Core/CNC%20Core/TelnetStream.cs#L142-L150) that prevented sending a blank line (a single carriage return character) to the controller, this is used to reset the error status. The working code is:
```
        public void WriteCommand(string command)
        {
            state = Comms.State.AwaitAck;

            if (command.Length == 1 && command != GrblConstants.CMD_PROGRAM_DEMARCATION)
                WriteByte((byte)command.ToCharArray()[0]);
            else
            {
                command += "\r";
                WriteString(command);
            }
        }
```

closed 2022-09-14 by terjeio