grblHAL/iMXRT1062 #46: Properly compile firmware for a dual-stepper-Y machine

szaghi, 2022-09-14:
Dear all,
I am trying to compile the firmware for a machine using a dual-stepper configuration for moving the Y axis. I am using GRBLHAL2000 board and I would like the auto-square enabled for the Y axis for which I am using two separate steppers (connected to Y and A drivers) and two separate end-stops. I read the documentation, but I am confused (I am also new to GRBL firmware), in particular, I am not sure which files have to be customized. I found that `my_machine.h` contains a section dedicated to GANGED/AUTO-SQUARE axis and I am wondering if defining `#define Y_AUTO_SQUARE       1` is enough or if there are other settings to be done.

Moreover, I am not sure if setting stepper/driver A as the second Y axis is a compile-time setting or not (and in the case where is this setting).

I am sorry if this help request is off-topic or if similar questions have been already answered.

Kind regards.
Stefano

P.S. A complete example of a configuration for a machine with dual-stepper-Y would be very precious for me, is there any example available?

terjeio, 2022-09-14:
> I found that my_machine.h contains a section dedicated to GANGED/AUTO-SQUARE axis and I am wondering if defining #define Y_AUTO_SQUARE 1 is enough or if there are other settings to be done.

This is the one, uncomment the line and recompile/reflash - ensure you have selected the correct board first (further up in the file).

> Moreover, I am not sure if setting stepper/driver A as the second Y axis is a compile-time setting or not (and in the case where is this setting).

All settings in _my_machine.h_ are compile time settings.

There are additional run time $-settings that becomes available after reflashing, these are related to limit switches offset adjustment and tolerances. The is also one for reversing the direction of the second motor.

> P.S. A complete example of a configuration for a machine with dual-stepper-Y would be very precious for me, is there any example available?

There are none available, what could be nice is a write-up of how to adjust the run time settings to ensure that the auto-squaring routine achieves a perfecty square result - it might be hard to mechanically adjust the limit switches for that. You can write it?

szaghi, 2022-09-15:
Hi @terjeio , thank you for the quick answer.
 
> This is the one, uncomment the line and recompile/reflash - ensure you have selected the correct board first (further up in the file).

Perfect. What about the `Y_GANGED` setting? Is to be set?
 
> 
> All settings in _my_machine.h_ are compile time settings.
> 

Yes, this was clear, I was in doubt on how set A driver as second Y axis, as a compile-time setting in `my_machine.h` or as run-time setting. 

> There are additional run time $-settings that becomes available after reflashing, these are related to limit switches offset adjustment and tolerances. The is also one for reversing the direction of the second motor.

Perfect. Thus, if I have understood well, to set A-driver as a second Y axis I have to rely on the run-time settings, right?

> There are none available, what could be nice is a write-up of how to adjust the run time settings to ensure that the auto-squaring routine achieves a perfecty square result - it might be hard to mechanically adjust the limit switches for that. You can write it?

If I will succeed, I will be very happy to share my experience and share my configs and workflow.

Kind regards,
Stefano

szaghi, 2022-09-15:
@terjeio I have just found a very interesting post of Phil Barrett concerning auto-square

https://www.grbl.org/single-post/it-s-hip-to-be-auto-square

If I understand well, defining `Y_GANGED` means that the first available driver, A in my case, is ganged with the Y driver, thus I do not need to search for a run-time setting (nice). Thus, for my config, I have just to define 

```c
#define Y_GANGED            1
#define Y_AUTO_SQUARE       1
```
The tricky part will be to adjust the limit switches in the correct position (I will report my experience).

Thank you again.

Kind regards,
Stefano

terjeio, 2022-09-15:
>  I have just found a very interesting post of Phil Barrett concerning auto-square

Nice find - I have not seen Phil's article about auto squaring before.

Enabling auto squaring for an axis automatically enables ganging so uncommenting `#define Y_GANGED 1` is not necessary.

> The tricky part will be to adjust the limit switches in the correct position (I will report my experience).

FYI you can set an offset with $171 for the Y-axis, up to +-10mm.

szaghi, 2022-09-16:
> Enabling auto squaring for an axis automatically enables ganging so uncommenting `#define Y_GANGED 1` is not necessary.

Wonderful, thanks.

> FYI you can set an offset with $171 for the Y-axis, up to +-10mm.

Thank you very much!

Kind regards, 
Stefano

P.S. I am closing this issue, your help has been very appreciated. If I will succeed with auto-squaring I will send you my feedback

closed 2022-09-16 by szaghi