grblHAL/iMXRT1062 #47: Teensy crashes with LITTLEFS

BehemothTheKitten, 2022-09-16:
Dear All,

Teensy crashes immediately after booting up when either LITTLEFS_ENABLE or WEBUI_ENABLE is set. The controller becomes inaccessible over USB or Ethernet. I used Arduino 1.8.19 and Teensyduino 1.57 to build grblHAL. Everything seems to work normally when these two plugins are disabled. 

Thank you.

terjeio, 2022-09-16:
Which version? Do the Teensy repeatedly reboot?
It could be due to littlefs failing to mount the drive due to "corrupted" flash.
Can you [comment out this line](https://github.com/grblHAL/Plugin_SD_card/blob/357cd53ebc69bb4d79df789a6054609f76d90a97/fs_littlefs.c#L325) to force a format before mounting?

Some related info in [this thread](https://forum.pjrc.com/threads/67232-Teensy-4-What-triggers-Halfkay-to-do-a-full-flash-erase/page2).

BehemothTheKitten, 2022-09-16:
Thank you for getting back to me. 

It is VER:1.1f.20220914 on a Teensy 4.1 and yes, it does keep keep rebooting every ~10 seconds.

I've been able to replicate the problem on two additional bare Teensy boards, including a brand new one.  Forcing a format, as you recommended, does resolve it. In fact, once I have flashed the modified firmware and formatted the file system, I can now use the original code without commenting out the IF statement. 

1. Build unmodified grblHAL 20220914 and upload it to a brand new Teensy.
2. Teensy keeps rebooting
3. Comment out the IF statement, build grblHAL, and upload it to Teensy
4. grblHAL works normally
5. Build unmodified grblHAL 20220914 and upload it to Teensy.
6. grblHAL continues to work normally

I've been able to use these steps to "fix" the problem on 3 different boards (2 bare controllers and one installed on T41U5XBB). However, unmodified grblHAL 20220914 refuses to work on a "virgin" Teensy.

terjeio, 2022-09-18:
Great, I guess it is due to assert statements in the littlefs code that triggers the reset. Cant't remember now if this happened for me as well as I worked with several implementations at the same time - and I had similar problems with the RP2040 driver.

What could be tried is to erase the flash area used for littlefs and see if it still crashes on mount. If not the perhaps a workaround for this is to release the erase code and instruct users to run it before enabling littlefs/webui?

BehemothTheKitten, 2022-09-18:
It looks like the fix is permanent and since I'm out of unused Teensy boards, I can no longer replicate the problem. I tried the 15 second reset, which did load the blinky program, but failed to overwrite the region used by littlefs.

Either that or check for a "file system formatted" flag in EEPROM (real or emulated) before attempting to mount the FS - if not set, format littlefs, and then set it. Since I have a deep appreciation of the Dunning-Kruger Effect and absolutely no understanding of your code, I will not bother with a useless pull request.

terjeio, 2022-09-27:
I will turn off littlefs assertions in the next commit and gamble on an [error is returned](https://github.com/ARMmbed/mbed-os/issues/11704#issuecomment-562259833](https://github.com/littlefs-project/littlefs/issues/390)) instead causing a format.

I'll keep this issue open for now, if somebody else experience crashes please report.

terjeio, 2023-08-26:
I believe this has been fixed in later builds.

closed 2023-08-26 by terjeio