grblHAL/iMXRT1062 #48: Coolant Flood/Mist pins reversed

rschell, 2022-09-26:
Built a T41U5XBB board recently (v2.09), flashed it originally using a pre-built 3 axis, auto squaring binary.  

Download the iMXRT1062 driver and built it using the Arduino IDE.  When uploaded I noticed that the Flood and Mist board outputs were reversed.  I swapped pins 18 and 19 in the T41U5XBB_map.h and my board correctly responds to a M7 command from bCNC now.  Might confirm with Phil if my changes are correct.

terjeio, 2022-09-27:
This is due to a [regression](https://github.com/grblHAL/iMXRT1062/blob/14054c3308d61b245ba91d20db83a6ce8f3de60b/grblHAL_Teensy4/src/driver.c#L482-L485) introduced in May last year, will fix in the next commit. Thanks for reporting.

rschell, 2022-09-29:
Fixed.  Thanks

closed 2022-09-29 by rschell