grblHAL/iMXRT1062 #49: Compile errors with latest code.

andrewmarles, 2022-10-02:
Trying to publish some updated builds, I am getting some compile errors that seem to be related mainly to the RTC functions.  I get these errors even with a completely fresh checkout of the core iMXRT1062 repo.

```
src\driver.c:2215:16: error: invalid initializer
src\driver.c:2217:13: error: incompatible type for argument 1 of 'rtc_set'
     rtc_set(t);
             ^

src\driver.c: In function 'get_rtc_time':
src\driver.c:2225:20: error: invalid initializer
         time_t t = rtc_get();

src\grbl\nuts_bolts.c: In function 'get_datetime':
src\grbl\nuts_bolts.c:287:22: error: storage size of 'dt' isn't known
     static struct tm dt;
                      ^

src\grbl\nuts_bolts.c:293:27: error: invalid application of 'sizeof' to incomplete type 'struct tm'
     memset(&dt, 0, sizeof(struct tm));
                           ^
```

terjeio, 2022-10-02:
Which Teensyduino version? It compiles and works for me with 1.54 (Arduino IDE).

andrewmarles, 2022-10-02:
PlatformIO.ini is pulling down 1.53 from line 47:

`platform = ${common_teensy.platform}@4.12`

I tried with the latest 4.17 and teensyduino 1.57, but that led to the errors from **https://github.com/grblHAL/iMXRT1062/issues/15**

Will try it with 1.54.

andrewmarles, 2022-10-02:
Modifying the platformIO.ini from the repo to pull in `https://github.com/WMXZ-EU/uSDFS` I am able to compile with 1.54.  It's not completely clear to me if this is the expected resolution for #15 ?

terjeio, 2022-10-02:
I was wrong, I have updated to 1.56...
1.57 reintroduces the _'msSCSICapacity_t' has not been declared_ error?

> It's not completely clear to me if this is the expected resolution for https://github.com/grblHAL/iMXRT1062/issues/15 ?

It is, the random delay bug was nasty as it did not show up in all grblHAL configurations.

andrewmarles, 2022-10-02:
Ok, at least I can claim some small grip on my sanity.  I will also try with 1.56.

andrewmarles, 2022-10-02:
With 1.56 and the patch it compiles.  I'll look at making a PR to update platformio.ini.

agent-r, 2022-10-21:
Could someone give a short discription what changes to make in platfomio.ini and what/how to patch?

dresco, 2022-10-21:
> Could someone give a short discription what changes to make in platfomio.ini and what/how to patch?

Under lib_deps, remove the two wwatson sources, and add `https://github.com/WMXZ-EU/uSDFS`. This still needs patching though, so after first build attempt, open up the `.pio/libdeps/teensy41/uSDFS` folder and copy in the updated files from the included uSDFS.zip.

Then change the platform line to `platform = ${common_teensy.platform}@4.15` - this should pull in the 1.56 teensyduino library.

@terjeio @andrewmarles I suppose the upstream library could be patched at installation with an extra_scripts directive, but perhaps easier to just point to a pre-patched fork?

terjeio, 2022-10-21:
> perhaps easier to just point to a pre-patched fork?

Yes I think so, or IMO rather include the source in the driver. A pity that the Arduino IDE is so simple and that it was not possible to get the uSDFS fixed...

BTW building with the new [web builder](https://github.com/grblHAL/core/discussions/203) is another alternative? Perhaps ditch Arduino IDE support when web builder is deemed stable?

szaghi, 2022-10-21:
Hi @dresco , I have tried to execute your suggestions, but I have failed. In particular I missed the point

> remove the two wwatson soruces

Which files have I to remove?

The other modifications should have been done correctly, but I have the following compiling errors:
```
Compiling .pio\build\teensy41\libd7d\uSDFS\utility\sd_spi.c.o
In file included from .pio\libdeps\teensy41\MSC-non-blocking\src\msc.h:31:0,
                 from .pio\libdeps\teensy41\MSC-non-blocking\src\MassStorageDriver.cpp:29:
C:\Users\stefa\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2056:2: error: 'msSCSICapacity_t' does not name a type
  msSCSICapacity_t msCapacity;
  ^
C:\Users\stefa\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2057:2: error: 'msInquiryResponse_t' does not name a type
  msInquiryResponse_t msInquiry;
  ^
C:\Users\stefa\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2058:2: error: 'msRequestSenseResponse_t' does not name a type
  msRequestSenseResponse_t msSense;
  ^
C:\Users\stefa\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2059:2: error: 'msDriveInfo_t' does not name a type
  msDriveInfo_t msDriveInfo;
  ^
C:\Users\stefa\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2078:31: error: 'msSCSICapacity_t' has not been declared
  uint8_t msReadDeviceCapacity(msSCSICapacity_t * const Capacity);
                               ^
compilation terminated due to -fmax-errors=5.
In file included from .pio\libdeps\teensy41\MSC-non-blocking\src\msc.h:31:0,
                 from .pio\libdeps\teensy41\MSC-non-blocking\src\MassStorageHost.cpp:31:
C:\Users\stefa\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2056:2: error: 'msSCSICapacity_t' does not name a type
  msSCSICapacity_t msCapacity;
  ^
C:\Users\stefa\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2057:2: error: 'msInquiryResponse_t' does not name a type
  msInquiryResponse_t msInquiry;
  ^
C:\Users\stefa\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2058:2: error: 'msRequestSenseResponse_t' does not name a type
  msRequestSenseResponse_t msSense;
  ^
C:\Users\stefa\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2059:2: error: 'msDriveInfo_t' does not name a type
  msDriveInfo_t msDriveInfo;
  ^
C:\Users\stefa\.platformio\packages\framework-arduinoteensy\libraries\USBHost_t36/USBHost_t36.h:2078:31: error: 'msSCSICapacity_t' has not been declared
  uint8_t msReadDeviceCapacity(msSCSICapacity_t * const Capacity);
                               ^
compilation terminated due to -fmax-errors=5.
Compiling .pio\build\teensy41\liba9e\lwip\api\api_lib.c.o
*** [.pio\build\teensy41\libf4b\MSC-non-blocking\MassStorageDriver.cpp.o] Error 1
*** [.pio\build\teensy41\libf4b\MSC-non-blocking\MassStorageHost.cpp.o] Error 1
```

Can someone help me to build the firmware?

Kind regards

dresco, 2022-10-21:
> Which files have I to remove? Can someone help me to build the firmware?

Replace these two lines in the platformio.ini with `https://github.com/WMXZ-EU/uSDFS`

https://github.com/grblHAL/iMXRT1062/blob/f94a0596bfd1fb03a1d5d74a054e5d7115e28ca6/grblHAL_Teensy4/platformio.ini#L32-L33

szaghi, 2022-10-21:
@dresco it worked, thank you very much!

szaghi, 2022-10-22:
Hi all, I am very sorry to bother you again, but I have a new problem with the latest firmware that I compiled yesterday with the help of @dresco : if I activate the spindle plugin I got the following error:

```
src\driver.c:1518:34: error: 'spindleConfig' undeclared (first use in this function)
         if(hal.spindle.config == spindleConfig)
                                  ^
src\driver.c:1518:34: note: each undeclared identifier is reported only once for each function it appears in
src\driver.c:1519:13: warning: implicit declaration of function 'spindleConfig' [-Wimplicit-function-declaration]
             spindleConfig();
             ^
*** [.pio\build\teensy41\src\driver.c.o] Error 1
```
with the previous version I used (4.12) this did not happened. Can someone help me? 
Kind regards,
Stefano

terjeio, 2022-10-22:
> .. I have a new problem with the latest firmware

@szaghi Sorry, I have indroduced a new bug.

Change this line to:
https://github.com/grblHAL/iMXRT1062/blob/f94a0596bfd1fb03a1d5d74a054e5d7115e28ca6/grblHAL_Teensy4/src/driver.c#L1522
to:

`#ifdef DRIVER_SPINDLE && SPINDLE_PWM_PIN`

as a quick fix.

szaghi, 2022-10-22:
@terjeio Thank you very much for the lighting help, it is really appreciated.

I'll test later today.
kind regards,
Stefano

szaghi, 2022-10-22:
> @szaghi Sorry, I have indroduced a new bug.
>
> Change this line to:
>
> https://github.com/grblHAL/iMXRT1062/blob/f94a0596bfd1fb03a1d5d74a054e5d7115e28ca6/grblHAL_Teensy4/src/driver.c#L1522
>
>
> to:
> `#ifdef [...]

@terjeio 

It seems to have not affected my error (that indeed seems to be into the function immediately above, line 1518). I still have the compiling error

```
src\driver.c: In function 'settings_changed':
src\driver.c:1518:34: error: 'spindleConfig' undeclared (first use in this function)
```

terjeio, 2022-10-22:
Oops - wrong line marked, change 1517 instead...

szaghi, 2022-10-22:
@terjeio it works!

Thank you again

karoria, 2022-12-05:
Thanks to all. This saved me too.

terjeio, 2022-12-05:
> .pio/libdeps/teensy41/uSDFS/src/uSDFS.h:27:17: error: redefinition of 'struct tm'

Is line 8 in this file `#define HAVETM`? If not then the required [patch](https://github.com/grblHAL/iMXRT1062/blob/master/patches/uSDFS.zip) has not been applied.

karoria, 2022-12-05:
@terjeio Thanks for reply. Yes I missed that patch but now I did follow both patches of network and SD card plugins but now compile error changes to something else as follows:

 *  Executing task in folder grblHAL_Teensy4: platformio run 

Warning! Ignore unknown configuration option `monitor_flags` in section [env]
Processing teensy41 (board: teensy41; platform: teensy@4.16; framework: arduino)
-----------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/teensy/teensy41.html
PLATFORM: Teensy (4.16.0) > Teensy 4.1
HARDWARE: IMXRT1062 600MHz, 512KB RAM, 7.75MB Flash
DEBUG: Current (jlink) External (jlink)
PACKAGES: 
 - framework-arduinoteensy @ 1.156.0 (1.56) 
 - tool-teensy @ 1.154.210805 (1.54) 
 - toolchain-gccarmnoneeabi @ 1.50401.190816 (5.4.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 94 compatible libraries
Scanning dependencies...
Dependency Graph
|-- uSDFS @ 1.1.3+sha.478e1d5
|   |-- USBHost_t36 @ 0.1
|-- lwip @ 2.0.2+sha.3127918
Building in release mode
Compiling .pio/build/teensy41/src/driver.c.o
Compiling .pio/build/teensy41/src/eeprom/eeprom_24AAxxx.c.o
Compiling .pio/build/teensy41/src/eeprom/eeprom_24LC16B.c.o
Compiling .pio/build/teensy41/src/encoder/encoder.c.o
Compiling .pio/build/teensy41/src/enet.c.o
Compiling .pio/build/teensy41/src/fans/fans.c.o
src/enet.c:136:38: warning: 'struct mdns_service' declared inside parameter list
 static void mdns_device_info (struct mdns_service *service, void *txt_userdata)
                                      ^
src/enet.c:136:38: warning: its scope is only this definition or declaration, which is probably not what you want
src/enet.c: In function 'mdns_device_info':
src/enet.c:141:5: warning: implicit declaration of function 'mdns_resp_add_service_txtitem' [-Wimplicit-function-declaration]
     mdns_resp_add_service_txtitem(service, "model=grblHAL", 13);
     ^
src/enet.c: At top level:
src/enet.c:146:39: warning: 'struct mdns_service' declared inside parameter list
 static void mdns_service_info (struct mdns_service *service, void *txt_userdata)
                                       ^
src/enet.c: In function 'netif_status_callback':
src/enet.c:193:9: warning: implicit declaration of function 'mdns_resp_init' [-Wimplicit-function-declaration]
         mdns_resp_init();
         ^
src/enet.c:195:29: warning: implicit declaration of function 'mdns_resp_add_netif' [-Wimplicit-function-declaration]
         if((services.mdns = mdns_resp_add_netif(netif_default, network.hostname, MDNS_TTL) == ERR_OK)) {
                             ^
src/enet.c:197:13: warning: implicit declaration of function 'mdns_resp_add_service' [-Wimplicit-function-declaration]
             mdns_resp_add_service(netif_default, network.hostname, "_device-info", DNSSD_PROTO_TCP, 0, MDNS_TTL, mdns_device_info, "version=" GRBL_VERSION);
             ^
src/enet.c:197:84: error: 'DNSSD_PROTO_TCP' undeclared (first use in this function)
             mdns_resp_add_service(netif_default, network.hostname, "_device-info", DNSSD_PROTO_TCP, 0, MDNS_TTL, mdns_device_info, "version=" GRBL_VERSION);
                                                                                    ^
src/enet.c:197:84: note: each undeclared identifier is reported only once for each function it appears in
Compiling .pio/build/teensy41/src/grbl/maslow.c.o
*** [.pio/build/teensy41/src/enet.c.o] Error 1
================== [FAILED] Took 7.48 seconds ==================

Environment    Status    Duration
-------------  --------  ------------
teensy41       FAILED    00:00:07.482
============= 1 failed, 0 succeeded in 00:00:07.482 =============

 *  The terminal process "platformio 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

Thanks for your time and efforts. Waiting for reply.

karoria, 2022-12-06:
Terje, I couldn't find updated lwIP patch. Disabling mdns protocol and compiling gives following error (pasted only critical lines):

src/networking/httpd.c: In function 'http_init_file':
src/networking/httpd.c:1959:17: error: 'vfs_file_t {aka struct <anonymous>}' has no member named 'is_custom_file'
         if (file->is_custom_file && (file->data == NULL))
                 ^
src/networking/httpd.c:1959:42: error: 'vfs_file_t {aka struct <anonymous>}' has no member named 'data'
         if (file->is_custom_file && (file->data == NULL))
                                          ^
*** [.pio/build/teensy41/src/networking/httpd.c.o] Error 1
========================= [FAILED] Took 11.39 seconds =========================

Environment    Status    Duration
-------------  --------  ------------
teensy41       FAILED    00:00:11.387
==================== 1 failed, 0 succeeded in 00:00:11.387 ====================

terjeio, 2022-12-06:
> I couldn't find updated lwIP patch.

It is [here](https://github.com/grblHAL/Plugin_WebUI/tree/main/lwIP%20patch/iMXRT1062). I guess it should be moved to the networking plugin...

The httpd.c errors above is likely due to lwipopts.h not beeing patched.

karoria, 2022-12-06:
OK. Now it compiles without "#define MDNS_ENABLE". I am OK with it as it doesn't add much value but still if you want to close the loop then I am ready to follow your instructions. If I #define MDNS_ENABLE and compile, following errors are shown:

src/enet.c:136:38: warning: 'struct mdns_service' declared inside parameter list
 static void mdns_device_info (struct mdns_service *service, void *txt_userdata)
                                      ^
src/enet.c:136:38: warning: its scope is only this definition or declaration, which is probably not what you want
src/enet.c: In function 'mdns_device_info':
src/enet.c:141:5: warning: implicit declaration of function 'mdns_resp_add_service_txtitem' [-Wimplicit-function-declaration]
     mdns_resp_add_service_txtitem(service, "model=grblHAL", 13);
     ^
src/enet.c: At top level:
src/enet.c:146:39: warning: 'struct mdns_service' declared inside parameter list
 static void mdns_service_info (struct mdns_service *service, void *txt_userdata)
                                       ^
src/enet.c: In function 'netif_status_callback':
src/enet.c:193:9: warning: implicit declaration of function 'mdns_resp_init' [-Wimplicit-function-declaration]
         mdns_resp_init();
         ^
src/enet.c:195:29: warning: implicit declaration of function 'mdns_resp_add_netif' [-Wimplicit-function-declaration]
         if((services.mdns = mdns_resp_add_netif(netif_default, network.hostname, MDNS_TTL) == ERR_OK)) {
                             ^
src/enet.c:197:13: warning: implicit declaration of function 'mdns_resp_add_service' [-Wimplicit-function-declaration]
             mdns_resp_add_service(netif_default, network.hostname, "_device-info", DNSSD_PROTO_TCP, 0, MDNS_TTL, mdns_device_info, "version=" GRBL_VERSION);
             ^
src/enet.c:197:84: error: 'DNSSD_PROTO_TCP' undeclared (first use in this function)
             mdns_resp_add_service(netif_default, network.hostname, "_device-info", DNSSD_PROTO_TCP, 0, MDNS_TTL, mdns_device_info, "version=" GRBL_VERSION);
                                                                                    ^
src/enet.c:197:84: note: each undeclared identifier is reported only once for each function it appears in
*** [.pio/build/teensy41/src/enet.c.o] Error 1
Compiling .pio/build/teensy41/src/grbl/maslow.c.o
========================= [FAILED] Took 7.77 seconds =========================

Environment    Status    Duration
-------------  --------  ------------
teensy41       FAILED    00:00:07.775
==================== 1 failed, 0 succeeded in 00:00:07.775 ====================

karoria, 2022-12-06:
Oh..It is a problem ! While I compiled without MDNS and flashed to teensy4.1, it shows success as usual but neither USB serial nor Ethernet is working. The device doesn't communicate at all. Any clues?

karoria, 2022-12-06:
@terjeio, I can confirm that Web Builder also has issues. I have selected T41U5XBB and T41BB5X pro both boards one by one and has selected relevant options in Network/WebUI pane but Ethernet, WS, FTP, HTTP, WebDAV enabled doesn't work. It generates hex file, also flashes well but no communication. I have my old pio project still working and its hex also flashes by teensy loader well and communication just works. I have seen this after above issues to confirm that the board doesn't have any problem. Till the time you come home and look in the matter, I am happy with my old firmware. Thanks a lot again.

terjeio, 2022-12-07:
@karoria I forgot one possible issue with littlefs that may cause a hang (an assert is triggered somewhere when an error should be returned). Comment out [this line](https://github.com/grblHAL/Plugin_SD_card/blob/c902f11fe05b7d7072dd096f952e41ad15f934e5/fs_littlefs.c#L325), recompile and reflash - then uncomment again. This will force a format of the littlefs storage area.

karoria, 2022-12-07:
Thanks. Will give it a try. I think i will be able to do that tomorrow.

karoria, 2022-12-08:
> @karoria I forgot one possible issue with littlefs that may cause a hang (an assert is triggered somewhere when an error should be returned). Comment out [this [...]

Yes! Commenting out and recompile and refresh makes communication. Should I uncomment and refresh again? I am getting webUI-3 page when I open my static IP, but don't know what to do next. Can I use my grblTouch with flash memory? If yes, how? I have and old style www directory which needs to be hosted as http server.

terjeio, 2022-12-08:
> Should I uncomment and refresh again? 

Yes.

>  but don't know what to do next.

The page presented allows you to upload data to littlefs. This can be whatever you want. If _index.html.gz_ (compressed _index.html_), _favicon.ico_, _preferences.json_ or _ap_login.html_ is found in littlefs they will [take priority](https://github.com/grblHAL/Plugin_WebUI/blob/ba3955efcb7277494b71b1e5b8a2c9f32eafe454/server.c#L713-L762) over files in the _www_ folder in the SD card file system (or in the embedded read-only file system if that is present).

> Can I use my grblTouch with flash memory?

Yes, provided that it is is compressed and fits in the available littlefs free space. Use the file upload button to upload.

karoria, 2022-12-08:
> Do you somehow erase the complete flash before writing the binary?

I followed your instructions of flashing with commented line first and then powered off the device and then flashing with uncommented line (as is). I don't know other method for erasing complete flash. Sorry for my incomplete knowledge in electronics area as I am having Mechanical background. I will dig more on low level driver though.

karoria, 2022-12-15:
Hi @terjeio, Are you back Home? When you get time please see to the matter. I am totally lost after trying hard.
Thanks,
Ravi

terjeio, 2022-12-17:
I will be back home at the end of January.

Have you tried with moving the storage area down in FLASH, eg. by 1Mb? 
BTW do you have an EEPROM for settings storage? If not then then it could be that the eeprom emulation code uses an overlapping area and corrupting littlefs data.

karoria, 2022-12-17:
> I will be back home at the end of January.
> 
> 
> 
> Have you tried with moving the storage area down in FLASH, eg. by 1Mb? 
> 
> BTW do you have an EEPROM for settings storage? If not then then it could be that the eeprom emulation code uses an overlapping area and corrupting littlefs data.

Thanks for taking time out of your busy schedule and guiding. Will see to it. I have not added extra EEPROM to teensy4.1 board. Does it mean I use eeprom emulation code?

terjeio, 2022-12-17:
> I have not added extra EEPROM to teensy4.1 board. Does it mean I use eeprom emulation code?

Yes. It is possible to disable this but then settings will not survive a cold restart.

terjeio, 2023-01-27:
@karoria

> and has selected relevant options in Network/WebUI pane but Ethernet, WS, FTP, HTTP, WebDAV enabled doesn't work

This only adds the protocol code to the firmware, but does not enable the protocols. Either reset the settings with `$RST=*` or enable the respective protocols with `$70`.

> if I compile without commenting and flash, communication doesn't work.

I got the same issue when back home, when connected via USB to my Win10 machine after a update to the latest grblHAL version it constantly tried to enumerate the connection. Likely due to the controller rebooting after crashing in the littlefs code. Forcing a littlefs format cleared the issue and now I am not able to reproduce. PRJC claims that [the flash is erased when programming](https://github.com/PaulStoffregen/LittleFS#program-memory), but this is not the case as any content I save to littlefs survives a programming cycle. So something is (randomly?) corrupting the littlefs storage area - but what?
I guess this will take some time to identify and fix...

terjeio, 2023-01-28:
@karoria - it seems I have cracked your issue, new build with the fix just commited.

I am closing this, please open a new issue if your problem persists.

closed 2023-01-28 by terjeio

karoria, 2023-02-13:
@terjeio 
Sorry, I can't believe why I missed your last messages. Thanks for all you've done. I will let you know in case still I have any issues.