grblHAL/iMXRT1062 #50: z-axis TLO and probing report with bCNC

rschell, 2022-10-24:
Trying to chase down two issues interfacing with bCNC (building with Arduino 1.8.19 IDE): 

1. The TLO default report setting in grbl/config.h lists Z-axis as the default.  Without changing config.h, the TLO report includes XYZ coordinates (for a three axis machine), bCNC is expecting only the Z coordinate, so the TLO always gets set to 0.0 (the X-axis value).  I have changed config.h in my local version to only report the Z-axis and have modified bCNC to accept any number of axis.  This one I have solved, but how would I select the Z-axis TLO report in your new build system?
2. When bCNC performs an autoleveling scan, it issues a G38.2 command and waits for its completion.  Apparently, it expects a PRB response but in my grblhal version none is given.  I can force a response using $#, by modifying bCNC, but I thought this was planned for.  Is there a way to get a PRB response on the completion of the probe?  I have tried several ways of changing DEFAULT_REPORT_PROBE_COORDINATES, but still don't receive a PRB response on the probe completion.  Any tricks I can try?

Haven't tried the new build system yet, would it be able to accommodate these types of customization's?

terjeio, 2022-10-25:
> ... how would I select the Z-axis TLO report in your new build system?

Everything that is configurable via #define symbols can be added to the Web Builder UI. My plan is to add tabs to the UI, likely one for _Advanced  features_, in order to keep it simple for most users. IMO this option would belong in that.

>  I have tried several ways of changing DEFAULT_REPORT_PROBE_COORDINATES, but still don't receive a PRB response on the probe completion.

Including [restoring settings to default](https://github.com/grblHAL/core/blob/b0d9e9e78a12c99c37993fa211517388768c0bc0/config.h#L26-L28)?

You can also change the behaviour at run time with `$10`, `$$=10` will list the available flags.

> Haven't tried the new build system yet, would it be able to accommodate these types of customization's?

Yes - but I need more time, I have just started...

closed 2022-10-25 by rschell

terjeio, 2022-10-25:
The [Web Builder](http://svn.io-engineering.com:8080/) has now been updated with new tabs, more to come later.

rschell, 2022-10-25:
Made my first hex file, testing now using bCNC.  Appears to be operating correctly.  Nice to have an alternative build system.

Options used: 3-axis, Y-ganged auto squaring, Z-axis TLO.