grblHAL/iMXRT1062 #51: How to input Motor Fault signal?

karoria, 2022-12-19:
Hi @terjeio 

Enabling #define MOTOR_FAULT_ENABLE 1  and defining its pin in custom board map doesn't work for me to take Motor Fault input. Is there anything else I am missing? I am using a little older version of grblHAL as I am finding difficulties to compile and run latest code. My startup messages are here for your quick reference:

[VER:1.1f.20220123:]
[OPT:VNMSL+,35,1024,5,0]
[NEWOPT:ENUMS,RT+,ES,TC,SED,ETH,FTP,SD]
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[DRIVER:iMXRT1062]
[DRIVER VERSION:220111]
[DRIVER OPTIONS:USB.2]
[BOARD:BOARD_VALAY]
[AUX IO:0,2,0,0]
[IP:192.168.0.77]
[NETCON:Websocket]
[PLUGIN:WebUI v0.02]
[PLUGIN:SDCARD v1.05]
ok

BTW, I need functionality similar to Estop for Motor Fault. Just the console message should state that the Alarm is caused by motor fault and not by Estop.

terjeio, 2022-12-20:
The signal has to be added to the [driver.c](https://github.com/grblHAL/iMXRT1062/blob/master/grblHAL_Teensy4/src/driver.c) input pin handling code (in several places).

karoria, 2022-12-20:
Okay. Thanks.

karoria, 2022-12-22:
Hi @terjeio 
Thanks for your continuous support. I got the Motor Fault signal working with changes in driver.c as instructed. Now I want to invert the signal. My present $14 setting is 78, so I have to add 256 to it (as motor fault is on bit 8) which makes it 334. But somehow, it doesn't accept $14=334 with a message "Error 52: undefined". Any clues?

terjeio, 2022-12-22:
You have to tell the core that the signal is available [here](https://github.com/grblHAL/iMXRT1062/blob/eb82b3ffd0340287c943857426558ed4fc3a0885/grblHAL_Teensy4/src/driver.c#L2425-L2434) by adding:

`hal.signals_cap.motor_fault = On;`

Hopefully this will work...

karoria, 2022-12-22:
Thanks @terjeio . It worked!

closed 2022-12-22 by karoria