grblHAL/iMXRT1062 #59: BLTouch Servo Probe

TerisGH, 2023-05-19:
Is it possible to use a BLTouch probe with grblHAL on teensy41?
I have a small sketch using the arduino Servo library which works,
when using the library from inside a plugin, the probe doesnt work.

This seems to be a similar attempt but directly controlling the servo:
https://github.com/grblHAL/core/issues/280
Trying to apply the changes to this repo, noticed for teensy "RC Servo/ESC for spindle" isnt supported.
What would it take to add support for servos?

terjeio, 2023-05-19:
Wait a bit for the servo plugin (and lowlevel support for this driver) to become available.

> Trying to apply the changes to this repo, noticed for teensy "RC Servo/ESC for spindle" isnt supported.

This option, when available, cannot be used for a BLTouch probe.

TerisGH, 2023-05-19:
Thank you, I was thinking Servo Spindle support would provide some of the low level functionality necessary for analogue outputs.
Any idea why the Arduino servo library would not work when used from within GRBLHal?
What would it take to add the low level support to this driver? Something I could try to adapt?

terjeio, 2023-05-20:
> Any idea why the Arduino servo library would not work when used from within GRBLHal?

Not really, but my guess is that you do not have code in place to initialize it. Or that grblHAL already uses the timer you want to use for the servo.

> What would it take to add the low level support to this driver?

Some new code in the driver _ioports.c_.

> Something I could try to adapt?

Yes, but wait until the ioports API has been extended.
Or you may write your own servo plugin that is board specific and handles everything locally.

TerisGH, 2023-05-20:
Must have been the timers, got it working with hardware pwm, thanks you so much!

terjeio, 2023-05-27:
FYI I now have added driver support for [PWM output via M67/M68](https://github.com/grblHAL/core/blob/master/changelog.md#20230526) and @wakass has a plugin in the pipeline that adds M280 support (from Marlin specs) on top of that.

closed 2023-05-27 by terjeio