grblHAL/iMXRT1062 #60: RC PWM Spindle control

300bar, 2023-05-29:
Thank you Terje for the implementation of the RC PWM spindle speed control, it is working well. It does everything needed for a spindle speed controlled by a RC PWM BLD driver. I just notice one thing, when setting the max spindle speed, it is not working, it accepts the input but stays at 1000rpms max. for calculating the pwm signal.

dresco, 2023-05-29:
Would be worth checking these settings, I think 1000rpm is the default maximum value;

$30: Maximum spindle speed in RPM
$31: Minimum spindle speed in RPM

300bar, 2023-05-29:
Yes I just realized that grblHal needs to be reset to take a new value for the $30 Max rpms.... before I was playing changing the $30 and it was ignoring me, staying at 1000rpms behaviour. By the way, 1000 rpms is not the max, actually it is really a small figure for a spindle speed. 
Thank you, everything is working now.

closed 2023-08-14 by terjeio