grblHAL/iMXRT1062 #66: How to add new pins and use them.

ToRoCu, 2023-09-02:
Hello,
I would like to use pins 48-50 (Teensy 4.1 from PSRAM) in the programme, how do I set this up?

terjeio, 2023-09-03:
For digital I/O add them as [aux ports](https://github.com/grblHAL/iMXRT1062/blob/a37fa39c393fc0f5755d946ff9d6f584b5556e14/grblHAL_Teensy4/src/T41U5XBB_map.h#L101-L115) in your map and use them via M62-M66.

ToRoCu, 2023-09-03:
Sry, I should have described better what I want to do.
I would like to use the pins in a plugin.
I have already spent a few hours with the code and am still unclear about the correct use of the functions/classes such as "claiming an auxiliary pin for exclusive use".
Is "Y2_LIMIT_PIN" already used somewhere in the program or is it a placeholder?... If so, for what?

There are still some comments missing, I will add some when I am done and make a pull request for it.

terjeio, 2023-09-03:
> I would like to use the pins in a plugin.

You will have to define them as [aux in and/or aux out](https://github.com/grblHAL/iMXRT1062/blob/a37fa39c393fc0f5755d946ff9d6f584b5556e14/grblHAL_Teensy4/src/T41U5XBB_map.h#L101-L115). By default such pins are made available for the M62-M66 commands but can be claimed by plugins for exclusive use meaning the can no longer be set or read by those M commands. Use the `$pins` command to get a list of pins and the function they are assigned to.

> Is "Y2_LIMIT_PIN" already used somewhere in the program or is it a placeholder?... If so, for what?

It is and is the limit (home switch) pin for the second Y2 motor (for auto-squared Y axis).

ToRoCu, 2023-09-04:
Hello terjeio,
I am amazed at how quickly your answers come.
First of all, thank you for your valuable work on this project.

After many hours of trying to understand the code and get what i want to do...I need help or a thought to move forward.
what i want to do is "actually" simple...i need two outputs (HIGH) on two AUX ports when one of the two Y axes (double motor) is on limit switch, as long as a step signal is generated.

Y_LIMIT_PIN= AUXOUTPUT6_PIN<-- HIGH when Y_LIMIT_PIN is triggered and a step signal is generated.
Y2_LIMIT_PIN = AUXOUTPUT7_PIN<-- HIGH when Y2_LIMIT_PIN is triggered and a step signal is generated.

I have added "#elif Y_DOUBLED == 6" to motor.pins.h to make Y2_LIMIT_PIN not dependent on the other axes.

I hope you can help me, even though I know it's a big ask.

Thank you

terjeio, 2023-09-05:
I am not sure I understand what you want to achieve but I am pretty sure you are complicating it by modifying _motor_pins.h_.
Plugins can do a lot of fancy stuff by intercepting HAL calls and/or subscribing to core events - and I wonder if this is the way forward for you.
First, when do you need the aux ports to be set high? At all times or just when homing?
If during homing you can intercept the HAL call that reads the limit switches since they are continuously polled.
If at all times the there are other options.
A more detailed description of what you want will help me provide better guidance.

ToRoCu, 2023-09-05:
The AUX ports must be switched during the home cycle (HIGH), otherwise it is not necessary.
However, I wonder how I can manage this without manipulating motor_pins.h so that I can use the second Y2_limit input without having to sacrifice the A-axis.
If the limit switch (Y_LIMIT_PIN) is triggered, then (AUXOUTPUT6_PIN) should be switched to (HIGH).
The AUX port (AUXOUTPUT6_PIN) should only output a (HIGH) signal as long as a step signal is generated on the Y_axis.
The same then applies to "Y2_LIMIT_PIN, AUXOUTPUT7_PIN".

terjeio, 2023-09-05:
Check [this piece of code](https://github.com/grblHAL/ESP32/blob/master/main/BlackBoxX32.c), it intercepts the HAL get limit switch status and modifies it when homing. Step signals are generated as long as the homing sequence is active, in `onHomingRateSet()` the _axes_ parameter holds which axis/axes. It is also possible to intercept the HAL call that ouput pulses if you want to check at such a low level.

If you rename the init function to `my_plugin_init()` the it will be automatically be called on startup (since it is defined as a "weak" function in the core),

ToRoCu, 2023-09-05:
How do I get access to the Y2_LIMIT_PIN?

terjeio, 2023-09-05:
The status is available in the [limit_signals_t struct](http://svn.io-engineering.com/grblHAL/html/structlimit__signals__t.html), `min2.y` - available either by calling `hal.limits.get_state` directly or by intercepting the call as done in the code linked above. During homing the `limitsGetState()` call in the linked code will be called at a high frequency so you do not have to add it to the real time loop callback. Interrupts are disabled for limit pins involved in the homing sequence so it is not possible to get hold of the status via intercepting the limits interrupt callback.
You can of course use MCU specific calls or register access to get hold of the status but that is not needed nor portable.

ToRoCu, 2023-09-09:
If I rename the ### void board_init (void) to ### void my_plugin_init (void), I get error messages like _warning: assignment from incompatible pointer type_ or _too many arguments to function 'on_homing_completed'_ when compiling.

ToRoCu, 2023-09-09:
some things are working now, i'm starting to see the big picture behind the code.
There are still 2 things: 
1. how can I stop the homing "success/OK" until both Y endstops are activated?
2. it takes too long from Y_LIMIT_PIN is pressed until a "HIGH" is output at the output port.

terjeio, 2023-09-09:
I still do not understand clearly what you want to achieve...

1. By writing your own homing routine? If a switch fails to activate the controller will raise an alarm when the distance traveled is more than the configured distance. Or for an auto squared axis when the second motor drives the the axis too far out of square.  Is this what you want to block?
2. What is too long? a few microseconds? milliseconds? ... Post your code, either in a repo or in a comment so I can take a look.

[hal.stepper.disable_motors](http://svn.io-engineering.com/grblHAL/html/hal_8h.html#a0e43b5696c87ff54b34c88ef1f8f1615) could be another call you need to intercept?

ToRoCu, 2023-09-09:
I would like to expand the Autosquaring so that I don't have to waste an extra axle like A.
Currently, no matter which sensor triggers (Limit_Y or Limit_Y2) the homing cycle is completed with Success. I want to prevent this and want the axis to continue driving until the second sensor has also triggered. When both sensors have triggered, Homing is DONE for Y axis.

Too long is over 500 milliseconds, when a sensor triggers, the output must go HIGH as quickly as possible.

```
static limit_signals_t limitsGetState (void)
{
    limit_signals_t state;
    axes_signals_t dir_outbits;

    state = limits_get_state();
    dir_outbits.y;

    bool M_lockY1 = state.min.y;
    bool M_lockY2 = state.min2.y;

    hal.port.digital_out(port_lockY1, M_lockY1);
    hal.port.digital_out(port_lockY2, M_lockY2);

    return state;
}
```
This funktion is running in My_Plugin.

terjeio, 2023-09-11:
> Too long is over 500 milliseconds, when a sensor triggers, the output must go HIGH as quickly as possible.

If outside of homing the delay will be up to the real-time request interval (from the sender). Please post the complete plugin code, the snippet above does not provide any context.

ToRoCu, 2023-09-11:
```
#include "driver.h"
 
static uint8_t port_lockY1;
static uint8_t port_lockY2;

static axes_signals_t homing = {0};
static limits_get_state_ptr limits_get_state;
static on_homing_completed_ptr on_homing_completed;
static limits_enable_ptr limits_enable;
static on_report_options_ptr on_report_options;

static limit_signals_t limitsGetState (void)
{
    limit_signals_t state;
    axes_signals_t dir_outbits;

    state = limits_get_state();

    bool M_lockY1 = state.min.y;
    bool M_lockY2 = state.min2.y;

  if(homing.y && dir_outbits.y) { 
    hal.port.digital_out(port_lockY1, M_lockY1);
    hal.port.digital_out(port_lockY2, M_lockY2);
  }else{
    hal.port.digital_out(port_lockY1, false);
    hal.port.digital_out(port_lockY2, false);    
  }
  return state;
}

//AUX OUT config
static void onReportOptions (bool newopt)
{
    on_report_options(newopt);  // Call previous function in the chain.

    if(!newopt)                 // Add info about us to the $I report.
        hal.stream.write("[PLUGIN:Plugin Autosquare]" ASCII_EOL);
}

static void output_warning (uint_fast16_t state)
{
    report_message("An output port is required for my_plugin!", Message_Warning);
}

void my_plugin_init (void) {

    port_lockY2 = hal.port.num_digital_out - 1;     // Claim the
    hal.port.num_digital_out--;                     // last free port.

    if(hal.port.set_pin_description)                // Add label to the pin for the $pins command
      hal.port.set_pin_description(Port_Digital, Port_Output, port_lockY2, "Lock_Y2");

    port_lockY1 = hal.port.num_digital_out - 1;     // Claim the
    hal.port.num_digital_out--;                     // last free port.

    if(hal.port.set_pin_description)                // Add label to the pin for the $pins command
      hal.port.set_pin_description(Port_Digital, Port_Output, port_lockY1, "Lock_Y1");

    on_report_options = grbl.on_report_options;     // Add our plugin to to the options report chain
    grbl.on_report_options = onReportOptions;       // to tell the user we are active.

    limits_get_state = hal.limits.get_state;
    hal.limits.get_state = limitsGetState;

}
```

terjeio, 2023-09-11:
You want to do auto squaring with two motors connected to a single stepper driver?
Or two stepper drivers/motors connected to the same step/dir output from the controller?
The output signals are for switching motors (or rather drivers) in/out?
I guess this doable by defining the step/dir pins for M3 equal to the Y axis, trapping [hal.stepper.disable_motors](http://svn.io-engineering.com/grblHAL/html/hal_8h.html#a0e43b5696c87ff54b34c88ef1f8f1615) is then the correct way to control the outputs as this called when switching between motors during homing for an auto-squared axis.

ToRoCu, 2023-09-11:
> I guess this doable by defining the step/dir pins for M3 equal to the Y axis, trapping [...]

I think you are making what I am trying to achieve more complicated than it is :D.
2 drivers on one STEP/DIR output.
If I do the AutoSquaring by defining M3, I have to waste a whole Axis just to align the portal once.
would mean I can only build a 4 axis CNC with the IMXRT1062.
Currently I only have the problem that I want to wait until both Y sensors have triggered to finish the homing process. 
```
if(Y_LIMIT_PIN && Y2_LIMIT_PIN) 
    Homing = success;
```

terjeio, 2023-09-11:
> If I do the AutoSquaring by defining M3, I have to waste a whole Axis just to align the portal once.
would mean I can only build a 4 axis CNC with the IMXRT1062.

[driver.c supports up to 6 axes](https://github.com/grblHAL/iMXRT1062/blob/0a8dd63d3126a70e6736f79151c4221fe92a120b/grblHAL_Teensy4/src/driver.c#L409-L429), 2 of which can be auto-squared (1 in an 5 axes configuration), you will need a new board map to take avantage of that though. E.g. if you add M5 definitions to the board map you can have either 6 axes or 5 axes with one auto-squared. Duplicating Y-axis definitions to the highest numbered motor in use (except the limit pin) and configure the Y-axis as auto-squared might work as suggested above. E.g. for a 5 axis configuration Y2 will be M5, in a 4 axis it will be M4. ABC-axis motors are assigned from M3 upwards and second motors for auto squared axes downwards from the highest numbered needed.

> Currently I only have the problem that I want to wait until both Y sensors have triggered to finish the homing process.

This is with the Y-axis _not_ configured as auto squared then?
If so you may have to write your own homing routine as homing is not just waiting for both switches to be triggered - possible since homing just got its own function pointer: `grbl.home_machine`. It is not possible to preempt the [core homing implementation](https://github.com/grblHAL/core/blob/049f0f13e45211237f3b096083afb8d371aad9fb/machine_limits.c#L249-L506) by returning early when both switches are triggered: pulloff, locate phase, pulloff, possibly square adjustment and then work envelope configuration has to be completed first.
Since the core [queries the driver](https://github.com/grblHAL/core/blob/049f0f13e45211237f3b096083afb8d371aad9fb/machine_limits.c#L516-L519) for auto squared axes prior to calling the homing routine you could perhaps fake that a single single motor output can be used for auto squaring and still use the core homing routine by redirecting `hal.stepper.get_ganged` to your plugin. You would then have to route the step signal to the correct drivers via trapping `hal.stepper_disable_motors`.

ToRoCu, 2023-09-11:
> [driver.c supports up to 6 axes](https://github.com/grblHAL/iMXRT1062/blob/0a8dd63d3126a70e6736f79151c4221fe92a120b/grblHAL_Teensy4/src/driver.c#L409-L429), 2 of which can be auto-squared (1 in an 5 [...]

I know, that is in the _motor_pins.h_.

The problem is that the IMXRT1062 does not have unlimited free GPIO's available, so i cannot use the common way.
In my case all GPIO's are already fully defined + the GPIO's on the underside(PSRAM), which is why this option is unfortunately not possible for me.

Currently, auto squared is not configured because it then immediately uses the A axis as M3. "which is why I initially wanted to add another option to _motor_pins.h_".

I will test a few ways.

Thanks :)