grblHAL/iMXRT1062 #68: I have run the compile both with Arduino IDE and PlatformIO with MCP3221 Enabled.

sntlewis, 2023-09-11:
I have run the compile both with Arduino IDE and PlatformIO with MCP3221 Enabled.

The following compile time errors occur on lines 129 and 386 in ioports.c

src\ioports.c: In function 'wait_on_input':
src\ioports.c:129:20: error: 'analog_n_in' undeclared (first use in this function)
     else if(port < analog_n_in)
                    ^
src\ioports.c:129:20: note: each undeclared identifier is reported only once for each function it appears in
src\ioports.c: In function 'ioports_init':
src\ioports.c:386:34: error: 'analog_n_in' undeclared (first use in this function)
         hal.port.num_analog_in = analog_n_in = 1;

I have found a MCP3221 "daughter card" on Phil-Barrett's github.

_Originally posted by @sntlewis in https://github.com/grblHAL/iMXRT1062/discussions/67_

terjeio, 2023-09-11:
Oops, a regression - can you try with [changing these lines](https://github.com/grblHAL/iMXRT1062/blob/0a8dd63d3126a70e6736f79151c4221fe92a120b/grblHAL_Teensy4/src/ioports.c#L42-L45) to:

```
#if MCP3221_ENABLE
static xbar_t analog_in;
static uint_fast8_t analog_n_in;
static enumerate_pins_ptr on_enumerate_pins;
#endif
```

I'll fix this in the next commit.

sntlewis, 2023-09-11:
Thank fixed it.  Now onto the next level of hardware development, order parts for the mcp3221 bob.

closed 2023-09-14 by terjeio