grblHAL/iMXRT1062 #69: aggiungere pin alla configurazione

distebia, 2023-10-07:
Salve, attualmente ho un finecorsa per l'asse X (-x), uno per l'asse Y (-y) e uno per la Z (-z) che utilizzo normalmente per l'homing. è possibile aggiungere un +X, +Y e +Z da usare solo come hard limit, niente homing? 
La mia asse Z, inoltre, è composta da un piano che sale e viene rilevato da un finecorsa induttivo e retrocede per impostare la giusta distanza del mio fuoco laser (co2). Per motivi di sicurezza posso mettere un secondo set di finecorsa (2) per l'asse Z, uno sopra e uno sotto, e attivare, solo su questi, l'hard limit oppure qualcosa che blocca la macchina se attivato? Quello sopra posizionato 3-5mm oltre il sensore induttivo in modo tale da scongiurare un errore di lettura dell'induttivo sotto, anche per evitare che il piano sale fino a sbattere contro la testa laser.
Se è possibile questa configurazione cosa devo cambiare e compilare?

terjeio, 2023-10-07:
Legg til pindefinisjoner for X_LIMIT_PIN_MAX, Y_LIMIT_PIN_MAX og Z_LIMIT_PIN_MAX i mappingfilen du benytter.

distebia, 2023-10-07:
sono nuovo con grblhal e sto cercando di imparare...devo compilare per teensy 4.1 e T41U5XBB..esattamente quale è il percorso del file da modificare?

terjeio, 2023-10-08:
Kopier [T41U5XBB_map.h](https://github.com/grblHAL/iMXRT1062/blob/master/grblHAL_Teensy4/src/T41U5XBB_map.h) til _my_machine_map.h_ og modifiser den. Fjern kommentar fra [denne linjen](https://github.com/grblHAL/iMXRT1062/blob/b7469ce0b1d520e14e9a7076929921919c72395c/grblHAL_Teensy4/src/my_machine.h#L32) for å kompilere med _my_machine_map.h_. Dette sikrer at endringene ikke blir overskrevet ved en oppdatering av kildekoden.

distebia, 2023-10-08:
ok, perfetto aggiungo X_LIMIT_PIN_MAX sotto X_LIMIT_PIN e indico il pin di riferimento?

[image]

Ma posso attivare l'hard limit solo per alcuni pin di limit?

terjeio, 2023-10-08:
> ok, perfetto aggiungo X_LIMIT_PIN_MAX sotto X_LIMIT_PIN e indico il pin di riferimento?

Korrekt, bruk definisjonene for M3_LIMIT_PIN, M4_LIMIT_PIN og SAFETY_DOOR_PIN for innganger med optokoblere.

> Ma posso attivare l'hard limit solo per alcuni pin di limit?

Nei, men vær oppmerksom på at hvilken inngang som blir bruk for homing er avhengig av om retning er invertert eller ikke ($23).

distebia, 2023-10-08:
Ho definito i pin di X_LIMITS_PIN_MAX, Y_LIMITS_PIN_MAX e Z_LIMITS_PIN_MAX in questo modo prendendo il relativo pin da m3 m4 e safety door. Basta questo su my_machine_map.h? 

[image]

L'homing è definito su -X -Y e niente Z. Homing in basso a sx solo per X e Y. Faccio homing con endstop -x e -y, mentre i +x e +y mi servono per hard limits, anche se uso i soft limit...una sicurezza in più.
Per quanto riguarda la Z la mia idea è di usare il sensore induttivo per alzare il piano e posizionare il materiale alla giusta distanza per il mio fuoco laser. Il piano sale e incontra il sensore induttivo. metti caso mi dimentico di posizionare il piatto metallico per essere rilevato oppure un guasto al sensore induttivo...il piano continua a salire e va dritto a sbattere contro la testa laser. Per questo volevo altri 2 sensori, uno sopra e uno sotto , per limitare e bloccare la corsa del piano. quello sopra un 5mm più in alto del sensore induttivo. L'asse Z avra 1 sensore ottico sotto, un sensore ottico sopra e un sensore induttivo montato nella testa del laser un goccio sotto il sensore ottico posizionato in alto.

distebia, 2023-10-15:
Salve, utilizzo platformio su vs code, eseguo git clone --recurse-submodules https://github.com/grblHAL/iMXRT1062.git e modifico in questo modo i seguenti file che ho rinominato in .txt per esere aggiunti qua.
[my_machine.h.txt](https://github.com/grblHAL/iMXRT1062/files/12910615/my_machine.h.txt)
[my_machine_map.h.txt](https://github.com/grblHAL/iMXRT1062/files/12910616/my_machine_map.h.txt)
ma ottengo questo errore
src\networking\networking.c:86:9: warning: unknown option after '#pragma GCC diagnostic' kind [-Wpragmas]
 #pragma GCC diagnostic ignored "-Wstringop-overflow"
dove sbaglio?

terjeio, 2023-10-16:
Pin definitsjoer er nok ok, test om de virker som de skal.

> src\networking\networking.c:86:9: warning: unknown option after '#pragma GCC diagnostic' kind [-Wpragmas]
#pragma GCC diagnostic ignored "-Wstringop-overflow"

Ignorer denne advarselen, den er for å unngå en annen - noen compilere er mer trøblete en andre...

distebia, 2023-10-16:
> Ignorer denne advarselen, den er for å unngå en annen - noen compilere er mer trøblete enn andre...

mi è successo la stessa cosa attivando, in fase di compilazione, anche la scheda sd e webui. l'errore era sempre di colore arancione, non rosso. stasera ricompilo con queste impostazioni e posto i risultati

distebia, 2023-10-16:
ho attivato WEBUI, ETHERNET e SDCARD e ricevo questo risultato:
[image]

[image]

quindi non sono errori?

terjeio, 2023-10-16:
Vel, det er bare kompilatoren / konfiguasjonen for iMRX1062 som produserer disse advarslene. Jeg har valgt å ikke bruke tid på å omgå de - kanskje senere? Koden kjører uten problemer.
For info: jeg bruker Arduino IDE for å kompilere, kompilatoropsjoner kan ikke settes fra dette primitive brukergrensensittet...

distebia, 2023-10-16:
Quindi non sono errori...seguirò il tuo consiglio per ompilare con arduino..prima di platformio ho provato con arduino ma avevo anche là errori..passo ad arduino e vediamo cosa succede, grazie

distebia, 2023-10-16:
con arduino ricevo questo, stessa configurazione

c:\Users\Biagio\Documents\Arduino\libraries\grblHAL_Teensy4\src\grbl\system.c: In function 'system_execute_line':
c:\Users\Biagio\Documents\Arduino\libraries\grblHAL_Teensy4\src\grbl\system.c:419:23: warning: writing 1 byte into a region of size 0 [-Wstringop-overflow=]
  419 |             *(--args) = '=';
      |             ~~~~~~~~~~^~~~~
c:\Users\Biagio\Documents\Arduino\libraries\grblHAL_Teensy4\src\grbl\system.c:360:42: note: at offset 1 into destination object 'line' of size [0, 2147483647]
  360 | status_code_t system_execute_line (char *line)
      |                                    ~~~~~~^~~~
c:\Users\Biagio\Documents\Arduino\libraries\grblHAL_Teensy4\src\grbl\system.c:424:21: warning: writing 1 byte into a region of size 0 [-Wstringop-overflow=]
  424 |             *args++ = '\0';
      |             ~~~~~~~~^~~~~~
c:\Users\Biagio\Documents\Arduino\libraries\grblHAL_Teensy4\src\grbl\system.c:360:42: note: at offset 1 into destination object 'line' of size [0, 2147483647]
  360 | status_code_t system_execute_line (char *line)
      |                                    ~~~~~~^~~~
c:\Users\Biagio\Documents\Arduino\libraries\grblHAL_Teensy4\src\networking\multipartparser.c: In function 'multipartparser_init':
c:\Users\Biagio\Documents\Arduino\libraries\grblHAL_Teensy4\src\networking\multipartparser.c:122:5: warning: 'strncpy' specified bound 70 equals destination size [-Wstringop-truncation]
  122 |     strncpy(parser->boundary, boundary, sizeof(parser->boundary));
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\Biagio\Documents\Arduino\libraries\grblHAL_Teensy4\src\networking\networking.c: In function 'networking_get_services_list':
c:\Users\Biagio\Documents\Arduino\libraries\grblHAL_Teensy4\src\networking\networking.c:96:13: warning: 'strncat' output truncated copying 3 bytes from a string of length 4 [-Wstringop-truncation]
   96 |             strncat(list, service_names[idx], strlen(service_names[idx]) - 1);
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\Biagio\Documents\Arduino\libraries\grblHAL_Teensy4\src\sdcard\fs_littlefs.c:185:14: warning: 'fs_getcwd' defined but not used [-Wunused-function]
  185 | static char *fs_getcwd (char *buf, size_t size)
      |              ^~~~~~~~~
c:\Users\Biagio\Documents\Arduino\libraries\grblHAL_Teensy4\src\webui\args.c: In function 'webui_get_bool_arg':
c:\Users\Biagio\Documents\Arduino\libraries\grblHAL_Teensy4\src\webui\args.c:56:12: warning: 'strncpy' specified bound depends on the length of the source argument [-Wstringop-truncation]
   56 |         if(webui_get_arg(argc, argv, strncpy(tmp, arg, strlen(arg) - 1)))
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\Biagio\Documents\Arduino\libraries\grblHAL_Teensy4\src\webui\args.c:56:56: note: length computed here
   56 |         if(webui_get_arg(argc, argv, strncpy(tmp, arg, strlen(arg) - 1)))
      |                                                        ^~~~~~~~~~~
c:\Users\Biagio\Documents\Arduino\libraries\uSDFS\src\utility\sd_sdhc.c: In function 'sd_isr':
c:\Users\Biagio\Documents\Arduino\libraries\uSDFS\src\utility\sd_sdhc.c:332:3: warning: this 'while' clause does not guard... [-Wmisleading-indentation]
  332 |   while(!(SDHC_IRQSTAT & SDHC_IRQSTAT_TC));//  SDHC_IRQSTAT &= ~SDHC_IRQSTAT_TC;
      |   ^~~~~
In file included from c:\Users\Biagio\Documents\Arduino\libraries\uSDFS\src\utility\sd_sdhc.c:51:
c:\Users\Biagio\Documents\Arduino\libraries\uSDFS\src\utility\sdio_priv.h:363:35: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the 'while'
  363 |         #define SDHC_MIX_CTRL     (USDHC1_MIX_CTRL) // Mixer Control
      |                                   ^
c:\Users\Biagio\Documents\Arduino\libraries\uSDFS\src\utility\sd_sdhc.c:335:5: note: in expansion of macro 'SDHC_MIX_CTRL'
  335 |     SDHC_MIX_CTRL &= ~(SDHC_MIX_CTRL_AC23EN | SDHC_MIX_CTRL_DMAEN) ;
      |     ^~~~~~~~~~~~~
c:\Users\Biagio\Documents\Arduino\libraries\uSDFS\src\utility\sd_sdhc.c: In function 'sd_CMD13_WaitForReady':
c:\Users\Biagio\Documents\Arduino\libraries\uSDFS\src\utility\sd_sdhc.c:881:5: warning: this 'while' clause does not guard... [-Wmisleading-indentation]
  881 |   { while ((SDHC_PRSSTAT & SDHC_PRSSTAT_CIHB) || (SDHC_PRSSTAT & SDHC_PRSSTAT_CDIHB)) ;
      |     ^~~~~
In file included from c:\Users\Biagio\Documents\Arduino\libraries\uSDFS\src\utility\sd_sdhc.c:51:
c:\Users\Biagio\Documents\Arduino\libraries\uSDFS\src\utility\sdio_priv.h:358:37: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the 'while'
  358 |         #define SDHC_IRQSTATEN      (USDHC1_INT_STATUS_EN) // Interrupt Status Enable register
      |                                     ^
c:\Users\Biagio\Documents\Arduino\libraries\uSDFS\src\utility\sd_sdhc.c:882:5: note: in expansion of macro 'SDHC_IRQSTATEN'
  882 |     SDHC_IRQSTATEN |= SDHC_IRQSTATEN_CCSEN;
      |     ^~~~~~~~~~~~~~
Opening Teensy Loader...
Memory Usage on Teensy 4.1:
  FLASH: code:297604, data:100932, headers:9012   free for files:7718916
   RAM1: variables:180288, code:294968, padding:32712   free for local variables:16320
   RAM2: variables:12416  free for malloc/new:511872