grblHAL/iMXRT1062 #71: valore ottimale planner $398

distebia, 2023-11-04:
Salve, ho provato ad aumentare il planner del pianificatore a $398=1000 e un'immagine in scala di grigi eseguita a 40000/min veniva eseguita ad una velocità maggiore (molto maggiore) rispetto a 40000/min e $398=35, ma ottengo un'immagine (in scala di grigi) molto confusa, come se il laser rimanesse acceso più a lungo e sincronizzato (impulso e passi) male. Utilizzo anche il plugin LB cluster. Con quali altri parametri potrei giocare per avere un'immagine migliore? utilizzo teensy4.1 + T41U5XBB

terjeio, 2023-11-05:
Dynamic response of laser do not match pixel rate (feedrate)?

Things to try in no particular order:

* Do not use LB clusters, may introduce loss of positional precision. grblHAL (and all Grbl ports I am aware of) uses single precision float numbers internally that can cause errors to accumulate when relative motion is commanded.
* Reduce feedrate.
* Increase PWM frequency.
* Maxing out $398 to the max allowable value may not always be optimal.

distebia, 2023-11-05:
esattamente quello che pensavo...avevo notato un x2 (ad occhio) impostando 40000 di velocità e $398=1000, rispetto a $398=35, ma perdevo di qualità e ho subito sospettato una non sincronizzazione nel movimento, ho anche giocato con $0=2 (da $0=4) ma i tb6600 dichiarano il val.min 2.3 (ottenevo immagini fantasma sdoppiati). Per il co2 consigliano 5000 hz (l'ho impostato a 10000). Tutto questo per cercare di ottenere delle incisioni a co2 di immagini in toni di grigio ad alta velocità, mi mantengo intorno ai 20000 e penso di non andare oltre

terjeio, 2023-11-08:
RF excited CO2 laser can be, IIRC, 10x+ faster (response time) than DC excited lasers. Downside is that they are a _lot_ more expensive...

distebia, 2023-11-24:
Ho fatto alcune prove e sembra che disabilitando il LB cluster e aumentando il $398 a 100 ottengo un'incisione a scala di grigi in meno tempo alla stessa velocità impostata. Potrebbe venire in aiuto il plugin PPI laser? Se si come lo configuro? utilizzo una T41U5XBB con teensy 4.1, grazie