grblHAL/iMXRT1062 #74: COOLANT_MIST_PIN changing from pin 24 to any other available pin breaks communication with board after flashing successfully.

karoria, 2023-12-14:
I am having a weird issue where in my customized board map, when I change pin 24 to other than COOLANT_MIST_PIN or even disabling COOLANT_MIST_PIN breaks USB as well as Ethernet communication after successfully compiling and flashing to Teensy 4.1. Any clues?

terjeio, 2023-12-14:
Oops, guards for the mist pin not beeing defined is missing, change [this code](https://github.com/grblHAL/iMXRT1062/blob/0d12cde1886e25bebae0b1ca56fa35e3e2495766/grblHAL_Teensy4/src/driver.c#L1492-L1514) to:

```
// Start/stop coolant (and mist if enabled).

// coolant_state_t is defined in grbl/coolant_control.h.
static void coolantSetState (coolant_state_t mode)
{
    mode.value ^= settings.coolant_invert.mask;

    DIGITAL_OUT(Flood, mode.flood);
#ifdef COOLANT_MIST_PIN
    DIGITAL_OUT(Mist, mode.mist);
#endif
}

// Returns coolant state in a coolant_state_t variable.
// coolant_state_t is defined in grbl/coolant_control.h.
static coolant_state_t coolantGetState (void)
{
    coolant_state_t state = { settings.coolant_invert.mask };

    state.flood = (Flood.reg->DR & Flood.bit) != 0;
#ifdef COOLANT_MIST_PIN
    state.mist = (Mist.reg->DR & Mist.bit) != 0;
#endif
    state.value ^= settings.coolant_invert.mask;

    return state;
}
```

Does this make it work?

karoria, 2023-12-17:
Thanks. I am out of station for some days. When I check, will inform you here.

karoria, 2023-12-27:
>Does this make it work?

That works @terjeio. Thanks! Please merge it in next commit.

terjeio, 2023-12-28:
It is [already merged](https://github.com/grblHAL/iMXRT1062/commit/41b1ce3a50f976cae5446632e91edb0e968ad0a2).

closed 2023-12-28 by terjeio