grblHAL/iMXRT1062 #76: Hello and I have Been Known to Misinterpret Ideas

silver2row, 2024-01-07:
Hello,

I know this is silly. I have a grblHAL board with Teensy4.1 attached and everything soldered in via the instructions...

1. I was moving the motors the other day by hand
2. I was receiving negative feedback on the controller
3. Is reverse polarity due to manual motor movement not collective with the grblHAL board and Teensy4.1 attachment?

I am asking only because I am terrible with the controls so far. I am still learning slowly. 

So, this is what happens. 

I move the axis, X and/or Y, manually and I can see the light on the board light up. Should I take the motors and detach them before handling manual motor movement? 

I know it is silly to ask but some reverse polarity protection would be okay with me. 

Seth

P.S. If this is uncalled for currently and the build does not require it, okay. I understand and I will fully agree to unplug things before handling manual movement. It just spooked me! So, I thought I would bring it up. Just a heads up for you or anyone really who likes this boards, like me, and who uses them to alter machinery. 

```
If there is reverse feedback control on the board itself so things do not get fried a bit, okay. 
```

I must have missed something while soldering the board and the attachments. Anyway, keep up the good work and do not quit. 

I know what you do works. I am not saying it does not work. I fired the laser the other day. It works! And for that, I am thankful.

I am still practicing the cartesian build with my cheap-o, depot machine I picked up for $99.99. So, my dilemma may be the motors or the grounding of the entire system. I will try to keep everyone updated once I configure things. This way, I can be of some assistance when you guys keep going!

terjeio, 2024-01-07:
> I move the axis, X and/or Y, manually and I can see the light on the board light up. Should I take the motors and detach them before handling manual motor movement?

The motors acts as generators when spun manually so yes it is best to disconnect them first unless the [stepper drivers](https://components101.com/modules/stepstick-protector) has backfeed protection.

silver2row, 2024-01-07:
Yes. Okay. I just wanted to check. My drivers are obviously a bit cheaper or at least they do no cover the back feedback protection aspect. Thank you.

Seth

P.S. I am goin' to be quiet now. Thank you again.

silver2row, 2024-01-07:
Once last thing...

On the hex page online to install the hex code for the imx (Teensy4.1), do I pick the grblHAL2000 or [T41U5XBB](https://www.grbl.org/grblhal-breakout-board)?

Seth

closed 2024-01-07 by silver2row

terjeio, 2024-01-07:
You select the board that matches the one you have. Or if you are using a "bare" Teensy select the board that provides the features you want. The board map contains the pin mappings.