grblHAL/iMXRT1062 #81: calo improvviso emissione fascio laser / pwm

distebia, 2024-02-10:
Salve, oggi stavo facendo una lavorazione su 3 livelli colore di lightburn, 1° livello foto modalità stucki, 2° riempimento di contorni e 3° taglio. Il problema si presenta dopo aver fatto perfettamente la foto, finisce la foto e passa al riempimento. Effettua si e no 20 linee di riempimento, impostato a 8000mm/min 20% 0.08 linee per millimetro, e all'improvviso mi accorgo che il laser si muove ma non ho emissione di fascio, l'amperometro da 0-30mA segna forse 2 e nel plexiglass si vede che il riempimento non c'è. Blocco tutto, rifaccio l'homing, seleziono soltanto 2° livello-riempimento e 3° livello-taglio e porto a termine il lavoro. Utilizzo teensy4.1+T41U5XBB. Non riesco a capire da cosa può dipendere tutto questo, la macchina funzionava prima e funzionava anche dopo, è come se il controller abbassa, ma non azzera, l'uscita pwm, dopo l'homing (senza toccare altro, tipo spegnere e rieccendere ecc...) tutto funziona come prima. Anche la temperatura era nel range valido e segnava 13.4 °C

terjeio, 2024-02-10:
Text exceeds 1000 character limit - english please!

distebia, 2024-02-10:
Hi, today I was working on 3 lightburn color layers, 1st sticki mode photo layer, 2nd outline filling and 3rd cut. The problem occurs after you have taken the photo perfectly, finish the photo and move on to filling. I carry out approximately 20 filling lines, set at 8000mm/min 20% 0.08 lines per millimetre, and suddenly I realize that the laser is moving but I have no beam emission, the 0-30mA ammeter reads perhaps 2 and in the plexiglass you can see that the filling is not there. I block everything, redo the homing, select only the 2nd fill level and the 3rd cut level and complete the job. Usingteensy4.1+T41U5XBB. I can't understand what all this could depend on, the machine worked before and also worked after, it's as if the controller lowers, but doesn't reset, the PWM output, after homing (without touching anything else, like turning it off and on again etc...) everything works as before. The temperature was also within the valid range and marked 13.4 °C

terjeio, 2024-02-11:
Which version ($I output)?
Which sender?
Is the 2nd layer in a separate file? Can you post it?

distebia, 2024-02-11:
[image]

utilizzo Lightburn e il lavoro è tutto su unico file che allego

[image]

[problema.zip](https://github.com/grblHAL/iMXRT1062/files/14233667/problema.zip)

terjeio, 2024-02-12:
I am not able to replicate this, and if I understand you correctly you cannot either?

distebia, 2024-02-12:
no, dopo che ho finito il lavoro mi sono accorto che il plexiglass aveva un graffio sotto la pellicola, ho rifatto il lavoro e non ho avuto problemi.