grblHAL/iMXRT1062 #82: Not an issue: I want to know MODBUS implementation.

karoria, 2024-03-12:
Hi @terjeio 
I want to implement MODBUS communication for my delta MS300 VFD. I want to know:
1. Can Modbus TCP work simultaneously when ethernet is main communication channel? I can connect teensy to network router for connecting different devices.
2. If not option 1, how to enable modbus RTU over serial communication to use GPIO 0 & 1 on teensy 4.1?
3. Any other points to keep in mind to setup in VFD and grblHAL.
Thanks in advance,
Ravi

karoria, 2024-03-21:
Hi @terjeio 
I have tried with different enable options for modbus spindles but no success. My latest spindle enabled is modvfd type. I set parameters 460 to 467 as per delta ms300 manual and found it is similar with GS20 vfd which is default in modvfd settings. But upon sending m3s2000 or so, it gives alarm14 spindle at speed timeout. Where I am wrong?

terjeio, 2024-03-25:
> Any method where I can check whether my board really send signals?

Does the RS485 breakout have Rx/Tx LEDs? If so you can se them blinking when communication is working.

karoria, 2024-03-25:
My board doesn't have leds for tx and rx pins but I can use dso if that helps. Should I use dso between tx and rx or tx with gnd and rx with gnd separately?

terjeio, 2024-03-26:
> Should I use dso between tx and rx or tx with gnd and rx with gnd separately?

tx with gnd and rx with gnd separately.

karoria, 2024-03-27:
> tx with gnd and rx with gnd separately.

Ok. I can see square waves coming from tx pin which is gpio-1 of teensy4.1. No movement in rx which should be obvious as no device is connected for replying back. Once I connect the vfd, rx also detected square waves in sync with  tx which looks like mirror of tx. But vfd doesn't get enabled or frquency change upon sending m3s3000 or so. I can see some quick movements in waves for a while when it gives alarm 14.

karoria, 2024-03-27:
Ok. A progress. I noticed now vfd goes in Run mode with M3 and stops with M5. It also goes in run mode with Rev with M4 command. But it doesnt change its rpm and generates alarm14 on each command. I have to clear alarm by unlock and then send m3 or m5 to make it effect.

terjeio, 2024-03-27:
The alarm is likely due to exceptions posted by the modbus code, if you comment out the `system_raise_alarm()` calls in the [exception handler](https://github.com/grblHAL/Plugins_spindle/blob/9c949fb50521d7a99fd8e262b9a49a34f406731f/vfd/modvfd.c#L247-L283) is the alarm going away?

Exceptions can be generated by the VFD, from timeout while waiting for a reply or a CRC error in the reply. 

Tip: when debugging use a terminal to communicate with the controller, not a sender - this since realtime report requests query data from the VFD which makes it harder to see what is going on.

karoria, 2024-03-29:
Thanks @terjeio for support. Vfd driver code for gs20 worked out of the box for delta ms300. It is working nicely.
BTW, stepper spindle code is working now? I want to try it in another machine.

terjeio, 2024-03-30:
> BTW, stepper spindle code is working now?

I think so, I have not tested it extensively. Give it a try!

karoria, 2024-03-31:
> I think so, I have not tested it extensively. Give it a try!

Just to be sure, do I need to configure VFD in position control mode instead of speed control if I use stepper spindle?

terjeio, 2024-03-31:
The VFD spindle has nothing to do with the stepper spindle so I do not understand what you mean. Currently the stepper spindle code "claims" the highest numbered axis motor to use it as a spindle. You may enable both a VFD and the stepper spindle and switch between them with the M104 command.

karoria, 2024-03-31:
I mean, will it run with simple step and direction control? How about acceleration/deceleration? If my B axis is the highest motor, then stepper spindle code will claim B axis as stepper spindle and apply acceleratiin settings of B axis to that spindle, right?

terjeio, 2024-03-31:
> will it run with simple step and direction control?

Yes.

> How about acceleration/deceleration?

Supported.

> If my B axis is the highest motor, then stepper spindle code will claim B axis as stepper spindle and apply acceleratiin settings of B axis to that spindle, right?

Yes, except the [setting units](https://github.com/grblHAL/core/blob/5afac36157e4dfd3043a7d15493224887ac5148a/stepper2.c#L136-L146) will be changed so the values will have to match those.

karoria, 2024-04-01:
Thanks for quick support.

karoria, 2024-04-02:
For SPINDLE_STEPPER, 
- I am not able to define spindle enable pin. Neither by M4_ENABLE_PIN nor by SPINDLE_ENABLE_PIN. Am I making any obvious mistake?
- Direction pin doesn't change its output to low or high. Instead, issuing M4S2000 command, it removes spindle from accessory state while rpm is conveyed in staus report.

terjeio, 2024-04-02:
> I am not able to define spindle enable pin. Neither by M4_ENABLE_PIN nor by SPINDLE_ENABLE_PIN. Am I making any obvious mistake?

No, I kind of made one as the basic code was for plasma THC support that is only active when there is motion. I have now added enable support to the higher level stepper spindle code that should work.
Note that the axis motor used for the stepper spindle still shows up in senders and can be used by motion commands. It is on my todo list to find a way to remove the axis from reports, or at least blocking motion commands for it.
And be aware that the stepper spindle code is flagged as experimental in the source.

> Direction pin doesn't change its output to low or high. Instead, issuing M4S2000 command, it removes spindle from accessory state while rpm is conveyed in staus report.

There were typos in the iMXRT1062 driver code, fixed now.

karoria, 2024-04-03:
Thanks for quick support. How to define enable pin now? By M4 enable or by Spindle Enable?

terjeio, 2024-04-03:
M4 enable.

karoria, 2024-04-03:
Thanks @terjeio. It works. But Enable behaviour is invert than what I need. It is active low, I need active High. How can I invert  enable behaviour?

terjeio, 2024-04-03:
It uses $4 for inverting the enable signals.

karoria, 2024-04-04:
That worked!