grblHAL/iMXRT1062 #85: Compilation problem after I formatted my macbook

karoria, 2024-04-25:
I am facing compilation issues as under:

Resolving teensy41 dependencies...
Platform Manager: Installing teensy @ 4.16
UnknownPackageError: Could not find the package with 'teensy @ 4.16' requirements for your system 'darwin_x86_64'

If I compile with teensy @4.17 or teensy @5.0 I get following SD error:

compilation terminated.
*** [.pio/build/teensy41/libd02/USBHost_t36/MassStorageDriver.cpp.o] Error 1
In file included from /Users/karoria/.platformio/packages/framework-arduinoteensy@1.157.220801/libraries/USBHost_t36/USBFilesystemFormatter.h:4:0,
                 from /Users/karoria/.platformio/packages/framework-arduinoteensy@1.157.220801/libraries/USBHost_t36/USBFilesystemFormatter.cpp:2:
/Users/karoria/.platformio/packages/framework-arduinoteensy@1.157.220801/libraries/USBHost_t36/USBHost_t36.h:2052:19: fatal error: SdFat.h: No such file or directory

terjeio, 2024-04-25:
You have custom code that uses the USBHost_t36 library?
I cannot find any references in the grblHAL code nor at the [PRJC github](https://github.com/search?q=repo%3APaulStoffregen%2Fcores%20USBHost_t36.h&type=code).

karoria, 2024-04-25:
No any custom code. I just cloned iMXRT repo and tried to compile it.

terjeio, 2024-04-25:
A platformio bug then? If you compile without WebUI/SD card support does the error go away?

karoria, 2024-04-29:
I have raised issue at pio. Got following response. It seems they have removed old version from server. Can we use latest version 5.0.0?

terjeio, 2024-04-29:
> Can we use latest version 5.0.0?

Possibly. But have to resolve the include error - some code is missing but not from grblHAL.
Again, if you compile without WebUI/SD card support does the error go away?

karoria, 2024-04-29:
Ok. Will check that and revert. Sorry I missed your instruction.

karoria, 2024-04-30:
Giving url link as under in platformio.ini file installs teensy version 4.16 and compiles successfully:

Instead of:
platform = ${common_teensy.platform}@4.16 

Change with:
platform = https://github.com/platformio/platform-teensy.git#v4.16.0

I can confirm that I had tried 4.18 and 5.0 versions with "just cloned" repo without anything changed in my_machine.h or anywhere else (that means webui and sd disabled) and they both versions throw above compilation errors.