grblHAL/iMXRT1062 #87: Single Block switch ignored

JasonTitcomb, 2024-10-22:
Great product. Really enjoying it.

Running the T41U5XBB board flashed from the web builder and ioSender .44
Start, Hold and Single Block buttons are hard wired.
I was testing a short program.
First I single block through the entire program and it stops at the end.
If I click the Reset button in ioSender at that point and cycle start again, it ignores that my single block switch is on.
Seems like I need to toggle it on and off before I start if _ do a reset.
[settings.txt](https://github.com/user-attachments/files/17482471/settings.txt)
[Pins.txt](https://github.com/user-attachments/files/17482472/Pins.txt)

terjeio, 2024-10-26:
This has to wait until I am back home and have access to a board.
FYI when I tested with the STM32F7xx it works, and the code is basically the same...

JasonTitcomb, 2024-11-04:
I created a pull request.

terjeio, 2024-12-11:
[PR](https://github.com/grblHAL/iMXRT1062/pull/88) already merged.

closed 2024-12-11 by terjeio