grblHAL/iMXRT1062 #94: No Motion control on iMXRT1062/T41U5XBB with April 5, 2025 build

rschell, 2025-04-06:
Built the April 5th version on web builder with my normal settings (4-axis, PWM spindle, full Ethernet, TLO-Z reporting, NGC expression support).  When I load the firmware via USB to the teensy4 and reconnect bCNC via ethernet to the controller, attempts to home or move result in no motion.

Prior March 28th, 2025 version works fine.  Any changes to the board configuration following April 5th commits?  Didn't build the March 29th commit, so I can't tell which of the latest commits is causing the issue.

Happy to do more testing, no sure which method to pursue.

terjeio, 2025-04-06:
I've messed up step timing, should be fixed now.

rschell, 2025-04-06:
Works now,  Thanks