grblHAL/iMXRT1062 #96: Problem new plugin to make the cnc works with dual x carrige like Idex

Achraf-Jannen, 2025-05-01:
I want to make the cnc works with dual x carrige like Idex So i tried to create a new plugin named Idex_plugin (two files .c and .h)

what are the steps to create a new plugin and this the code i created (got many errors )

Idex_plugin.c :
// Plugins/idex_plugin/idex_plugin.c
#include "driver.h"
#include "idex_plugin.h"
#include "grbl/stepper.h"
#include "grbl/gcode.h"
#include "grbl/system.h"
#include"grbl/nuts_bolts.h"

// IDEX Mode Enum
typedef enum {
    IDEX_MODE_FULL_CONTROL = 0,
    IDEX_MODE_AUTO_PARK,
    IDEX_MODE_DUPLICATION,
    IDEX_MODE_MIRRORED
} idex_mode_t;

// Global state
static idex_mode_t idex_mode = IDEX_MODE_FULL_CONTROL;
static uint8_t active_tool = 0; // 0 = X1 (X), 1 = X2 (A)
static float duplication_offset = 100.0f;
static float park_pos_x1 = 0.0f;
static float park_pos_x2 = 350.0f;

// Axis indices
#define AXIS_X 0
#define AXIS_A 3

extern const gpio_t enable_pins[];

static void switch_tool(uint8_t tool)
{
    active_tool = tool;
    if(tool == 0) {
        gpio_set(&enable_pins[AXIS_X]);     // Enable X
        gpio_clear(&enable_pins[AXIS_A]);   // Disable A
        report_message("Switched to X1 (T0)");
    } else {
        gpio_clear(&enable_pins[AXIS_X]);   // Disable X
        gpio_set(&enable_pins[AXIS_A]);     // Enable A
        report_message("Switched to X2 (T1)");
    }

    if(idex_mode == IDEX_MODE_AUTO_PARK) {
        if(tool == 0) {
            gc_sync_position_axis('A', park_pos_x2);
        } else {
            gc_sync_position_axis('X', park_pos_x1);
        }
    }
}

static status_code_t idex_mcode_handler(uint_fast16_t code, char *params)
{
    if(code == 605) {
        uint8_t mode;
        if(sscanf(params, "S%hhu", &mode) == 1) {
            if(mode <= 3) {
                idex_mode = (idex_mode_t)mode;
                char msg[40];
                sprintf(msg, "IDEX Mode set to %u", mode);
                report_message(msg);
                return Status_OK;
            } else {
                return Status_InvalidStatement;
            }
        }
    }
    return Status_Unhandled;
}

static status_code_t idex_tcode_handler(uint_fast16_t code)
{
    if(code == 0 || code == 1) {
        switch_tool((uint8_t)code);
        return Status_OK;
    }
    return Status_Unhandled;
}

void idex_plugin_init(void)
{
    gc_register_mcode_handler(605, idex_mcode_handler);
    gc_register_tool_select_handler(idex_tcode_handler);

    report_message("IDEX Plugin Initialized");
}

terjeio, 2025-05-01:
The code above is not for grblHAL, did you use AI to generate it?

Achraf-Jannen, 2025-05-01:
yes

Achraf-Jannen, 2025-05-01:
i am trying to use idex property from marlin. Is it possible ?

terjeio, 2025-05-01:
> i am trying to use idex property from marlin. Is it possible ?

I do not know what idex is, do you have a link?

Achraf-Jannen, 2025-05-01:
this is the link 
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/Configuration_adv.h
and this is the part : 
// @section idex

/**
 * Dual X Carriage
 *
 * This setup has two X carriages that can move independently, each with its own hotend.
 * The carriages can be used to print an object with two colors or materials, or in
 * "duplication mode" it can print two identical or X-mirrored objects simultaneously.
 * The inactive carriage is parked automatically to prevent oozing.
 * X1 is the left carriage, X2 the right. They park and home at opposite ends of the X axis.
 * By default the X2 stepper is assigned to the first unused E plug on the board.
 *
 * The following Dual X Carriage modes can be selected with M605 S<mode>:
 *
 *   0 : (FULL_CONTROL) The slicer has full control over both X-carriages and can achieve optimal travel
 *       results as long as it supports dual X-carriages. (M605 S0)
 *
 *   1 : (AUTO_PARK) The firmware automatically parks and unparks the X-carriages on tool-change so
 *       that additional slicer support is not required. (M605 S1)
 *
 *   2 : (DUPLICATION) The firmware moves the second X-carriage and extruder in synchronization with
 *       the first X-carriage and extruder, to print 2 copies of the same object at the same time.
 *       Set the constant X-offset and temperature differential with M605 S2 X[offs] R[deg] and
 *       follow with M605 S2 to initiate duplicated movement.
 *
 *   3 : (MIRRORED) Formbot/Vivedino-inspired mirrored mode in which the second extruder duplicates
 *       the movement of the first except the second extruder is reversed in the X axis.
 *       Set the initial X offset and temperature differential with M605 S2 X[offs] R[deg] and
 *       follow with M605 S3 to initiate mirrored movement.
 */
//#define DUAL_X_CARRIAGE
#if ENABLED(DUAL_X_CARRIAGE)
  #define X1_MIN_POS X_MIN_POS    // Set to X_MIN_POS
  #define X1_MAX_POS X_BED_SIZE   // A max coordinate so the X1 carriage can't hit the parked X2 carriage
  #define X2_MIN_POS    80        // A min coordinate so the X2 carriage can't hit the parked X1 carriage
  #define X2_MAX_POS   353        // The max position of the X2 carriage, typically also the home position
  #define X2_HOME_POS X2_MAX_POS  // Default X2 home position. Set to X2_MAX_POS.
                                  // NOTE: For Dual X Carriage use M218 T1 Xn to override the X2_HOME_POS.
                                  // This allows recalibration of endstops distance without a rebuild.
                                  // Remember to set the second extruder's X-offset to 0 in your slicer.

  // This is the default power-up mode which can be changed later using M605 S<mode>.
  #define DEFAULT_DUAL_X_CARRIAGE_MODE DXC_AUTO_PARK_MODE

  // Default x offset in duplication mode (typically set to half print bed width)
  #define DEFAULT_DUPLICATION_X_OFFSET 100

  // Default action to execute following M605 mode change commands. Typically G28X to apply new mode.
  //#define EVENT_GCODE_IDEX_AFTER_MODECHANGE "G28X"
#endif

terjeio, 2025-05-01:
Ok. You can hijack an axis with plugin code, e.g. A, as a second X but with some limitiations. A X2 designated motor used for ganged motion can _not_ be controlled independently - at least not without driver changes.

S0 mode may not be possible without core changes. I would like to see the gcode used to control two independent X axes simultaneously... One at a time should be easy, see S2 and S3 motion below. 

S1 mode - perhaps. Homing may be an issue. Or not - just home A in the opposite direction of X?. Toolchanges might be tricky.

S2 and S3 modes - IMO can be done.

For S2 and S3 motion [intercept](https://github.com/grblHAL/Plugin_odometer/blob/1b72cfa883c514e006128ae43335107ab7b8c6e0/odometer.c#L269-L270) the `hal.stepper.pulse_start` call and [copy step and dir bits](https://github.com/grblHAL/core/blob/390c11bb7ef58d8c8d32e936f01b85d9709a0d84/stepper.h#L87-L88) from X to A, inverting dir in S3 mode before passing it on to the driver.

Trapping M605 [in a plugin](https://github.com/grblHAL/Templates/blob/master/mcodes.c) is easy - but implementing the functionality might not be?

If you want or need to you can [trap stepper enable](https://github.com/grblHAL/core/blob/390c11bb7ef58d8c8d32e936f01b85d9709a0d84/stepper.c#L186) too and modify which steppers gets enabled.

FAB619, 2025-07-22:
Hi, Thanks a lot for your detailed reply and for pointing out the constraints.  
Your notes about hijacking an axis (e.g. A as X2) and intercepting `hal.stepper.pulse_start` to copy/invert step/dir bits are very helpful.  

Based on what you explained, my plan is:  
- Start by implementing only **S2 (duplication)** and **S3 (mirrored)** modes because they seem doable without deep core changes.  
- Later explore S0 and S1 which would need extra logic for toolchange and homing.  

I’m thinking of writing a plugin that:  
- Hooks into motion generation to duplicate X pulses to A (with optional inversion for S3).  
- Adds a small parser hook for `M605 S0/S1/S2/S3` to switch modes at runtime.

**A few questions:**  
- What’s the best place in GRBL HAL to trap and handle custom M‑codes like `M605`?  
- Are there existing plugins where `hal.stepper.pulse_start` is intercepted that I could use as reference?  
- Do you foresee any issues with planner timing or stepper enable handling when duplicating signals?

Any advice or pointers would be great. Thanks again for your time and for maintaining GRBL HAL — I’d be happy to share progress or even a draft skeleton once I get started.

FAB619, 2025-07-22:
As requested, here’s a sample G‑code sequence we currently use on a Marlin IDEX setup to test duplication and mirrored modes.  
This shows the expected workflow and commands we’d like to replicate on GRBL HAL:

/*;;;Start Gcode;;;
G28                  ; Home all axes
M605 S2              ; Enable DUPLICATION mode
M605 S3              ; Switch to MIRRORED mode (in marlin it works only after sending M605 S2 for version 2.1.x)
G21                  ; Set units to millimeters
G91                  ; Relative positioning
G1 X50.751 Y-0.08 Z-20 F3000
G1 Z-12.5 F3000      ; Additional Z move
M106                 ; Fan on
;;;Start Prog;;;
G1 X-50.751
G1 X-2.901 Y0.019 F720
G1 X-2.9 Y0.061
G1 X-4.948
G1 X-1.178 Y-0.031
G1 X-1.162 Y-0.088
G1 X-1.145 Y-0.146
G1 X-1.127 Y-0.201
G1 X-1.107 Y-0.255
G1 X-1.086 Y-0.309
G1 X-1.064 Y-0.36
G1 X-1.039 Y-0.409
G1 X-1.013 Y-0.459
G1 X-0.987 Y-0.505
G1 X-0.959 Y-0.551
G1 X-0.928 Y-0.595
G1 X-0.897 Y-0.638
G1 X-0.865 Y-0.68
G1 X-0.83 Y-0.719
G1 X-0.795 Y-0.758
G1 X-0.758 Y-0.795
.........
.........
.........
;;;End Gcode;;;
M107                 ; Fan off
G1 Z12.5 F3000       ; Raise Z
G1 X20               ; Move X
M84                  ; Disable motors *\

Let me know if you’d like additional samples (e.g., toolchange sequences or homing routines) or a draft of how we plan to hook this into the GRBL HAL workflow.

terjeio, 2025-07-23:
> What’s the best place in GRBL HAL to trap and handle custom M‑codes like M605?

In the [plugin init function](https://github.com/grblHAL/Templates/blob/5385b30376d72ecdaceb3b3b1fff0e720b6a6cd0/my_plugin/Stepper_enable_control/my_plugin.c#L207-L211). Name it `my_plugin_init()` initially - your version will override a weak implementation when compiling.

> Are there existing plugins where hal.stepper.pulse_start is intercepted that I could use as reference?

Yes, [I linked to one](https://github.com/grblHAL/Plugin_odometer/blob/1b72cfa883c514e006128ae43335107ab7b8c6e0/odometer.c#L269-L270) above.

> Do you foresee any issues with planner timing or stepper enable handling when duplicating signals?

Planner timing no, stepper enable depends on setting $37 - with current defaults no. The stepper enable function can also be [intercepted](https://github.com/grblHAL/Templates/blob/5385b30376d72ecdaceb3b3b1fff0e720b6a6cd0/my_plugin/Stepper_enable_control/my_plugin.c#L213-L214).