grblHAL/iMXRT1062 #98: Spindle encoder resolution

LaurentLM, 2025-07-08:
Hi,

The spindle speed reports on my lathe seems to fluctuate more than physically possible. I thought a lower resolution encoder would average errors a little bit better and changed from a 600 PPR to a 100 PPR encoder.
Both encoders are optical encoders and output clean signals (checked with the oscilloscope).

I found that with with setting `$38  Spindle pulses per revolution (PPR)` set to `100`, I only get 4 reports of spindle RPM asp part of serial status messages, and then they stop. 

When stopping and starting the spindle again, I occasionally get 4 more RPM reports. 

Changing the PPR seems to have an impact on how many RP reports I get, but it seems that for `$38` set to anything below 170 PPR, the RPM reports become unstable and fail after a few status reports. 

Using a teensy 4.1 on a T41U5XBB Spindle sync board.
Firmware freshly flashed 20250702 (but the issue was already present before the firmware update).

I think the question has been asked before, but without a clear answer. What would a sensible encoder resolution be, taking into account threading accuracy, and interrupt load. I appreciate there wont be a "one answer fits all", but an idea of range would be useful.

terjeio, 2025-07-11:
What is your $340 setting (at speed tolerance)?
When I retested with my simulator the RPM report is very stable - typically within +-1RPM. Tested with $30 set to 100 and 120 (simulator default).
I have not tested for threading accuracy since I do not have access to a precision lathe nor the necceseary measuring equipment. My only way to test is by enabling [PID-logging and using the PID tuner](https://github.com/grblHAL/core/wiki/Spindle-sync#tuning) in ioSender to get an idea about the accuracy from the reported position vs, the calculated position.
Interrupt load is kept low by using a MCU counter to count the pulses and setting the treshold for firing the interrupt based on the PPR setting - higher PPR -> fewer interrupts.

LaurentLM, 2025-07-15:
Spindle speed tolerance is disabled (set to 0). I believe I might have found a hardware issue that could cause that. I won't be able to check that until next week, but I have good hopes that this solves the issue 🤞.
I'll report my findings here. Closing issue for now.