grblHAL/iMXRT1062 #99: I can't compile to use 2 pwm laser spindles

distebia, 2025-07-14:
Hi, I'm trying to compile for imxrt1062, t41u5xbb, so I can have two spindles in laser mode (CO2 and diode) and use one at a time. I'm attaching parts of the code used in

my_machine_map.h
// Define auxiliary output pins
#define AUXOUTPUT0_PIN      (31U)
#define AUXOUTPUT1_PIN      (32U)
#if SPINDLE_ENABLE & (1<<SPINDLE_PWM2|1<<SPINDLE_PWM2_NODIR)
#define AUXOUTPUT0_PWM_PIN  (33U)
#else
#define AUXOUTPUT2_PIN      (33U)
#endif
#define AUXOUTPUT3_PIN      (12u) // Spindle enable
#define AUXOUTPUT4_PIN      (11u) // Spindle direction
#define AUXOUTPUT5_PIN      (13u) // Spindle PWM
#define AUXOUTPUT6_PIN      (19u) // Coolant flood
#define AUXOUTPUT7_PIN      (18u) // Coolant mist

// Define driver spindle pins
#if DRIVER_SPINDLE_ENABLE & SPINDLE_ENA
#define SPINDLE_ENABLE_PIN      AUXOUTPUT3_PIN
#endif
#if DRIVER_SPINDLE_ENABLE & SPINDLE_PWM
#define SPINDLE_PWM_PIN         AUXOUTPUT5_PIN
#endif
#if DRIVER_SPINDLE_ENABLE & SPINDLE_DIR
#define SPINDLE_DIRECTION_PIN   AUXOUTPUT4_PIN
#endif

my_machine.h
// If none are specified the default PWM spindle is instantiated.
// Spindle definitions can be found in grbl/spindle_control.h.
// More here https://github.com/grblHAL/Plugins_spindle
//#define SPINDLE0_ENABLE         SPINDLE_HUANYANG1
  #define SPINDLE0_ENABLE         SPINDLE_PWM0_NODIR
  #define SPINDLE1_ENABLE         SPINDLE_PWM2_NODIR
//#define SPINDLE1_ENABLE         SPINDLE_PWM0
//#define SPINDLE2_ENABLE         SPINDLE_PWM1
//#define SPINDLE2_ENABLE         SPINDLE_NONE
  #define SPINDLE_OFFSET          1 // Uncomment to enable settings for laser spindle XY-offset.
//#define SPINDLE0_ENABLE         SPINDLE_PWM0
//#define SPINDLE1_ENABLE         SPINDLE_PWM2 

config.h 
#if !defined N_SPINDLE || defined __DOXYGEN__
#define N_SPINDLE 2
#endif

/*! \def N_SYS_SPINDLE
\brief Defines number of simultaneously active spindles supported - minimum 1 (none), maximum 8.
*/
#if !defined N_SYS_SPINDLE || defined __DOXYGEN__
#define N_SYS_SPINDLE 1
#endif

$pins
[PIN:16,Feed hold]
[PIN:17,Cycle start]
[PIN:14,Emergency stop]
[PIN:15,Probe]
[PIN:20,X limit min]
[PIN:23,X limit max]
[PIN:21,Y limit min]
[PIN:28,Y limit max]
[PIN:22,Z limit min]
[PIN:36,Aux in 0,P0]
[PIN:30,Aux in 1,P1]
[PIN:34,Aux in 2,P2]
[PIN:35,Aux in 3,P3]
[PIN:41,Aux in 4,P4]
[PIN:28,Aux in 5,P5]
[PIN:29,Aux in 6,P6]
[PIN:2,X step]
[PIN:4,Y step]
[PIN:6,Z step]
[PIN:3,X dir]
[PIN:5,Y dir]
[PIN:7,Z dir]
[PIN:10,X enable]
[PIN:40,Y enable]
[PIN:39,Z enable]
[PIN:12,Spindle on]
[PIN:13,Spindle PWM]
[PIN:31,Spindle 2 PWM]
[PIN:18,Mist]
[PIN:19,Flood]
[PIN:32,Aux out 1,P0]
[PIN:11,Aux out 3,P1]
[PIN:33,Aux analog out 0,Spindle PWM]
[PIN:0,RX,UART1]
[PIN:1,TX,UART1]

but the $spindles command only lists the first spindle

distebia, 2025-07-16:
I changed my_machine.h like this and I can get m3/m4 on pwm0 but only m3 on pwm2

#define SPINDLE0_ENABLE         SPINDLE_PWM0
#define SPINDLE1_ENABLE         SPINDLE_PWM2_NODIR
//#define SPINDLE1_ENABLE         SPINDLE_PWM2_NODIR
//#define SPINDLE2_ENABLE         SPINDLE_NONE
#define SPINDLE_OFFSET          1 // Uncomment to enable settings for laser spindle XY-offset.

Basically, I'd also like full control of m3/m4 for pwm2.
I change spindles with m104p0 and m104p1, but I also see m104q0 and m104q1 in the documentation (is there a difference between p and q?)
Furthermore, I have the values $770 and $771 for laser spindle XY-offset...how do I manage them? Do I type m104p1 and set, for example, $770=-75? If so, it accepts it but has no effect.

terjeio, 2025-07-17:
> so basically I can't have 2 lasers with $32=1 in the same system?

You can, with controllers/boards that supports PWM0 and PWM1 spindles - very few does.

> what's missing is $32=1?

Yes, laser mode requires the PWM value to be precalculated in the foreground process, it is later set by the stepper interrupt routine and this should not be bogged down by unneccesary calculations. The ioports API does not support precalulation of the PWM value nor outputting it directly from interrupt contexts - and I have no plans to change that.

So the iMXRT1062 driver has to be updated to allow the PWM1 spindle to be configured. I have it running on my bench now, but not yet in a version that is ready for release.

distebia, 2025-07-17:
I seem to remember that MKS DLC32 has this feature, or maybe I'm wrong... even FluidNC works on MKS. It's a real shame not to have additional PWM spindles on the IMXRT1062, it's incredibly fast...

terjeio, 2025-07-18:
>  or do I have to specify PWM1??

Yes.

distebia, 2025-07-18:
[image]

[image]

terjeio, 2025-07-18:
delete line 2075

distebia, 2025-07-18:
I deleted line 2075 in driver.c and compiled it without problems. I loaded the firmware, but I don't have clear control over the second spindle, and to be honest, not even the first. I switch with m104p1 and launch m3s100, the meter reads 4.6v. If I launch m3s1000, it still reads 4.6v, with m5, it still reads 4.6v... I switch with m104p0 and I have no output, the meter reads 0v...
But the lines in my_machine.h should be like this:
#define SPINDLE1_ENABLE SPINDLE_PWM0
#define SPINDLE2_ENABLE SPINDLE_PWM1
or
#define SPINDLE0_ENABLE SPINDLE_PWM0
#define SPINDLE1_ENABLE SPINDLE_PWM1
or exactly how?

terjeio, 2025-07-20:
> But the lines in my_machine.h should be like this:
#define SPINDLE1_ENABLE SPINDLE_PWM0
#define SPINDLE2_ENABLE SPINDLE_PWM1
or
#define SPINDLE0_ENABLE SPINDLE_PWM0
#define SPINDLE1_ENABLE SPINDLE_PWM1
or exactly how?

Both should work, but the second is preferred. 
There was a mixup of min/max RPM for the second PWM spindle settings that could cause it to malfunction - I have fixed this and also updated the code so that no reeboot is needed to enable laser mode for it.
FYI my T41U5XBB board is currently in use so I have used a Pro variant to test the code - this uses nearly identical pin mappings so I assume the T41U5XBB would work with the new code.

distebia, 2025-07-20:
[image]

It seems we're there, I have the pwm2 output from aux2..is that correct? I send m104p0 from the console and I can turn on the PWM (CO2 laser, primary for me), I do the frame and everything is ok. I launch m104p1 from the console and I can turn on the PWM2 on aux2 with g1m3s100f1000 (diode, secondary for me) but it does nothing if I send g1m4s100f1000, it doesn't turn on at all...also I compensated the offsets with $770=-65.50 and $771=3.55...after launching m104p0 I block the primary laser, I draw a square and write it, everything is perfect..if I type m104p1 I should switch to the secondary laser, I redo the frame of the same square, the PWM2 turns on but does not move with the set offsets, so it is in another position...where am I going wrong?

distebia, 2025-07-20:
also the commands $520 and $521 accept them but one overwrites the other, if I set $520=1, I save and reread and I also get $521=1... I don't understand how I have to proceed

distebia, 2025-07-21:
OK, today I'll try adding the line to offset.c and see if it changes anything. What's not clear to me is the spindle switching mode. Are the m104p0 and m104p1 commands correct? Is that all there is to it? When I execute m104p1, in addition to switching the output to pwm2, shouldn't it also apply the offsets set to $770 and $771?

terjeio, 2025-07-21:
> When I execute m104p1, in addition to switching the output to pwm2, shouldn't it also apply the offsets set to $770 and $771?

If you use the ioSender spindle selector drop-down M104 commands are sent so no need to add them from gcode or the MDI. Offsets are applied when a spindle selected event is fired and the target spindle is not the default spindle and it is flagged as laser capable, and reverted when switching from it to another.

The [$spindlesh](https://github.com/grblHAL/core/wiki/Report-extensions#spindles) command lists the spindles with capabilities, e.g:
```
[SPINDLE:0|0|0|*DLIRV|PWM|0.0,1000.0]
[SPINDLE:1|1|0|DLIRV|PWM2|0.0,1000.0]
```

The `L` in the `DLIRV` string above is for laser capability.

terjeio, 2025-07-23:
$770 and $771 specifies how much to move the axes on a spindle change, it does _not change any offsets_. You will have to set either G92 or a G5x offset to whatever position is needed after the switch. Should the plugin be updated to have settings for which (if any) offset to change - G92 or the currently active offset?

distebia, 2025-07-28:
Perfect, thanks. I've got something on my mind and can't seem to figure it out... I need to control a solenoid valve that switches the air flow from CO2 to the diode after a tool change. My idea is to create a Lightburn macro and send m104p1, then g55 (where I first set the offsets with l10 p2) and a command to enable the aux output like m64 p0 and m65p0... but when I send the m64 p0 command I get the error code 20... do I need to enable something in my_machine.h and recompile?

distebia, 2025-07-29:
solved, with #define SPINDLE1_ENABLE SPINDLE_PWM1_NODIR I made the aux0 output available and with m64p0 I measure 5v on aux0... if I do $pins I see however that aux1 is occupied by enable for pwm2 and therefore not available as a relay output. do I necessarily have to have the enable pin to control the diode (pwm2)?

terjeio, 2025-07-30:
>  do I necessarily have to have the enable pin to control the diode (pwm2)?

With the current code - yes. A PWM only spindle could be added but I guess that would have to use a "virtual" on pin to keep the on state available for the core.

terjeio, 2025-08-05:
Is the PWM2 spindle enabled in settings?

distebia, 2025-08-05:
I'm a careless person, it was disabled on $511, I'm waiting for the solenoid valve (to be controlled with m64) to divert the air flow from the CO2 to the diode and I'm ok, thanks

terjeio, 2025-08-27:
IIRC Aux2 is driven by a ULN2003, this has a VCE(sat) of around 1V according to the datasheet (I measure ~0.7V at the output on my STM32 board with a 2K7 load). So Aux2 will never reach 0V in its off state. You can use use the TTL Aux2 output instead?